dresden-elektronik/deconz-rest-plugin #5672: Socket-outlet, Exxact, Wiser, double, 16 A, connected (WDE002182)

cobch, 2022-01-14:
This device shows up in DeConz WebUI with all information such as Power (W) and Consumption (kWh), but only on/off and state is provided in API to i.e Home Assistant. After adding  "ZHAConsumption" and a "ZHAPower" items with DDF (see #5563), power and consumption sensors are added in Home Assistant but state is reported as "unknown".

[image]

## Device

https://www.se.com/ww/en/product/WDE002182/socketoutlet-exxact-wiser-double-16-a-connected-white/

- Product name: Socket-outlet, Exxact, Wiser, double, 16 A, connected.
- Manufacturer: Schneider Electric.
- Model identifier: SOCKET/OUTLET/2.
- Device type : 
  - Sensor
  - Switch

## Screenshots

### Node Info panel
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes

[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

### Simple Metering
[image]

[image]

### Electrical measurement
[image]

### Diagnostics
[image]

[image]

Smanar, 2022-01-15:
On your DDF

```
      "uuid":[
         "$address.ext",
         "0x01",
         "0x0b04"
      ],
```

Need to have 0x06

```
            "read":{
               "at":"0x0400",
               "cl":"0x0b04",
               "ep":0,
               "fn":"zcl"
            },
```

Same ep is 0x06.

You have this error 4/5 time, after that if you read the value in deconz it will update it in HA but only 1 time, If I m right after that need to set the bind/report (that sill missing in the DDF).

cobch, 2022-01-15:
I have tried to look around at the DDF and to change the endpoint to 0x06 via the deconz WebUI and save the DDF file again. But unfortunately I did not manage it. Should just I write "0x06"  next to endpoint instead of "auto" for each case? It seems to be a value between 0 and 255 that is accepted there... Maybe should I edit the json file directly but did not know how to do yet.

ON/OFF:
[image]

POWER SENSOR:
[image]

CONSUMPTION SENSOR:
[image]

Here is all my DDF preview:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "SOCKET/OUTLET/2",
  "product": "SOCKET/OUTLET/2",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/schneider/exxact.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "read": {
            "at": "0x0400",
            "cl": "0x0b04",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0400",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val; }"
          }
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}

cobch, 2022-01-15:
@Smanar I did manage to change the endpoint to 0x006. A bit tricky since you have to select the subdevice name directly (in grey). I have now the consumption reporting correctly in HA (((((-:
BUT the power is not reported at all and still the "unavailable" property is still there. Do you know why?

Smanar, 2022-01-16:
>Should just I write "0x06" next to endpoint instead of "auto" for each case?

Ha yes, I had same issue too, I think on my side I have used "6".

>Maybe should I edit the json file directly but did not know how to do yet

On the windows title you have the DDF path to edit it.

>BUT the power is not reported at all

```
"name": "state/power",
"refresh.interval": 300,
"read": {
"at": "0x0400",
"cl": "0x0b04",
"ep": 0,
"fn": "zcl"
},
```

On your capture you have it on the cluster 0x0702 not the 0x0B04

cobch, 2022-01-16:
@Smanar Your are so great. It works now for Power report (-: I have filled "6" and changed the cluster to "0x702" and see what I get in HA:
[image]

I still have some questions (-: It sounds to take some time to get the power updated from Deconz to HA. How can I force to update every second? I get as well a strange unit with coma "XX,XXX". Is it possible to change it in deconz DDF? See the graph below:
[image]

Last question is regarding the binding in my DDF. It looks like that. Normal or should I edit something more?

[image]

Smanar, 2022-01-16:
>I get as well a strange unit with coma "XX,XXX". Is it possible to change it in deconz DDF?

You mean 55,900 W ? Yeah I think 55 W is enought, but idk if it s the decimal or the 1k ? It seem for the 1k it's "," and the decimal "." ? You can just divide the value

Before
`if (Attr.val != -32768) { Item.val = Attr.val; }`

After
`if (Attr.val != -32768) { Item.val = Attr.val / 1000; }`

>I still have some questions (-: It sounds to take some time to get the power updated from Deconz to HA. How can I force to update every second?
>Last question is regarding the binding in my DDF. It looks like that. Normal or should I edit something more?

Same answer, Its strange your device work without binding/report, it was probably set by deconz by defaut.
Select an attribute, double click on it, on the new popup press "read config", you will have the setting used by deconz for the reporting (Min/Max/Change)

You can use the one you prefer on the last tab you have showed https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel

Min is the minimum time between all report
Max is the maximum time between all report
Change is the minimum change that can trigger a report.

But 1 report every second is realy too much for me.

cobch, 2022-01-16:
It was kW so the "1000" divisor made it (-:
Last things to deal with is the min/max and threshold to trigger as you mentioned above. According to the link you send me, I would need to drag and drop but I can not have boh windows (main window and DDF  editor) open at the same time. It surely a nwebie question but see how it is presented below:

Main window:
[image]

DDF editor:
[image]

Mimiix, 2022-01-17:
> Same answer, Its strange your device work without binding/report, it was probably set by deconz by defaut.

Polling probably.

Smanar, 2022-01-17:
>Polling probably.

Yep, possible too.

For your issue https://forum.phoscon.de/t/ddf-editor-how-to-drag-bindings/1351/11
If it can help ? Else can edit manully the DDF ? you have the path in the window title ? (I don't see it on yur capture, but the ddf is empty)

cobch, 2022-01-17:
You @Smanar are so great!!! I finally managed with the floating windows to edit the bindings and with the followings settings I get a really good report of power and consumption that react in real time when I switch ON and OFF the plug. SO happy!

[image]

[image]

[image]

I will wait a bit before closing the issue to see if something wrong happen in the near futur. Otherwise I wonder how I can give this work to the community for integrating this outlet in the next deconz update?

Mimiix, 2022-01-17:
> Otherwise I wonder how I can give this work to the community for integrating this outlet in the next deconz update?

That would be great! Can you share the ddf json here?

manup, 2022-01-18:
I now have a set of Wiser devices here for testing including socket and the plug, a DDF would be very welcome.

cobch, 2022-01-18:
Here comes my DDF. Thanks again for your support.

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "SOCKET/OUTLET/2",
  "product": "SOCKET/OUTLET/2",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/schneider/exxact.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 6,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x06",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300,
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 6,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val; }"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "read": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 6,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val / 1000; }"
          }
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 6,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val; }"
          }
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x06",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 6,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 6,
            "eval": "Item.val = Attr.val"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 6,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}

Mimiix, 2022-01-18:
@cobch Let me know if this works properly for you, if so, i'll make a PR out of it :)
Can you let us know?

cobch, 2022-01-18:
@Mimiix So far everything works really great!

cobch, 2022-01-24:
@Mimiix After one week everything is working good. I have tested the config with a second outlet and everything (power and consumption) came outside the box (-:

Mimiix, 2022-01-24:
@cobch great! can you make a Pr? Just go to the file folder and add it in this repo.

cobch, 2022-01-26:
@Mimiix I really would like to learn how to do it. Could you helping making a PR?

cobch, 2022-01-27:
Thanks @Mimiix . It went ok but unfortunately I have created to PR exactly identical. How can I remove one of them?

Mimiix, 2022-01-27:
I don't see an identical one? I've merged it anyway :)

cobch, 2022-01-28:
Great to see that I managed it (-:

cobch, 2022-01-28:
Should I close this issue?

jlaur, 2022-02-01:
@Mimiix, @cobch - I was going to create an issue for this, then found this as already created, even with merged PR. Now the question: My power outlet is LK Wiser, i.e. introduced for the Danish market. It seems similar, but a little different:

[image]

What is required to get this one supported too?

jlaur, 2022-02-01:
@Smanar - thanks. I think I'll need to gain some deeper understanding before being able to do this, as I'm totally new to deCONZ. I have a fresh RaspBee installation with the LK power outlet paired and working. I tried to copy the DDF contents into /usr/share/deCONZ/devices/wiser/fuga_socket_outlet.json with just these modifications based on the assumption that it would be quite similar:

```
"modelid": "LK/OUTLET/1",
"product": "LK/OUTLET/1",
"path": "/devices/wiser/fuga_socket_outlet.json",
```

I restarted deCONZ and it seems that this file was picked up. However, when using the REST API and query for `/sensors`, it doesn't show up. It's still available as `/lights/2` as it was also before adding the DDF:

```
{
    "etag": "f28e05304b75b460c8b1d104f8242cda",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-02-01T21:42Z",
    "manufacturername": "Schneider Electric",
    "modelid": "LK/OUTLET/1",
    "name": "On/Off plug-in unit 2",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "002.006.018 R",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:00:00:00:00:00:00:00-00"
}
```

But no data for Simple Metering or Electrical Measurement.

I then tried to follow this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

But here I hit the wall, at least for today, for example when trying to determine "good endpoint". I wanted to verify if this is correct for the version of the outlet:

```
"uuid": [
"$address.ext",
"0x06",
"0x0b04"
],
```

Smanar, 2022-02-01:
The endpoint is the 06 on top, on blue square it's clusters.

Try with "gold" as status in the DDF.

To be sure for the path, select your node, and make "edit DDF", you will have on the window title the path of the DDF used.

jlaur, 2022-02-06:
> Try with "gold" as status in the DDF.

Thanks, that did the trick! Or rather, it led me to understanding that this DDF is quite new and replacing legacy code, thus found this:

[image]

Now it works perfectly.

closed 2022-03-07 by github-actions[bot]