dresden-elektronik/deconz-rest-plugin #5676: Develco Products EMIZB-132 (EMI Norwegian HAN) reports voltage at 10x proper value

WTell, 2022-01-17:
## Describe the bug

The Develco Products EMIZB-132 (EMI Norwegian HAN) reports voltage at 10x proper value. For example, the device is currently reporting voltage at 2334 V, while actual voltage is measured to be 233 V.

## Steps to reproduce the behavior

Read the voltage from the device. The device reports voltage at 10x nominal values. It is assumed that this behaviour is intentional on the part of the manufacturer, perhaps trying to achieve a higher resolution within the confines of an integer value.

## Expected behavior

Voltage is expected to be reported in a nominal, three-digit range, preferably around 230 or 400 V. The device reports a "AC Current Divisor" value of 10, which I assume that deCONZ should divide the reading by before it is reported by the API.

## Screenshots

Data from the device:

```
    "manufacturername": "Develco Products A/S",
    "modelid": "EMIZB-132",
    "name": "Electricity Sensor",
    "state": {
      "current": 490,
      "lastupdated": "2022-01-17T08:45:51.436",
      "power": 0,
      "voltage": 2333
    },
    "swversion": "2019-07-11 09:10",
    "type": "ZHAPower",
```
[image]

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 0x26720700
 - deCONZ version: 2.13.4-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

SwoopX, 2022-01-17:
Deconz uses the appropriate voltage divisor through #5084 therefore, I currently do not see any todo (expect when it is proven that the device reported divisor is wrong).

WTell, 2022-01-17:
Thanks for the quick feedback! I don't understand how it could be using the right divisor, seeing as the sensor in the screenshot above currently reports a voltage of 2383 through the API (GET http://deconz/api/key/sensors/29 ):

```{
	"config":{
		"on":true,
		"reachable":true
	},
	"ep":2,
	"etag":"",
	"lastannounced":"2022-01-13T12:00:01Z",
	"lastseen":"2022-01-17T11:34Z",
	"manufacturername":"Develco Products A/S",
	"modelid":"EMIZB-132",
	"name":"Electricity Sensor",
	"state":{
		"current":500,
		"lastupdated":"2022-01-17T11:34:41.471",
		"power":0,
		"voltage":2383
	},
	"swversion":"2019-07-11 09:10",
	"type":"ZHAPower",
	"uniqueid":"00:15:"
}```

It seems to me that the calculation is not performed. Am I missing something?

I assume that the values reported in the GUI are pre-calculation, as they should be the raw attributes returned by the ZigBee device. Is this correct?

Smanar, 2022-01-17:
RMS voltage = 2322 and divisor is 10 so need to have 2322 /10 = 232 V
RMS current = 96 and divisor is 10 so need to have 9.6 A

Both have issue

WTell, 2022-01-17:
It seems that there is also an issue with the value reported in `state.current`. I assume that it is deCONZ' intention to report this value in whole amperes. The number returned through the API seems to have been multiplied by 10, and not divided by 10.

Further, there is a problem with how the API reports power draw. For reasons unknown to me, the physical metering device is shown as two separate sensors through the API - one is a `ZHAConsumption` sensor and the other is a `ZHAPower` sensor. These are from the same physical device (meter reader). The output is: 

```
  "27": {
    "config": {
      "interfacemode": 2,
      "on": true,
      "reachable": true
    },
    "ep": 2,
    "etag": "3e...",
    "lastannounced": "2022-01-13T12:00:01Z",
    "lastseen": "2022-01-17T11:52Z",
    "manufacturername": "Develco Products A/S",
    "modelid": "EMIZB-132",
    "name": "Electricity Sensor",
    "state": {
      "consumption": 76216000,
      "lastupdated": "2022-01-17T11:52:51.563",
      "power": 1617
    },
    "swversion": "2019-07-11 09:10",
    "type": "ZHAConsumption",
    "uniqueid": "00::5e-02-070"
  },
  "29": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 2,
    "etag": "ef...",
    "lastannounced": "2022-01-13T12:00:01Z",
    "lastseen": "2022-01-17T11:52Z",
    "manufacturername": "Develco Products A/S",
    "modelid": "EMIZB-132",
    "name": "Electricity Sensor",
    "state": {
      "current": 480,
      "lastupdated": "2022-01-17T11:52:51.565",
      "power": 0,
      "voltage": 2382
    },
    "swversion": "2019-07-11 09:10",
    "type": "ZHAPower",
    "uniqueid": "00::5e-02-0b04"
  }
```

ZHAPower sensor reports state.power as 0, while the ZHAConsumption sensor reports state.power as 1617 (which is the correct wattage from 0x0304. The ZHAPower sensor should report the same reading, as some applications now report my house as drawing 480 amps at 2382 volts but at 0 watts.

Smanar, 2022-01-17:
>For reasons unknown to me, the physical metering device is shown as two separate sensors through the API - one is a ZHAConsumption sensor and the other is a ZHAPower sensor. These are from the same physical device (meter reader)

That is device fault ^^. Deconz make a 1 sensor by endpoint, but it can remove the not working one (the state/power in the ZHAPower).
There is so much issues, the 2 firsts can be solved with DDF, but removing the not working field isn't working yet (there is someone that is trying that on the forum, without success yet)
So idk if the corrective need to be using DDF or code edition ?

SwoopX, 2022-01-17:
Reopening this one as I should've looked a bit longer at the PR I'm referencing and the associated issue, however, this is not a bug.

What we see here is presumably a change in divisors/multipliers due to changes made in the device firmware. In the initial device support request, they were all set to 1. Having read up a bit on the device back in the days, I know Develco rejected to provide more up-to-date firmwares and ppl got stuck on several issues... However, iirc, they have recently shared one of the latest.

Coming back to the topic, the key in how to treat the raw values actually appears to be the firmware version. To make things even more complicated, the firmware version resides in the 0x8000 attribute of the basic cluster, which deconz cannot properly interpret, as they chose to use the raw byte values of the 3 or 4 bytes provided to represent the version (that's why just nonsense is displayed).

That being said, I do not see any viable change in the current (legacy) code to be made and even feasable with rational effort to get this on the right track. Alternatively, as already suggested by Smanar, a DDF should be created for that device. For it going public, value calculation should take the firmware version into account.

Sorry it's not as easy as changing one or two lines, but we've already seen rising differences in behavior based on different firmware versions.

Smanar, 2022-01-18:
So yes, go for the DDF ^^.
Do you know how it work ? It's not the first plug added with DDF for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5672

You can start from a previous one https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577
Or from scratch using the editor https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

WTell, 2022-02-09:
Thanks for the reminder, github bot ^_^

Posting this just to keep the issue open – I’ll look into this over the weekend.

WTell, 2022-02-13:
So, I promised I'd look into this over the weekend – which I did – but I must admit that gettint the DDFs sorted out (i.e., learning the syntax and how to get this right) is taking more time that I'd care to admit. I'll keep you posted as the work progresses.

My goal is to create a DDF that takes into account the firmware version like @SwoopX pointed out, so that it could be included in the official release and benefit everybody.

SwoopX, 2022-02-13:
@WTell Check out the current code base to learn how the true firmware version is read and mapped to `swversion` https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/develco

Just reuse the JS and check one of the DDFs.

closed 2022-03-14 by github-actions[bot]