dresden-elektronik/deconz-rest-plugin #5679: Xiaomi power plug (ZNCZ04LM) sporadically not showing Consumption, Power and Temperature

marc965, 2022-01-17:
## Describe the bug

I have several zigbee devices installed.
After a long time of succesfull operation need to reboot.
After reboot all Devices work fine accept all 5 Xiaomi power plug (lumi.plug.mmeu01, ZNCZ04LM, all Version 09-06-2019)
are sporadically not delivering Power, Consumption and Temperature Values.
The on/off functionality is still available.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]
[image]
[image]
sporadically you see "nicht verfügbar" (not available) and the graphs are broken
[image]
[image]
[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
     NAS: Synology VMM, Home Assistant core-2021.12.9, supervisor-2021.12.2, Home Assistant OS 7.1
 
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
  Home Assistant deCONZ Add-on 6.11.1

 - Firmware version: (26xxyy00)
   266B0700

 - deCONZ version: (2.xx.yy)
   2.13.04 / 18.12.2021
 
- Device: (ConBee I / ConBee II / RaspBee I / RaspBee II):
  Conbee II

 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II:
   yes

 - Is there any other USB or serial devices connected to the host system? If so: Which?
    no

## deCONZ Logs

21:30:43:436 poll node 04:cf:8c:df:3c:8a:2b:36-15-000c
21:30:43:437 Poll ZHAPower sensor node Power 18
21:30:43:555 skip create link for 0xB2A3 (lqi: 0) - 0x258F (lqi: 0)
21:30:43:701 Poll APS request to 0x04CF8CDF3C8A2B36 cluster: 0x0B04 dropped, cluster doesn't exist
21:30:44:437 poll node 04:cf:8c:df:3c:8a:2b:36-16-000c
21:30:44:438 Poll ZHAConsumption sensor node Consumption 19
21:30:44:701 Poll APS request to 0x04CF8CDF3C8A2B36 cluster: 0x0702 dropped, cluster doesn't exist
21:30:45:437 poll node 54:ef:44:10:00:02:ab:4c-15-000c
21:30:45:438 Poll ZHAPower sensor node Power 22
21:30:45:702 Poll APS request to 0x54EF44100002AB4C cluster: 0x0B04 dropped, cluster doesn't exist
21:30:46:437 poll node 04:cf:8c:df:3c:8a:2b:36-01
21:30:46:437 Poll light node Zisterne_Motor
21:30:46:461 Node data 0x54ef44100002ab4c profileId: 0x0104, clusterId: 0x000C
21:30:46:462 SensorNode id: 22 (Power 22) no longer available
21:31:51:437 poll node 04:cf:8c:df:3c:8a:2a:cd-15-000c
21:31:51:438 Poll ZHAPower sensor node Power 16
21:31:51:611 Wait 0s till query finished
21:31:51:700 Poll APS request to 0x04CF8CDF3C8A2ACD cluster: 0x0B04 dropped, cluster doesn't exist
21:31:53:436 poll node 04:cf:8c:df:3c:8a:2a:cd-16-000c
21:31:53:437 Poll ZHAConsumption sensor node Consumption 17
21:31:53:700 Poll APS request to 0x04CF8CDF3C8A2ACD cluster: 0x0702 dropped, cluster doesn't exist
21:33:59:117 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('21', 'Consumption 21', 'ZHAConsumption', 'lumi.plug.mmeu01', 'LUMI', '04:cf:8c:df:3c:8a:34:0e-16-000c', '09-06-2019', '{"consumption":67290,"lastupdated":"2022-01-17T20:33:00.157"}', '{"on":true,"reachable":false,"temperature":1500}', '{"d":81,"ep":22,"in":[12,0],"p":260}', 'normal', '1', '2022-01-17T20:33Z', '2022-01-14T15:21:01Z')
21:33:59:118 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('20', 'Power 20', 'ZHAPower', 'lumi.plug.mmeu01', 'LUMI', '04:cf:8c:df:3c:8a:34:0e-15-000c', '09-06-2019', '{"current":0,"lastupdated":"2022-01-17T20:33:00.157","power":0,"voltage":229}', '{"on":true,"reachable":false,"temperature":1500}', '{"d":81,"ep":21,"in":[12,0],"p":260}', 'normal', '1', '2022-01-17T20:33Z', '2022-01-14T15:21:01Z')
21:33:59:119 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('22', 'Power 22', 'ZHAPower', 'lumi.plug.mmeu01', 'LUMI', '54:ef:44:10:00:02:ab:4c-15-000c', '09-06-2019', '{"current":153,"lastupdated":"2022-01-17T20:32:24.875","power":36,"voltage":233}', '{"on":true,"reachable":true,"temperature":1800}', '{"d":81,"ep":21,"in":[12,0],"p":260}', 'normal', '1', '2022-01-17T20:33Z', '2022-01-14T15:21:02Z')
21:33:59:119 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('23', 'Consumption 23', 'ZHAConsumption', 'lumi.plug.mmeu01', 'LUMI', '54:ef:44:10:00:02:ab:4c-16-000c', '09-06-2019', '{"consumption":34380,"lastupdated":"2022-01-17T20:32:24.875"}', '{"on":true,"reachable":false,"temperature":1800}', '{"d":81,"ep":22,"in":[12,0],"p":260}', 'normal', '1', '2022-01-17T20:33Z', '2022-01-14T15:21:02Z')
21:35:49:611 Poll ZHAConsumption sensor node Consumption 12
21:35:49:872 Poll APS request to 0x04CF8CDF3C802626 cluster: 0x0702 dropped, cluster doesn't exist
21:35:50:610 poll node 04:cf:8c:df:3c:8a:34:0e-01
21:35:50:611 Poll light node BORG008htpc
21:35:50:662 Poll APS request to 0x04CF8CDF3C8A340E cluster: 0x0006 dropped, values are fresh enough
21:35:51:612 poll node 04:cf:8c:df:3c:80:26:26-15-000c
21:35:51:613 Poll ZHAPower sensor node Power 11
21:35:51:873 Poll APS request to 0x04CF8CDF3C802626 cluster: 0x0B04 dropped, cluster doesn't exist
21:35:52:609 poll node 54:ef:44:10:00:02:ab:4c-15-000c
21:35:52:610 Poll ZHAPower sensor node Power 22
21:35:52:873 Poll APS request to 0x54EF44100002AB4C cluster: 0x0B04 dropped, cluster doesn't exist
21:35:53:611 poll node 04:cf:8c:df:3c:8a:2b:36-01
21:35:53:611 Poll light node Zisterne_Motor
21:35:53:662 Poll APS request to 0x04CF8CDF3C8A2B36 cluster: 0x0006 dropped, values are fresh enough
21:35:54:610 poll node 04:cf:8c:df:3c:8a:2b:36-01
21:35:54:610 Poll light node Zisterne_Motor
21:35:54:662 Poll APS request to 0x04CF8CDF3C8A2B36 cluster: 0x0006 dropped, values are fresh enough
21:35:55:612 poll node 04:cf:8c:df:3c:8a:2a:cd-01
21:35:55:613 Poll light node Zisterne_Elektronik
21:35:55:662 Poll APS request to 0x04CF8CDF3C8A2ACD cluster: 0x0006 dropped, values are fresh enough
21:35:56:612 poll node 84:71:27:ff:fe:a7:af:7c-01

## Additional context

Mimiix, 2022-01-18:
Asked devs to check.

marc965, 2022-01-18:
Xiaomi power plug (ZNCZ04LM)
What do you mean with "Asked devs to check."?

Mimiix, 2022-01-18:
> What do you mean with "Asked devs to check."?

I asked the developers to check this issue.

> Can i do something to help solving this problem?

 Did you read other issues on this plug on the git? I remember some being there. Other than that, waiting :)

marc965, 2022-01-18:
> > What do you mean with "Asked devs to check."?
> 
> I asked the developers to check this issue.
> 
> > Can i do something to help solving this problem?
> 
> Did you read other issues on this plug on the git? I remember some being there. Other than that you have to wait sweat_smile

Several issues found on the git but all closed.

Mimiix, 2022-01-18:
Did you read them trough? Might be some explaining perhaps in there. Either way, lets wait for a dev to check in.

marc965, 2022-01-18:
> Did you read them trough? Might be some explaining perhaps in there. Either way, lets wait for a dev to check in.

Ok, thanx for helping.
Another information, perhaps it can help: turning this Device from "off" to "on" and back to "off" seems to have an impact on this device. It suddenly begin to operate normaly for a litte while...

This evening i will try to unpair and repair a device to check if it works properly...

felgy86, 2022-01-19:
Hi.

I´ve been crawling the web for some days now trying to find someone with the same issue as I.... and found it now! (I found several similar, but now exactly the same...)
I have 2x ZNCZ04LM plugs for some while now. One of them is working without issues, but the second one after reset started behaving exactly has you describe. All 3 sensors are with missing data sporadically:

Example of today:

[image]

I'm running last HA version (2021.12.10) in a VM and ConBee II with last version (2.13.04) and Firmware (26660700).
I´ve now reset the plug so many times without improvement...

Comparing debug logs (good plug vs. bad plug), the good one is updating the sensors a lot more often than the bad one. 

I do see some suspicious errors (by the way, the same as yours...):
22:10:12:475 Poll APS request to 0x04CF8CDxxxxx cluster: 0x0702 dropped, cluster doesn't exist
22:10:13:468 Poll APS request to 0x04CF8CDxxxxx cluster: 0x0B04 dropped, cluster doesn't exist

Let me know if further details are needed.

Thanks!

SwoopX, 2022-02-06:
Tldr;

The recommended conbee/raspbee firmware version to run the device is 0x26720700. If that was not the case while first pairing the device, the firmware should be updated and the devices reset and re-paired. Additionally, with the upcoming stable release, I added a respective DDF to hopefully force the device into some more sane behavior (I suspect the plug's firmware to be buggy, as it behaves differently then all other devices).

felgy86, 2022-02-06:
I am (was) running last available firmware version for ConBee 2 and multiple resets were performed... nothing changed. I'll wait for next firmware release. 

Thanks

felgy86, 2022-02-07:
@SwoopX, I´ve just upgraded to 2.14.01 / 06/02/2022 and issue seems to be solved.

Appreciate your time looking at this issue!

Cheers

marc965, 2022-02-08:
Seem to find a solution for this now: 

deCONZ: Home Assistant Integration 6.11.1
Conbee II: Version 2.13.04 / 18.12.2021, Firmware: 266B0700

I play arround with deCONZ and find this:
[image]

Default for "Reporting Interval" is 300. I change it to 10!
[image]

Please see the difference.
Can anyone confirm this?

What about the Conbee II CPU Load? Where can i check this?
Is there an impact on the Conbee II CPU Load?

marc965, 2022-02-11:
@SwoopX 
i think its a question of the device firmware, deCONZ Application and signal strength. Now, with the new firmware and Application, it seems to work better.

Do you have an suggestion for a device like this with more reliability? I need Power, Consumption and Current without on/off functionality but as plug.

I am very interested in Zigbee technics and how to edit own DDF files, but i am not ready yet to understand this. I' am staying tuned...

marc965, 2022-03-05:
Still not Working for me...
deCONZ Current version: 6.12.0
ConBee II 2.14.01 / 6.2.2022 Firmware 266B0700
Home Assistant OS 7.4, supervisor-2022.01.1, core-2022.3.1

Delete all Xiaomi power plug (ZNCZ04LM) devices and repair.
No change in behaviour...

marc965, 2022-04-01:
Unfortunately with the last Update a further bug has come up with last deCONZ update:

The PowerOn Setting is now more configurable, it was possible before:
[image]
 now its greyed out:
[image]

closed 2022-04-30 by github-actions[bot]