dresden-elektronik/deconz-rest-plugin #5682: ThirdReality Water Leak Sensor - 3RWS18BZ

devWaves, 2022-01-19:
## Device

- Product name: THIRDREALITY Zigbee Water Leak Sensor
- Manufacturer: Third Reality. Inc
- Model identifier: 3RWS18BZ
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Switch ?

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

### On/Off

### Other clusters that are not mentioned above
   ### Power Configuration
   ### IAS Zone
   ### OTAU

devWaves, 2022-01-19:
I was able to get the water leak sensor detected by the rest api using the following files and rebuilding the rest api

I referenced this PR to build the files https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5518

de_web_plugin_private.h , de_web_plugin.cpp , bindings.cpp

[de_web_plugin.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7893985/de_web_plugin.txt)
[de_web_plugin_private.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7893986/de_web_plugin_private.txt)
[bindings.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7893987/bindings.txt)

I have also attached the DDF before and after rebuilding the rest api
[ddfBeforePRChange.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7893995/ddfBeforePRChange.txt)
[ddfAfterPRChange.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7893996/ddfAfterPRChange.txt)

Smanar, 2022-01-19:
Hello, but there is a problem, ATM c++ code edition are locked, we need to make only with DDF (if it's possible)
So without editing the 3 files (it's for that the PR was not merged)

devWaves, 2022-01-19:
hey @Smanar . I included the ddf file after the PR was made. ddfAfterPRChange.txt . I was under the assumption that was the file that would be needed. Sorry I just started looking at this yesterday and havent had a chance to look at ddf docs and how all this stuff works

Smanar, 2022-01-20:
No problem, you have a short description here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
But yes for this kind os sensor (that use the cluster IAS) better to start from another DDF sensor.

To test it, you just need a deconz versioin that support DDF without file compilation.

And BTW, there is probably an issue for this maufacture https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5681#issuecomment-1017526763

devWaves, 2022-01-30:
I created a pull request. https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5728

The DDF is working for me and reporting when water is detected. It is also reporting battery level but is it divided by 2. Not sure why as the DDF configured value is not divided by 2. Eitherway it works good enough. I just have to multiple battery percentage by 2 to get real value

Smanar, 2022-01-30:
On config/battery on expression try to replace
`Item.val = Attr.val`

by

`Item.val = 2* Attr.val`

devWaves, 2022-01-30:
using ```Item.val = 2* Attr.val``` doesn't seem to change anything

the battery value is correct when I look at the node in deconz, but through the rest api it is halfed

Smanar, 2022-01-30:
It's the cluster 0x0001, attribute 0x0021 ?
Try with the "strict mode", it 's probably due to legacy code.

devWaves, 2022-01-31:
[image]

[image]

shows correctly in deconz. not in phoscon

I will try the strict mode etc later

Mimiix, 2022-05-07:
@Smanar what's the status of this one?

zahti, 2022-05-07:
I just bought 2 of these sensors, and would like to start using them. Please let me know how I can help!

Smanar, 2022-05-08:
Ha, manup have skipped this one, there is a PR, working, only the battery can missing, but since the time, perhaps it work better now.

@zahti can you try the DDF on the PR ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5728

PR #6826 (2023-04-16): Fix Third Reality water sensor (3RWS18BZ) battery value
Update 3RWS18BZ_water_sensor.json to fix #5728 PR comments and fix #5682 issue
closed 2022-06-07 by github-actions[bot]