dresden-elektronik/deconz-rest-plugin #5686: TS004F 58 - 4 Gang Scene Switch - _TZ3000_xabckq1v

superMoYoX, 2022-01-20:
## Device

- Product name: Scene Switch / 4 Buttons Zigbee Scene Switch
- Manufacturer: _TZ3000_xabckq1v
- Model identifier: TS004F
- Device type : Please remove all unrelated device types. 
   - Switch
  

## Screenshots

[image]

### Groups

### Scenes

### On/Off

### Level Control

superMoYoX, 2022-01-20:
Hi !

Already shows up in Phoscon but acts weirdly (it seems to broadcast on/off commands to all ZigBee lights/sockets of my setup).

[image]

manup, 2022-01-20:
I have not much experience with Tuya based switchtes but the broadcasting looks like a default configuration of the switch, perhaps we need to create a group binding for it. In the DDF file we can use the new "auto" group mechanism as it is used by Kobold and the ubisys C4.

superMoYoX, 2022-01-21:
Hi !

BTW, this device can "switch" to another mode by long pressing the two right buttons.
Some people using a "ZigBee" plugin on Jeedom servers have succeeded to use properly this device:
https://community.jeedom.com/t/ajout-switch-tuya-4-boutons-ts004f/59494/152

Smanar, 2022-01-21:
This device is broken https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5574

Smanar, 2022-01-23:
@superMoYoX I think you are using jeedom ? so not able to compile the code to test changes ?

superMoYoX, 2022-01-25:
Hi @Smanar !

I can share my time with you if you want. (with my poor french's english)
I don't know how to compile the "code" you said, but actually I'm not stranger to programming stuff.

If you have a tutorial or smthing like that or can explain to me, I'll take it :)

Smanar, 2022-01-25:
Don't worry, if you have problem I m on discord too, and I m french.
So the PR is in waiting list and already working for some device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5702

The problem is in jeedom deconz is integrated in jeedom, so IDK if they change some setting like path. To be sure to not break something, just take a look if you can find the lib file on "classic" path` /usr/share/deCONZ/plugins/libde_rest_plugin.so`
If you have this file, exaclty at this path, you can make a backup of it in another folder, so you will be able to make a rollback if we have problem.

You need too a recent deconz , and I know it's jeedom that decide or not if you can update it by defaut.

If you have an old deconz version or have a different path, I prefer give up, don't want to break your machine.

superMoYoX, 2022-01-27:
Hi Smanar !

I looked for the file you mentionned. The only one I found on my Debian is under :
/usr/share/deCONZ/plugins/libde_rest_plugin.so
as you told.
I can't be sure he's the Jeedom one or an old let by a previous install try ...
BTW, the deCONZ version downloaded by Jeedom when doing a "local install" is 2.13.04 .

If you have some time, I 'm available this week-end to talk with you on Discord (in FR :) )
Just let me know !

Smanar, 2022-01-28:
All seem fine.
Just make a backup of the lib file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder, it will be the only file modified on your OS

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

```
sudo apt install deconz-dev
git clone --branch switch_issue_1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Thoses lines will install the dev package, clone the code, compile it and replace the file.

After that just need to re-include the device (deconz will crash during the file replacement if you don't stop it, so you surely need to restart it)

The compilation will use a lot your machine the first time, but next one (if needed) will be faster

If you have error message, stop all and give me the error (on discord if you want)

superMoYoX, 2022-01-29:
Hi Smanar !

Ok I'm available to try everthing you want ^^'.

I did all you wrote.
I installed QT5-Default btw to compile the code.

But the qmake exited with errors and didn't create the libdre_rest_plugin.so file :

`de_web_plugin.cpp:15930:27: warning: unused variable ‘item’ [-Wunused-variable]
             ResourceItem *item = sensor->item(RConfigGroup);
                           ^~~~
de_web_plugin.cpp:15926:14: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
         else if (sensor->modelId() == QLatin1String("HG06323") ||                // LIDL Remote Control
              ^~
de_web_plugin.cpp:15931:13: note: ...this statement, but the latter is misleadingly indented as if it were gua                               rded by the ‘if’
             if (!item)
             ^~
de_web_plugin.cpp:15931:18: error: ‘item’ was not declared in this scope
             if (!item)
                  ^~~~
de_web_plugin.cpp:15931:18: note: suggested alternative: ‘tm’
             if (!item)
                  ^~~~
                  tm
de_web_plugin.cpp: At global scope:
de_web_plugin.cpp:15993:1: error: expected declaration before ‘}’ token
 }
 ^
make[1]: *** [Makefile.Release:1381: release/de_web_plugin.o] Error 1
make[1]: *** Attente des tâches non terminées....
make[1] : on quitte le répertoire « /root/deconz-rest-plugin »
make: *** [Makefile:40: release] Error 2
`

Smanar, 2022-01-29:
Arf, my bad, coding typo
Have updated the code

```
sudo apt install deconz-dev
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

superMoYoX, 2022-01-29:
Done !
File replaced, deconz service restarted, TS004F re-included !

superMoYoX, 2022-01-29:
It seems to not broadcast anymore, I receive correct values when pressings buttons (1002, 2002, 3002, 4002 for button 1, 2, 3, 4)

Smanar, 2022-01-29:
ok so the PR is working too with this device ?

superMoYoX, 2022-01-29:
What's the PR?

superMoYoX, 2022-01-29:
OK, the pull request
I got it 😅
Thanks a lot for the work Smanar

Smanar, 2022-01-30:
Lol, yep, I have presumed the answer was "yes" so have finished the PR.

BusyGorilla, 2022-02-01:
hello,
I've just got one of these TS004F switches and found its not working with deCONZ as it should :(

I've read through all the above posts and am I right in saying that the switch will get added and then work with deCONZ on future updates? (sorry for the noob question :) )

Smanar, 2022-02-01:
I can't say https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733

This PR is more a fix than a new device, but sensible.

closed 2022-03-02 by github-actions[bot]

jokob-sk, 2022-06-19:
Would love to see support for these as well

Smanar, 2022-06-19:
This device is supported, or at least particvally, I have just see it in the code.

jokob-sk, 2022-06-29:
Thanks @Smanar for commenting - installed the newest version and indeed it seems to be working well. Tried it previously on an older version where the device was recognised as a light source :D - thanks for the great work team!