dresden-elektronik/deconz-rest-plugin #5688: Add 3A Nue Smart Door Sensor (Feibit FB56-DOS06HM1.1)

paulharken, 2022-01-21:
Hello

First time request here! Just want to say thank you dresden for all the hard work you do. I've been able to setup our home the way we want with a lot of different devices and this is the first time I've encountered any issues. Closed Issue #4400 seems very close to it however the device I have is model "FB56-DOS06HM1.1", whereas issue 4400 has it listed as "FB56-DOS06HM1.3"

Device is branded as a "3A/Nue ZigBee Smart Door Sensor", model "HGZB-13A". Paired it using Phoscon GW web interface via the sensor panel and it didn't show up as a sensor; it shows up as a "warning device" under lights. It gets sent over to HomeAssistant as well but shows up as a siren, not a door sensor (assuming because deconz is reading it as part of an intrusion alarm system?).

Screenshots below of what information I can gather. 

Can I get any more information? 

**Node info**
[image]

**Basic Cluster**
[image]

**Identify Cluster**
[image]

**IAS WD Cluster**
[image]

**IAS Zone Cluster**
[image]

paulharken, 2022-02-13:
Not stale.

paulharken, 2022-06-12:
Still open

Mimiix, 2022-06-12:
Sure

paulharken, 2022-07-04:
Still open

paulharken, 2022-07-26:
Still open

paulharken, 2022-08-18:
Still waiting

Mimiix, 2022-08-18:
@Smanar  can this be done with ddf?

Smanar, 2022-08-18:
I think yes, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Feibit,co.ltd",
  "modelid": "FB56-DOS06HM1.1",
  "vendor": "3A/Nue",
  "product": "Smart Door Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    },
    {
      "type": "Warning device",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

closed 2022-09-17 by github-actions[bot]