dresden-elektronik/deconz-rest-plugin #5693: MOES BHT-002GBLZB Thermostat

MrTh0m, 2022-01-22:
## Device
- Product name: MOES BHT-002GBLZB 
- Manufacturer: _TZE200_aoclfnxz
- Model identifier: TS0601
- Device type : router

## Screenshots
[image]
[image]

### Basic
[image]
### Groups
[image]
### Scenes
[image]
### Tuya specific Cluster
no attributes
### OTAU
[image]
### TIME
no attributes

Smanar, 2022-01-22:
It's not this one ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5468

MrTh0m, 2022-01-22:
> It's not this one ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5468

It seem, yes. I've looking for but not found. 
So is that mean is already supported ?

Smanar, 2022-01-22:
Yeap, but with tuya all is possible (they changes lot of thing regulary), so you can have issue.
The device can need 2 inclusions and can be invisible on phoscon, you need to check direclty the API (phoscon/help/API information) or use a third app.

MrTh0m, 2022-01-23:
Exactly,
At 1st pairing  the thermostat was seen but cannot do lot on Home Assistant. 
At 2nd pairing, after deleting the object. thermostat is seen, i've perform a reboot of deconz and then i don't see it anymore.
It is remain on HA. I can 
- Up or down the degrees 
- Shutdown completly the thermostat
- Switch on screen light
but that is all

MrTh0m, 2022-01-23:
Thanks you for the API's page, never going here before. 
So for my thermostat, i see that :

       "22": {
            "config": {
                "heatsetpoint": 1900,
                "locked": false,
                "mode": "heat",
                "offset": -300,
                "on": true,
                "preset": "program",
                "reachable": true,
                "schedule": {},
                "schedule_on": false,
                "setvalve": false
            },
            "ep": 1,
            "etag": "7c32a10f3bb08bf413be2e2926023b4a",
            "lastannounced": "2022-01-22T14:30:44Z",
            "lastseen": "2022-01-23T09:23Z",
            "manufacturername": "_TZE200_aoclfnxz",
            "modelid": "TS0601",
            "name": "Thermostat 22",
            "state": {
                "lastupdated": "2022-01-23T08:58:43.219",
                "on": false,
                "temperature": 1650
            },
            "type": "ZHAThermostat",
            "uniqueid": "8c:f6:81:ff:fe:56:74:63-01-0201"
        }

Smanar, 2022-01-23:
You can use config/preset with "auto" and "program"
You have the childlock config/locked
You have the config/mode with "off" and "heat"

MrTh0m, 2022-01-24:
You mean, i need to append some values ?

Smanar, 2022-01-24:
Heu not append, edit, you need to use thoses fields to control the device, if I m right you have a combo somewhere in HA to control the preset for exemple, for mode it's on widget bottom. For childlock  IDK where you can see it in HA, but the device have it.

MrTh0m, 2022-01-29:
Yes, i've widget to control HAVC but goes wrong. Only 
- up or down temperature work
- and shutdown or turn on thermostat.
The others function fail.

HA discover well the thermostat from Deconz but all funtion or preset don't work.

Below, what HA know from thermostat :

> hvac_modes:
>   - auto
>   - heat
>   - 'off'
> min_temp: 7
> max_temp: 35
> preset_modes:
>   - auto
>   - boost
>   - comfort
>   - complex
>   - eco
>   - holiday
>   - manual
> current_temperature: 17.5
> temperature: 18
> preset_mode: null
> offset: -300
> locked: false
> friendly_name: Thermostat 22
> supported_features: 17

For eg, below when i hit "confort" preset, i've issue :

[image]

Smanar, 2022-01-29:
>You can use config/preset with "auto" and "program"

You have not the "program" preset somewhere ? or can use custom one ?

>You have the config/mode with "off" and "heat"

Here I think "desactivé" and "marche forcée" will work.

Yo have the childlock in HA
>locked: false

Sorry, but all tuya TRV/thermostat are differents, and they have all differents features, it s hard for the HA plugin to make a widget compatible for all possible stuff.

closed 2022-02-27 by github-actions[bot]