dresden-elektronik/deconz-rest-plugin #5697: MOES - TV01 Zigbee Thermostat Radiator Valve

nukleuz80, 2022-01-22:
Hello,

can you please add the following device?
After pairing as sensor i get three new sensors (temperature, humidity and alarm)

## Device

- Product name: MOES - TV01 Zigbee Thermostat Radiator Valve
- Manufacturer: _TZE200_e9ba97vf
- Model identifier: TS0601
- Device type : 
  - Thermostat

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
  
[image]

  - Node Info panel
  
[image]

### Basic
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### Tuya Specific Cluster
[image]

The three sensors from the API after pairing:

`{
    "config": {
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "aa0bbe17dbde1c62bb33e00ddd90250e",
    "lastannounced": "2022-01-22T17:00:17Z",
    "lastseen": "2022-01-22T21:21Z",
    "manufacturername": "_TZE200_e9ba97vf",
    "modelid": "TS0601",
    "name": "Temperature 112",
    "state": {
        "lastupdated": "none",
        "temperature": null
    },
    "type": "ZHATemperature",
    "uniqueid": "2c:11:65:ff:fe:c6:1c:47-01-0402"
}`

`{
    "config": {
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "edf815f1d0438673e184b987706adaf3",
    "lastannounced": "2022-01-22T17:00:17Z",
    "lastseen": "2022-01-22T21:21Z",
    "manufacturername": "_TZE200_e9ba97vf",
    "modelid": "TS0601",
    "name": "Humidity 113",
    "state": {
        "humidity": null,
        "lastupdated": "none"
    },
    "type": "ZHAHumidity",
    "uniqueid": "2c:11:65:ff:fe:c6:1c:47-01-0405"
}`

`{
    "config": {
        "enrolled": 0,
        "humiditymaxthreshold": null,
        "humidityminthreshold": null,
        "melody": null,
        "on": true,
        "pending": [],
        "preset": null,
        "reachable": true,
        "temperaturemaxthreshold": null,
        "temperatureminthreshold": null,
        "volume": null
    },
    "ep": 1,
    "etag": "2d7824e88ceba17184533a1493c70484",
    "lastannounced": "2022-01-22T17:00:17Z",
    "lastseen": "2022-01-22T21:21Z",
    "manufacturername": "_TZE200_e9ba97vf",
    "modelid": "TS0601",
    "name": "Alarm 114",
    "state": {
        "alarm": null,
        "lastupdated": "none"
    },
    "type": "ZHAAlarm",
    "uniqueid": "2c:11:65:ff:fe:c6:1c:47-01-0500"
}`

Smanar, 2022-01-23:
This is not in the code at all, no clones, nothing, and as the moment lot of thing are locked, I think this device will be not included before some time, sorry.

closed 2022-02-22 by github-actions[bot]

PastCoder, 2022-10-20:
@nukleuz80 Did you find meanwhile any way to make the _TZE200_e9ba97vf work? I am asking because I am as well trying to figure out how to make it work in deconz (and iobroker). I see it in the deconz GUI but nowhere else. 
Maybe there are now more people interested so that I could make sense to raise the request again!?

Smanar, 2022-10-20:
Nothing will move before this PR will be merged https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054
The next step in deconz will be propably TRV.

nukleuz80, 2022-10-21:
> @nukleuz80 Did you find meanwhile any way to make the _TZE200_e9ba97vf work? I am asking because I am as well trying to figure out how to make it work in deconz (and iobroker). I see it in the [...]

No, i sent it back and baught a new one which is compatible...