dresden-elektronik/deconz-rest-plugin #5700: Device name

rohrsh, 2022-01-23:
## Device

- Product name: Aeotec Smart Plug
- Manufacturer: LDS
- Model identifier: ZB-ONOFFPlug-D0005
- Device type: Switch (with power monitoring)

This is an Australian 240v smart plug. 
https://www.smarthome.com.au/smartthings-smart-plug.html

I have the same situation as this query from 2019: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2121
The device registers just fine as an on/off plug, **but no power monitoring.** 

I thought the device was identical to the "Samsung SmartPlug 2019 Model ID: 7A-PL-Z-J3" but I don't see the power measurements coming through to Home Assistant. 

Thanks for any help. Happy to provide more information. 

## Screenshots

[image]

[image]

[image]

[image]

[image]

Mimiix, 2022-01-23:
Missing screenshots.

rohrsh, 2022-01-24:
Screenshots updated. Please let me know if any information is still missing. My apologies for any errors, this is my first device submission. Thank you!

Mimiix, 2022-01-24:
Looks good, thanks :)

cammurray, 2022-02-14:
@rohrsh by rebooting, do you mean rebooting the device, deconz, or hass? My device is stuck showing the on/off but no power monitoring. Have reset deconz and hass a couple of times but don’t get the sensor showing.

rohrsh, 2022-02-14:
Actually I'm not sure. All I know is that I didn't change any part of the Aeotec device config and a couple of days later I started getting power reports in Home Assistant. During that time I did work on other items and reboot my Home Assistant. 

In a troubling turn of events, I did Zigbee remove and then re-add my device in another part of my house and after 24 hrs still hasn't started any power monitoring. 

So there's room for improvement with this device

cammurray, 2022-02-17:
Interesting, mine has been up for about 3-4 days now, but I'm not getting anything in home assistant yet. deCONZ GUI seems to be able to read the data if I click the read button.

[image]

[image]

SwoopX, 2022-02-17:
I would normally close this one, as it is indeed 100% identical to the Samsung but I also recall it was painful to have it working back in the days...

I'd recommend to quickly generate a DDF for it, as a fully working one should be generated out of the box. It should be stored at the same location where the other files are residing. Just ensure the status is set to `Gold` instead of `Draft`.

You can also share it here for some tweaking and for it to be integrated in future releases.

cammurray, 2022-02-17:
Ok. I seem to have found the 'magic sauce' for this one while going through the ~hundred comments on the Samsung equivelant for this device.

When pairing this device, the device adds a 'light' type sensor in deconz but this sensor doesn't expose any power metering through. If you -wait- for the device to go out of pairing mode, and then go to add a new device in deconz, deconz finds another device somehow - it's this second pairing that gives you the power monitoring capabilities. This is.. weird..?

@SwoopX let me know if I can do anything to assist make this a bit easier. I'd be happy to 'generate a DDF' for it, but I don't know what that is :)

reg98643t, 2022-02-24:
> Ok. I seem to have found the 'magic sauce' for this one while going through the ~hundred comments on the Samsung equivelant for this device.
>
> When pairing this device, the device adds a 'light' type [...]

Could you elaborate on what exactly you did to get it to work?

cammurray, 2022-02-24:
@reg98643t what I did is actually in my comment..

"When pairing this device, the device adds a 'light' type sensor in deconz but this sensor doesn't expose any power metering through. If you -wait- for the device to go out of pairing mode, and then go to add a new device in deconz, deconz finds another device somehow - it's this second pairing that gives you the power monitoring capabilities"

Let me know how that's not clear and I'll try rephrase.

reg98643t, 2022-02-24:
@cammurray 

For the second device addition, in Deconz did you try to add a light, sensor or switch? How long did you wait after pairing? Did you have anything plugged into the plug and was it drawing power?

Essentially I've tried what you said and it didn't work and I'm wondering if there are some important factors not mentioned.

reg98643t, 2022-02-24:
I think I might have worked it out, I plugged a light in and left it going then redid the process and it seems to have been the missing link.

reg98643t, 2022-02-24:
Trying a second one and it hasn't worked, might need to be drawing power for some time before it works?

reg98643t, 2022-02-25:
Yeah I have not been able to replicate the process for any further plugs.

reg98643t, 2022-02-25:
@SwoopX how can @cammurray do the DDF thing?

reg98643t, 2022-02-25:
@SwoopX @cammurray 

I have got it working and I have a working DDF, any idea how I can get the DDF out of the Deconz VM?

SwoopX, 2022-02-26:
You get a DDF when richt clicking on the respective node (must have been previously selected) and then choose `Edit DDF`. That can be saved at a location of your choice. It should already contain pretty much everything which is required, but the bindings might be an exception (they previously must have been triggered to show up).

I, personally, can always copy and text from/to a VM, be it via ssh, vnc, or just vmware workstation 🤷‍♂️ 

Would be great to share it here upfront to check it for completeness.

SwoopX, 2022-02-27:
Hm, from what I can tell, the DDF should more look like the innr SP 120, except for the first few lines of course
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/innr/sp_120.json

rohrsh, 2022-03-21:
Still unresolved as far as I am aware

reg98643t, 2022-03-21:
Sorry guys, I have it working perfectly for myself but I have no way to share it yet.

rohrsh, 2022-04-15:
Zzzzzzzzz

reg98643t, 2022-04-19:
Sorry all haven't had time to try and work out how to share it yet

nemccarthy, 2022-05-15:
Any luck on this?

Smanar, 2022-05-15:
Try this DDF
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "LDS",
  "modelid": "ZB-ONOFFPlug-D0005",
  "vendor": "Aeotec",
  "product": "Aeotec Smart Plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val * 10;"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

closed 2022-06-13 by github-actions[bot]

Man-Grove, 2023-11-29:
This worked for me, including power metering, at the second pairing (the initial didn't work). However, it's identified as a light and not as a plug, which is very weird (and wrong?). It's the only plug I have that's identified as a light.

Man-Grove, 2023-11-30:
Disregard that -- I don't have the same version...