dresden-elektronik/deconz-rest-plugin #5706: MOES ZTS-EUB 3 gang switch

Try2Fly, 2022-01-24:
## Device

- Product name: Moes ZigBee Smart Switch
- Manufacturer: _TZE200_tz32mmtza
- Model identifier: TS0601.
- Device type : 
 
  - Switch (3-gang)
 

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

sn0wf4ke, 2022-02-08:
I have this one as 2 Gang Switch. It seems to be the same.
I`m on version 2.4.0 beta
Here are my screenshots:

[image]
[image]
[image]
[image]
[image]

closed 2022-03-09 by github-actions[bot]

forast3r, 2022-08-19:
> I have this one as 2 Gang Switch. It seems to be the same. I`m on version 2.14.0 beta Here are my screenshots:
> 
> [image] [image] [image] [image] [image]
> 
> and here is a debug log from pairing:
> 
> ```
> 17:47:55:533 device announce 0x60A423FFFE2FC20E (0x0408) mac capabilities 0x80
> 17:47:55:533 set fast probe address to 0x60A423FFFE2FC20E (0x0408)
> 17:47:55:533 FP indication 0x0000 / 0x0013 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:55:533                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:55:534 device announce 0x60A423FFFE2FC20E (0x0408) mac capabilities 0x80
> 17:47:55:536 ZDP get node descriptor for 0x0408
> 17:47:55:536 ZDP get node descriptor for 0x0408
> 17:47:55:916 DB UPDATE device_descriptors SET data = x'0240800210525200002c520000', timestamp = 1644338875 WHERE device_id = (SELECT id FROM devices WHERE mac = '60:a4:23:ff:fe:2f:c2:0e') AND endpoint = 0 AND type = 2
> 17:47:55:917 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240800210525200002c520000', 1644338875 FROM devices WHERE mac = '60:a4:23:ff:fe:2f:c2:0e'
> 17:47:55:921 FP indication 0x0000 / 0x8002 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:55:921                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:55:922 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) cluster 0x8002
> 17:47:55:927 [2] get active endpoints for 0x60a423fffe2fc20e
> 17:47:55:928 ZDP get active endpoints for 0x0408
> 17:47:56:012 FP indication 0x0000 / 0x8005 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:012                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:012 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) cluster 0x8005
> 17:47:56:013 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) clear timeout on cluster 0x8005
> 17:47:56:019 [3] get simple descriptor 0x01 for 0x60a423fffe2fc20e
> 17:47:56:019 ZDP get simple descriptor 0x01 for 0x0408
> 17:47:56:076 FP indication 0x0000 / 0x8005 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:076                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:076 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) cluster 0x8005
> 17:47:56:082 wait response fastEnddeviceProbe() 0x60A423FFFE2FC20E, elapsed 62 ms
> 17:47:56:196 DB UPDATE device_descriptors SET data = x'0104015100010400000400050000ef0219000a00', timestamp = 1644338876 WHERE device_id = (SELECT id FROM devices WHERE mac = '60:a4:23:ff:fe:2f:c2:0e') AND endpoint = 1 AND type = 4
> 17:47:56:196 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104015100010400000400050000ef0219000a00', 1644338876 FROM devices WHERE mac = '60:a4:23:ff:fe:2f:c2:0e'
> 17:47:56:206 FP indication 0x0000 / 0x8004 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:206                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:206 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) cluster 0x8004
> 17:47:56:207 ZDP indication search sensors 0x60A423FFFE2FC20E (0x0408) clear timeout on cluster 0x8004
> 17:47:56:212 [4.1] Get manufacturer code
> 17:47:56:213 [4.2] get basic cluster attr 0x0004 for 0x60a423fffe2fc20e
> 17:47:56:539 FP indication 0x0104 / 0x0000 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:539                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:539 Clear fast probe timeout for cluster 0x0000, 0x60A423FFFE2FC20E
> 17:47:56:545 [4.1] Get model ID
> 17:47:56:545 [4.2] get basic cluster attr 0x0005 for 0x60a423fffe2fc20e
> 17:47:56:752 FP indication 0x0104 / 0x0000 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:752                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:752 Clear fast probe timeout for cluster 0x0000, 0x60A423FFFE2FC20E
> 17:47:56:758 [4.1] Get sw build id
> 17:47:56:758 [4.2] get basic cluster attr 0x4000 for 0x60a423fffe2fc20e
> 17:47:56:974 FP indication 0x0104 / 0x0000 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:974                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:56:974 Clear fast probe timeout for cluster 0x0000, 0x60A423FFFE2FC20E
> 17:47:56:980 [4.1] Get date code
> 17:47:56:980 [4.2] get basic cluster attr 0x0006 for 0x60a423fffe2fc20e
> 17:47:57:246 FP indication 0x0104 / 0x0000 (0x60A423FFFE2FC20E / 0x0408)
> 17:47:57:246                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:47:57:246 Clear fast probe timeout for cluster 0x0000, 0x60A423FFFE2FC20E
> 17:47:57:893 Bind response success for 0x00158d0007063fa1 ep: 0x01 cluster: 0x0001
> 17:47:58:092 Bind response success for 0x00158d0007063fa1 ep: 0x01 cluster: 0x0201
> 17:47:58:092 skip configure report for cluster: 0x0201 attr: 0x0008 of node 0x00158D0007063FA1 (seems to be active)
> 17:47:58:167 GW firmware version is up to date: 0x26660700
> 17:47:58:189 ZCL attribute report 0x00158D000774348B for cluster: 0x0201, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 17:48:00:708 FP indication 0x0104 / 0x000A (0x60A423FFFE2FC20E / 0x0408)
> 17:48:00:709                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:48:03:816 FP indication 0x0104 / 0xEF00 (0x60A423FFFE2FC20E / 0x0408)
> 17:48:03:817                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:48:03:989 FP indication 0x0104 / 0xEF00 (0x60A423FFFE2FC20E / 0x0408)
> 17:48:03:990                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:48:04:076 ZCL attribute report 0x00124B002549D4E5 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000
[... cut ...]
> 17:48:20:888 ZCL attribute report 0x7CB03EAA0A008CEA for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000
> 17:48:25:583 FP indication 0x0104 / 0xEF00 (0x60A423FFFE2FC20E / 0x0408)
> 17:48:25:584                       ...     (0x60A423FFFE2FC20E / 0x0408)
> 17:48:25:003 ZCL attribute report 0x7CB03EAA0A04CEFA for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000
> 17:48:32:412 emit event/tick: 0x7cb03eaa00a8f18b
> ```

Got a similar one (with same manufacturer and product numbers). Did someone manage to get it working?