dresden-elektronik/deconz-rest-plugin #5709: Request to add Mini Zigbee Switch TMZ02

mielikki33, 2022-01-25:
## Device

- Product name: Tuya_TMZ02L
- Manufacturer: Unknown
- Model identifier: Unknown
- Device type : 
  - Switch (2 gangs)

Reference : https://zigbee.blakadder.com/Tuya_TMZ02L.html

## Screenshots

### Basic
[image]
[image]
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### On/Off
[image]
[image]

### Otau
[image]

Mimiix, 2022-01-25:
Can you please update the post with the Model id / manu name? Its in your secondscreen:
[image]

mielikki33, 2022-01-25:
Thank you, I just updated the post.

mielikki33, 2022-01-25:
I hope the information is complete for you. I have 11 modules to install in my home, but I cannot pair them in Phoscon with Home Assistant.

thanks in advance

Mimiix, 2022-01-25:
We need to wait for a dev. But be adviced: It could take a while to be integrated as DDF's aren't yet useable for Switches / Tuya. Even more so: If a dev can help integrate it, you need the possibility to compile yourself (with help of the dev ofc) on a native installation. That being said: You can't do that with the Home Assistant Addon or Docker version.

mielikki33, 2022-01-25:
OK thanks for your answer. I'm on Raspberry Pi 4 with a Conbee 2 key.

closed 2022-02-24 by github-actions[bot]

ContraNL, 2022-03-11:
If there is more information needed for the devs please let us know. I too have multiple units and want to integrate them in Home Assistant.

Smanar, 2022-03-11:
Hello,
The problem is this device use a new tuya cluster, the 0xe001
We are waiting for the "classic one" the 0xEF00 is supported in the DDF core, so I think this new one will not be supported since long time.

ContraNL, 2022-03-11:
Do you mean it won’t be supported for a long time or you think it will be supported soon?

Smanar, 2022-03-12:
It will need more time to be supported (than devices using the classic tuya cluster)

ContraNL, 2022-03-12:
@Smanar do you know of a similar switch that is supported?

Smanar, 2022-03-12:
What are you searching ? A module without neutral for 16A ? (16A is a lot for light)

ContraNL, 2022-03-12:
@Smanar no, with neutral and just for lighting.

Smanar, 2022-03-12:
So just a module, with neutral, for lightning ? no dimmer ?
You can find some of them in Legrand, Tuya, Aqara, Philips, Sonoff, Loratap, ect ...

ContraNL, 2022-03-12:
@Smanar, do you have a tuya example that works? I now have multiple units of this device (40 euro’s worth) and they are not working. I don’t want the same to happen seeing I often have to wait 4 weeks to receive them from e.g. aliexpress to find out they are not supported o Phoscon/Home Assistant.

ContraNL, 2022-03-12:
[image]

I know this will probably work seeing the dimmers are supported. But I don’t want RF :S

Smanar, 2022-03-13:
You have for exemple this one with neutral [_TYZB01_iuepbmpv](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3595)

This one with DDF without neutral https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5839

But tuya are cheaper, if you want to paid so much take the philips one, I can see it at 43 euros on amazon.

ContraNL, 2022-03-13:
@Smanar, [#3595](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3595) is the Moes I was talking about. Even though I don’t want RF, I ordered these now. I rather have support, then not have RF as an option.
I will use these until there is support for the above mentioned units.

Thanks for the information.

Smanar, 2022-03-13:
I don't see something about RF on the device,, but there is so much tuya clone .....

ContraNL, 2022-03-13:
[image]
See image, this is the unit mentioned in your first link and mine. These units have zigbee as well as RF and on Aliexpress there is mention of support on Home Assistant, so I hope they will work.

Smanar, 2022-03-13:
On my first link, I haven't the same https://zigbee.blakadder.com/Moes_MS-104Z-1.html for the _TYZB01_iuepbmpv
It s a classic problem with tuya, even on the same order, you can have different model, visualy same, but not reconized by the same manaufacture name.

ContraNL, 2022-03-13:
@Smanar, mmm then I hope they will work seeing some people are reporting they do in de comments of the product on aliexpress.

Smanar, 2022-03-13:
Ha yes, right, I m just seing "wifi" on it.
If you have the manufacture name, I can say if deconz is able to support it or not (if it don't work natively, for exemple if you have one without the tuya cluster), but the problem is you have it only when you connect the device ...

ContraNL, 2022-03-13:
@Smanar I know… that’s the problem. They should that it on the aliexpress product pages. But we’ll see, I just hope it had thr “classic” cluster so you can help me if it’s not supported 😉