dresden-elektronik/deconz-rest-plugin #5710: Cover direction for another cover module

pyrech, 2022-01-25:
## Describe the bug

I configured my new cover switch modules - named LoraTap ZigBee Curtain Switch module. They are recognized inside Phoscon as `_TZ3000_dbpmpco1` and they are "working". But I had to invert the motor wiring to make the up/down direction correct. I saw that's a frequent problem with some cover modules and I guess we need a similar fix as https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5415.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.13.04)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2022-01-25:
@Smanar There we go again 😂

Smanar, 2022-01-25:
Lol, a tuya reversed ? something new.

So first, a personnal thx to @pyrech , then there is 2 approchs now for this kind of bug
- Code editing like your link
- Create a DDF file like here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5447#issuecomment-962651289

If it s just a fix, can be validated fast, but there is so much PR in waiting list (and I can't give a date), and you need to compile the code to test the result.
Using DDF, it's fast, just need to use deconz but have 1 issue, IDK if it's still a probleme, but it work only in "Strict mode".

Choose your weapon.

Mimiix, 2022-01-25:
I prefer the ddf fix

pyrech, 2022-01-26:
Thanks for your fast answers.

I'm not sure to understand what is really DDF. By browsing through code and issues, I understand that its some json files that are bundled by the plugin to override configuration for specific devices, am I correct?

About the strict mode, I also don't know what it implies or not :sweat_smile:

Smanar, 2022-01-26:
>By browsing through code and issues, I understand that its some json files that are bundled by the plugin to override configuration for specific devices, am I correct?

Yes, In the future the plugin will use only DDF.

About DDF, there is a video now https://www.youtube.com/watch?v=rgTQpOVg8w0
A link here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

And can work natively if you have a deconz version > 2.13 without compilation. But realy easier if you can access to the GUI.

>I also don't know what it implies or not

Me too, lol. But and IDK if the issue persist, but the test is fast. And I think it depend if the device is included with DDF, or if the DDF is used after the inclusion.

pyrech, 2022-01-27:
Thanks for the explanations.

> Just give me your chose, I will explain you how to do

If DDF is the future, let's go for it (but no emergency, even if my covers' state are a bit broken I can still manage them from HomeAssistant by inverting the motor cables :wink:)

About the DDF editor inside the GUI, I must admit I'm a bit lost even after reading the cheat-sheet and the linked issue :confused:

Smanar, 2022-01-28:
Will be easy, realy few users have problem with it (generaly it's for bind)

- Enable DDF, can take "basic" to start, then "hybrid" and "strict"
- Right clic on the node, then edit DDF

If you are lucky the DDF is almost filled else
- Fill first the "basic cluster identification", take "gold" as status for tests.
- Drag and drop a "window covering device" from right to left panel
- Configure state/on and state/lift with

ZCL for both
Endpoint 0x01 (or just1)
Cluster 0x0102
Attribute 0x0008

Then on the DDF menu can take "hot reload"

You can take a look on logs with the flag "DDF" you will some usefull logs.

And now reverse the return
- select state/lift
- edit the "expression" field, use instead `Item.val = 100 - Attr.val`
- select state/open
- edit the "expression" field, use instead `Item.val = (100 - Attr.val) < 100`
- can remove state/on and state/bri
- re make "hot reload"

Smanar, 2022-01-30:
@pyrech you are stucked somewhere ?

pyrech, 2022-02-01:
Sorry, did not have access to my installation until now.

Sadly as you guessed I'm stuck on the bindings part. I tried many of the tips given in the linked thread, even by accessing to the VNC session with a native client on my Ubuntu laptop but no luck until now, the binding window remains empty and I can't add anything :confused:

[image]

Did I miss something?

pyrech, 2022-02-01:
Ok, I found how to detach the "Cluster info" the binding window from the main window. So I followed the rest of your instructions but I probably did something wrong as I don't see any change. Here is the preview DDF:

```{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dbpmpco1",
  "modelid": "TS130F",
  "product": "TS130F",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/lift",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          }
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift",
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = 100 - Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = (100 - Attr.val) < 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0102"
    }
  ]
}```

Smanar, 2022-02-02:
You have 2 TYPE_WINDOW_COVERING_DEVICE on your DDF, only one is usefull (one with (1) and one with (2) ) The second one is more completed.

```
          "name": "state/open",
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = (100 - Attr.val) < 100",
            "fn": "zcl"
          }
```

You have set the binding, but still need the report.
To set the binding drag&drop on the left panel the cluster > done
To set the reportijg drag&drop the attribute 0x0008 (from the same cluster) on the right panel.

If values are not set by defaut can use thoses one :
```

        min = 1;
        max = 300;
        change = 1;
```

pyrech, 2022-02-02:
I completed the DDF with your latest instructions but I see no change in the cover behaviour. Do I need to reboot something to make it work? Or what is wrong with ma config?

Here is the latest DDF:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_dbpmpco1",
  "modelid": "TS130F",
  "product": "TS130F",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift",
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = 100 - Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "read": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = (100 - Attr.val) < 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2022-02-03:
`  "status": "Draft",`
Mean disabled, try direclt with "gold"

pyrech, 2022-02-03:
I changed the status then tried with all 3 modes but nothing changed. In logs I see some lines related to cover. Maybe some could explain what's wrong :

```
22:46:21:165 Websocket 172.30.32.1:52852 send message: {"attr":{"id":"5","lastannounced":null,"lastseen":"2022-02-03T21:46Z","manufacturername":"_TZ3000_dbpmpco1","modelid":"TS130F","name":"Volet roulant bureau","swversion":null,"type":"Window covering device","uniqueid":"a4:c1:38:1d:96:4c:e1:7a-01"},"e":"changed","id":"5","r":"lights","t":"event","uniqueid":"a4:c1:38:1d:96:4c:e1:7a-01"} (ret = 335)
22:46:21:167 ZCL read attr 0xA4C1381D964CE17A, ep: 0x01, cl: 0x0102, attr: 0x0008, mfcode: 0x0000, aps.id: 129, zcl.seq: 216
22:46:21:208 a4:c1:38:1d:96:4c:e1:7a-01/state/open expression: Item.val = (100 - Attr.val) < 100 --> false
22:46:21:224 a4:c1:38:1d:96:4c:e1:7a-01/state/lift expression: Item.val = 100 - Attr.val --> 100
22:46:21:239 0xA4C1381D964CE17A: update ZCL value 0x01/0x0102/0x0008 after 0 s
22:46:21:240 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:241 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:243 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:244 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:257 ZCL read attr 0xA4C1381D964CE17A, ep: 0x01, cl: 0x0102, attr: 0x0008, mfcode: 0x0000, aps.id: 131, zcl.seq: 217
22:46:21:299 a4:c1:38:1d:96:4c:e1:7a-01/state/open expression: Item.val = (100 - Attr.val) < 100 --> false
22:46:21:303 a4:c1:38:1d:96:4c:e1:7a-01/state/lift expression: Item.val = 100 - Attr.val --> 100
22:46:21:305 0xA4C1381D964CE17A: update ZCL value 0x01/0x0102/0x0008 after 0 s
22:46:21:306 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:307 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:308 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
22:46:21:310 No Tuya productId entry found for manufacturername: _TZ3000_dbpmpco1
```

Smanar, 2022-02-04:
```
22:46:21:299 a4:c1:38:1d:96:4c:e1:7a-01/state/open expression: Item.val = (100 - Attr.val) < 100 --> false
22:46:21:303 a4:c1:38:1d:96:4c:e1:7a-01/state/lift expression: Item.val = 100 - Attr.val --> 100
```

Can you check if you need open = false, lift  = 100 or the reverse at this moment ?

If all if fine you need to have exaclty thoses value in the device json.

pyrech, 2022-02-08:
If I understand your question, it seems to me that using `(100 - Attr.val) < 100` and `100 - Attr.val` expressions are correct. 

But the thing I dont' understand is that in no case the direction of the cover change, whatever configuration I tried

Smanar, 2022-02-08:
Ha, I have a doubt.
It's the return that is reversed ?
So try with the reverse

```
"eval": "Item.val =  Attr.val",
"eval": "Item.val = ( Attr.val) < 100",
```

In strict mode

Or its when you are making the request ? And here I think I have another problem.

pyrech, 2022-02-08:
I use the action buttons from the Cluster Info panel.

[image]

Even with reverse configuration, no change in the cover direction ("Up / Open" still close the cover, "Down / Close" still open it)

Smanar, 2022-02-09:
Ok, so was the bad situation.
I m asking to other devs if it's possible using DDF, I have a doubt now.

For me DDF can only reverse the return, not the request made.
So as workaround, the only way I m seing is reversing the wire, to have the "send request" reversed, and use in same time the DDF to return the "status report"

pyrech, 2022-03-07:
(Just bumping the issue to keep it open until status reported is reversed out of the box 😉)

Smanar, 2022-03-27:
@pyrech The last deconz beta start tu support a "write" feature for tuya https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5868

Wana make a try with your device ?
If yes, can you share again the last DDF you are using ?

pyrech, 2022-04-18:
Still waiting for a fix 😁

Smanar, 2022-04-18:
Ha yes, no more news for the moment.
In WIP tuya can now support write command, but it will work only on tuya cluster, so nothing for "standard" cluster, I will ask for "fresh" news for that.

Smanar, 2022-04-20:
@pyrech I can edit the code to add a DDF field like "config/reversemode"
So will be possible to do something using DDF later, but problem for that need c++ code edition, I can make some simulation but to test the final code, I realy need someone to test code.

BTW you don't have a linux machine will full OS to test my changes ? 5I will give you the commands to use)

pyrech, 2022-04-20:
My daily work is on a linux computer so I can probably test out some things if you give all the commands.

Smanar, 2022-04-21:
Ok have started the code, and I probably have a covering module somewhere, if it don't use tuya cluster, I will be able to make test on my side.
I will make fist tests this week end I think.

Smanar, 2022-04-24:
So I have a problem, we can't use "config/reversemode" for light device.
So waiting for devs decision I have set a workaround, on the DDF just add a `"revers":true`

```
{
  "name": "state/open",
  "parse": {
    "at": "0x0008",
    "cl": "0x0102",
    "ep": 1,
    "reverse": true,
    "eval": "Item.val = Attr.val < 100",
    "fn": "zcl"
  }
}
```

To reverse ONLY the open/close request, or can use them in the "lift" JSON.
For the moment
- NOT working for on and bri.
- Act only on request, not the return, so you need to make like before in the JSON on the "eval" field.

To test the code, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone --branch covering_new https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder.

Then make a try with the actual DDF, them add a `"reverse":true`

pyrech, 2022-04-28:
I'm not at home until a few days, so I can't test it at the moment, sorry.

I will try to test it when I'm back. Thanks!

Smanar, 2022-05-06:
@pyrech Have updated the code for the "stop" request.

Smanar, 2022-05-07:
Ok the "stop" request is working too, but it seem there is strange behaviour.

> I get frequent artifacts, like the shutter moving back to the previous position after it has moved, or it stuttering when using the stop button.
> Or for example it is 90% open, and I press the "Open/Up" button (via API from HA) it opens to 100% then goes back down a bit.
> 
> Almost feels like something is always fighting my inputs.
> This is especially true after performing multiple actions in succession

If you can confirm or share some logs ? (with "info" "info_l2" and "ddf")

Smanar, 2022-06-06:
Ok still no one to test the code ?

Smanar, 2022-07-05:
I m still waiting someone to test the code ^^
I had another tester, but he have just left for z2m (on the other issue)

pyrech, 2022-07-05:
Hi, sorry for the lack of activity. I'm still interested in the fix but I didn't touch my home automation stuff for many months and I must admit I'm too afraid to break something when testing and compiling a new deconz manually :confused:

Smanar, 2022-07-06:
I have tried it with a module, a QS-zigbee-C01, no problem on my side, but I try it as stand alone mode.
Someone else have tried it with a _TZ3000_j1xl73iw but have strange result https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5710#issuecomment-1120240367

So the code is at least almost working, but I need someone that can try it with real covering and have enough time to make some tests.

closed 2022-08-05 by github-actions[bot]