dresden-elektronik/deconz-rest-plugin #5711: RaspBee firmware upgrade issue

ausfas, 2022-01-26:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

ausfas, 2022-01-26:
[image]

Mimiix, 2022-01-26:
Hi There,

This is a long issue with the addon not able to update the firmware. As we don't maintain the addon, we can't fix it.

The recommended way is to use an native OS and follow this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Closing this issue.

Kind regards,
Dennis

closed 2022-01-26 by Mimiix

ausfas, 2022-01-26:
dCONZ has listed homeassistant  on https://phoscon.de/en/raspbee, however the manual solution is what I tired and has not worked for me so I have raised this issue.

Mimiix, 2022-01-26:
If it's a user question, you should use the forum as described in #5113