dresden-elektronik/deconz-rest-plugin #5714: Technicolor XHK1-TC

tyk0s, 2022-01-27:
## Device

- Product name: Technicolor XHK1-TC
- Manufacturer: Technicolor
- Model identifier: TKA105
- Device type : Keypad

### Basic
[image]
[image]
[image]

### Identify
[image]

### Power Configuration
[image]

### Poll Control
[image]

### Temperature measurement
[image]

### IAS Zone
[image]

### Diagnostics
[image]

### OTAU
[image]

tyk0s, 2022-01-27:
updated..

Mimiix, 2022-01-27:
Are you planning to re-create it or do i need to reopen this one?

tyk0s, 2022-01-27:
> Are you planning to re-create it or do i need to reopen this one?

please reopen :)

tyk0s, 2022-01-27:
Similiar to [Xfinity Keypad XHK1-UE](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4560)
looks the same but different manufacturer if this can help somehow

tyk0s, 2022-02-20:
What more can i do? Any devs?

tyk0s, 2022-03-03:
no interest from devs, closing.

closed 2022-03-03 by tyk0s

BabaIsYou, 2023-01-23:
@tyk0s 
May be could try this DDF, adapted from the one generated by code for XHK1-UE

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Technicolor",
  "modelid": "TKA105",
  "product": "XHK1",
  "sleeper": false,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0402"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:44",
            "ep": 255,
            "eval": "const vmin = 36; const vmax = 60; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "default": 0
        }
      ]
    },
    {
      "type": "ZHAAncillaryControl",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0501"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ],
        "out": [
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:44",
            "ep": 255,
            "eval": "const vmin = 36; const vmax = 60; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/action",
          "default": "disarmed"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
        },
        {
          "name": "state/panel",
          "default": "disarmed"
        },
        {
          "name": "state/seconds_remaining",
           "default": 0
        },
        {
          "name": "state/tampered",
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0000"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    }
  ]
}
```

tyk0s, 2023-01-23:
Cool, but not using conbee stick anymore.

PR #6695 (2023-04-14): DDF for Technicolor XHK1-TC Keypad
Fix #5714 
BTW add DDF for legacy supported XHK1-UE Xfinity keypad