dresden-elektronik/deconz-rest-plugin #5716: Device name

Myth5656, 2022-01-27:
## Device

- Product name: SMaBiT AV2010/29A
- Manufacturer: SMaBiT
- Model identifier: AV2010/29A
- Device type : 
  - Other: Waning Device (siren)

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### IAS Zone
[image]
### IAS WD
[image]
###OTAU
[image]

Myth5656, 2022-01-27:
Sending "select" and "lselect" at the moment via openhab turns on the siren for a second. I can hear a short beep for both commands.

Myth5656, 2022-01-28:
The blink command turns on the siren and light.
So it's just a software / parameter problem.

SwoopX, 2022-01-29:
Yes, it's a parameter problem. You need to find out the right combination (or the one you're after) in the IAS WD cluster.  So play with the warning options. I would assume strobe is supported and I'd recommend to stick to warning mode 0 for the left play with siren level. Do not forget to set the warning duration.

Myth5656, 2022-01-29:
@SwoopX do you or does anybody else have an example how to pass multiple values like warning mode and duration in one command from openhab?
I would post possible combinations when I found a working one. At the moment I just send "blink", siren then is on for about 5sec with blinking lights.
Preset values like select and lselect are not working.

SwoopX, 2022-01-30:
This is neither possible via openhab nor any other home automation software. That's what deconz GUI is for.

Myth5656, 2022-01-30:
@SwoopX for my understanding: I need to find out which parameters are necessary to turn on the alarm? And software changes are done afterwards?

Myth5656, 2022-01-30:
> This is neither possible via openhab nor any other home automation software. That's what deconz GUI is for.

Here I have some information about the parameters, tested with deCONZ:

siren Level 0 = Siren sound on "typical burglary sound", constantly running
siren Level 1 = Siren sound on "typical burglary sound", with short breaks
warning section strobe = 5 long beeps, lights always on
warning mode 0+1 = lights on, flashing. (only in combination with siren level 0+1)
warning mode 2+3: not available
warning duration = on in seconds (only with siren level 0+1)

squawk 0 = 5 short beeps, (kind of acknowledge sound), no light
squawk 1 = not available
strobe squawk section = not available
sound for disarmed = 5 long beeps, lights alway on (same like strobe in warning section)
Values "strobe level" and "strobe duty cycle" have no effect.

At the moment only the blink  command triggers a function which is probably "warning section strobe".
Its missing to activate "a siren level with warning mode". It would make sense to have "siren level 0 with warning mode 0" available for two different duration (select and lselect).

Myth5656, 2022-02-04:
@SwoopX I provided details for behavior with the parameters, but I saw that device merges are stopped.
Is it possible to solve my issue with DDF? Or how could it be solved?

SwoopX, 2022-02-05:
> Is it possible to solve my issue with DDF?

No, this cannot be handled via DDF but must be hardcoded.

The current implementation for other Smabit devices is burglar (warning mode 0 checked), No strobe (strobe unchecked), very high sound (siren level 1+2 checked) for both `select` and `lselect`. It's interesting to see that the different siren levels apparently are reposible for different sounding schemes in your case. Also, using strobe only would probably not exactly be zigbee compliant.

However, the specs do not dictate which is the LSB/MSB, rarely leading to vendors implementing it differently. So far, only one device is known by us doing it differently. That would presumably result in the following: Warning 2+3 are responsible for loudness, warning 0 for strobe and siren 0 would be the warning level burglar (siren 1 = fire - that could explain the shounding schemes).

There's no way I can help you fiddling out the "right" combination other than the above, I'm afraid.

Myth5656, 2022-02-06:
@SwoopX
In your implementation `select` and `lselect` would work, the problem is the `duration`
The parameter needs to be set to a higher value.
With your implementation the siren goes on and off directly.
So this is probably the **bug** here.

SwoopX, 2022-02-10:
You probably aren't aware that for `lselect`, you can pass `duration` as a 2nd parameter to make to sound for the defined time?

Myth5656, 2022-02-11:
yes you are right, I am not aware of this. In my logs I see that /state/alert property is called
`"address":"/lights/5/state/alert" is called. `
I tried diffent json formats, but always get an error returned from DeconzBaseThingHandler

`Sending command {"state":lselect} to channel deconz:warningdevice:00212E077C9D:00124b001ff57e5301:alert failed: 400 - [{"error":{"address":"/lights/5/state/alert","description":"invalid value, {&quot;state&quot;:lselect}, for parameter, alert","type":7}}]`
Should the request be send to /state ?

SwoopX, 2022-02-13:
The request must go to `state/`, something like

```
{
    "alert": "lselect",
    "duration": 10
}
```

Myth5656, 2022-03-18:
@SwoopX 
Sorry for my late response: I just tried the request you mentioned:

`{
    "alert": "lselect",
    "ontime": 10
}`
Its still the same, the siren just beeps very short. Ontime value can be higher, it doesn´t  make a difference.
Can you reOpen the ticket?

Myth5656, 2022-03-18:
Hi all, please re-Open the issue. I provided more Information, the proposed solution doesn´t work. CheersAndy  Gesendet von Mail für Windows Von: github-actions[bot]Gesendet: Freitag, 18. März 2022 03:00An: dresden-elektronik/deconz-rest-pluginCc: Myth5656; MentionBetreff: Re: [dresden-elektronik/deconz-rest-plugin] SMaBiT AV2010/29A outdoor sirene (new model) (Issue #5716) As there has not been any response in 28 days, this issue will be closed. @ OP: If this issue is solved post what fixed it for you. If it is not solved, request to get this opened again.

	
		
			
				
			
		
		
			
				Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
				www.avast.com

Mimiix, 2022-03-18:
Sure.

Myth5656, 2022-04-09:
@SwoopX

Are there any news? The rest rest call did not work either, please check my last post. Best regards
Andy

closed 2022-05-09 by github-actions[bot]