dresden-elektronik/deconz-rest-plugin #5725: Aqara E1 Water Leak Sensor SJCGQ13LM

cwm77, 2022-01-29:
## Device

- Product name: E1 Water Leak Sensor
- Manufacturer: Aqara
- Model identifier: SJCGQ13LM
- Device type :
  - Sensor

## Screenshots

### Basic

[image]
[image]
[image]

### Identify
[image]

### Alarms
N/A

### Device Temperature
N/A

### Groups
N/A

### Scenes
N/A

### On/Off
N/A

### Level Control
N/A

### Color Control
N/A

### Simple Metering
N/A

### Diagnostics
N/A

## Other clusters that are not mentioned above
### Power Configuration
[image]

### IAS Zone
[image]
[image]

### Lumi specific
[image]
[image]
[image]
[image]

### Identify
[image]

### OTAU
[image]
[image]

Mimiix, 2022-01-29:
Manufacturer name and model Id in the template are wrong. Please adjust it so it matches the basic cluster.

cwm77, 2022-01-29:
> Manufacturer name and model Id in the template are wrong. Please adjust it so it matches the basic cluster.

Done

SwoopX, 2022-01-29:
All screenshots are only showing the default values, so I'm afraid this is not helpful. Please exchange the screenshots with some where all the attributes have been read.

cwm77, 2022-01-29:
> All screenshots are only showing the default values, so I'm afraid this is not helpful. Please exchange the screenshots with some where all the attributes have been read.

Ok, I updated the screenshots. The clue was to push the button of the sensor while reading. Not every screenshot got changed, but I checked everyone. Please let me know if something is missing.

cwm77, 2022-01-29:
I wanted to ask if it is already possible to foresee if the integration of this sensor into the software will take a lot of effort and time? According to zigbee.blakadder.com, the two previous models are already supported by Deconz.

https://zigbee.blakadder.com/Aqara_SJCGQ11LM.html
https://zigbee.blakadder.com/Aqara_SJCGQ12LM.html

For this sensor, here is the link, by the way:

https://zigbee.blakadder.com/Aqara_SJCGQ13LM.html

SwoopX, 2022-01-29:
> ... 100% Zigbee 3.0 compatible
That's the best joke I read since a long time (I'm afraid)

No, the links unfortunately do no help.

Mimiix, 2022-01-29:
@SwoopX what's missing?

SwoopX, 2022-01-30:
@Mimiix Read attributes

cwm77, 2022-01-30:
I would also be interested to know exactly which attributes are the problem now?  I am not a dev, but if I can help in any way, for example testing, please let me know. Hope the laugh I generated didn't knock you off your chair ;) .

cwm77, 2022-01-30:
> @Mimiix Read attributes

I would also be interested to know exactly which attributes are the problem now? I am not a dev, but if I can help in any way, for example testing, please let me know. Hope the laugh I generated didn't knock you off your chair ;-).

Mimiix, 2022-01-31:
@SwoopX Which ones?

Smanar, 2022-01-31:
>What exactly is wrong with it now?

IDK, waiting for SwoopX answer.
But not possible using code for next devices, can use only DDF, so this device will be not included in deconz like the previous one.
And honnestly I have never used the lumi cluster in DDF, so not sure if this device can be added as it ?

cwm77, 2022-02-07:
> 

Has this something to do with this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733 ?

Mimiix, 2022-02-07:
Yes.

cwm77, 2022-02-11:
> As already mentioned twice, the attributes from all but 2 clusters have NOT been read.

After Update deCONZ to v2.14.1 read Attributes again. Something changed, HAppy now?

4val0v, 2024-03-16:
Is there currently support for this sensor ?

SwoopX, 2024-03-20:
No, it isn't. However, the subsequently added screenshots appear to be complete now and drafting should be doable. Anybody able to test an initial DDF once ready?

PR #7676 (2024-03-24): DDF for Aqara E1 Water Leak Sensor SJCGQ13LM
Should be fairly straight forward. Closes #5725
closed 2024-03-24 by manup