dresden-elektronik/deconz-rest-plugin #5729: Can’t find aqara mini switch(lumi.remote.b1acn01) in Phoscon

monpelaud, 2022-01-30:
Hi,
This issue is described here:
[https://community.home-assistant.io/t/cant-find-aqara-mini-switch-lumi-remote-b1acn01-in-phoscon/372987](url)
Best regards
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2022-01-30:
>ConBee II stick v.2.13.04

Hello, I don't see something new for this device, and It don't use DDF yet ...
Have you tried to use the switch just after the pairing procedure, every 2/3s, to prevent the device going to sleep ?

If you have the GUI, can you try that https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues ?

Mimiix, 2022-01-30:
Fixed in 2.14.0.

Please use the beta for this.

closed 2022-01-30 by Mimiix