dresden-elektronik/deconz-rest-plugin #5731: LoraTap Tuya ZigBee 3.0 Wireless 2 Gang Remote Control Switch

sinus61, 2022-01-30:
## Device

- Product name: LoraTap Tuya ZigBee 3.0 Wireless 2 Gang Remote Control Switch 
- Manufacturer: _TZ3000_t8hzpgnd
- Model identifier: TS0042
- Device type : 

  - Remote
https://www.loratap.com/products/ss6200zb

When pairing it creates two lamps and one sensor. But no buttonevent comes through.
I think there are already some TS0042 around, so maybe this has just added similar.

## Screenshots

[image]
[image]

### Basic
[image]

### Power Configuration 1
[image]

### On/Off 1
[image]
[image]

### Power Configuration 2
[image]
[image]

### On/Off 2
[image]

[image]

sanluca, 2022-01-30:
I am interested, I too have these two button remote controls but also four button ones. it only recognizes two-button remote controls but as two lights, I am waiting for news on how to configure them
thank you

sinus61, 2022-01-31:
I have also the 4 gang:

Product name: LoraTap Tuya ZigBee 3.0 Wireless 4 Gang Remote Control Switch

Manufacturer: _TZ3000_ufhtxr59

Model identifier: TS0044

Device type :

    Remote
    https://www.loratap.com/products/ss6400zb

You get 4 lamps if you pair and a sensor, but no buttonevent.

sanluca, 2022-01-31:
mine is not seen

Smanar, 2022-02-01:
Need to check the manufacture name, all device can be different even visualy same.

"_TZ3000_t8hzpgnd" is not supported at all, and as PR are locked, need to wait for a battery switch support with DDF.

janjurek, 2022-02-02:
Hi there, this might be probably redundant information, but as to confirm this one, the Manufacturer is little different for me as well (_TZ3000_**eee8nrt2l**) but I too have a 4 button one with model identifier: TS0044.

It's correctly read inside the deCONZ Raspbee UI however I'm unable to make it usable inside the Phoscon API (thus not possible to use). I don't even get any lamps it's just ignored for me in API. Hope the PRs are unlocked soon, this switch is quite affordable and portable, it would be great addition to the network for me :)

[image]

sanluca, 2022-02-10:
but who does it depend on that  PR are free?

Smanar, 2022-02-11:
I m not sure to understand ?

sanluca, 2022-02-21:
I did not understand what it takes to make them work, what are PRs? thanks

Smanar, 2022-02-21:
Ha, PR are Pull Request, and all are blocked for the moment https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733

and for the moment the core don't support battery switch.

hoodlumj3, 2022-03-15:
Mr stale bot, this is not stale, this is dependant on other dev factors. TY

janjurek, 2022-04-06:
Not stale still waiting for something to happen though. TY

janjurek, 2022-04-30:
Not stale still waiting for development dependecy.

vmercep, 2022-05-15:
any updates from this one? I have same problem, "_TZ3000_t8hzpgnd" only two lamps and switch with no buttons.

vmercep, 2022-05-16:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_t8hzpgnd",
  "modelid": "TS0042",
  "product": "TS0042",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/ts0042.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}

Smanar, 2022-05-17:
Perfect, I have edited the DDF a little (remoive path and change status to gold) but it seem complete

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_t8hzpgnd",
  "modelid": "TS0042",
  "product": "TS0042",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

On deconz go to help / debug view, select "info" and "info_l2" as falg, I think you will see error message about a missing table when you press a button.
Can you share the error message ?
If we are lucky your DDF is complete and just need to edit another json file.

vmercep, 2022-05-22:
Hi,

OK now I have restarted and see that I have only switch which I cant add to light control. Another problem is that I am using deconz and docker image , Here is log

4:52:24:043 Websocket 172.18.0.1:41338 send message: {"attr":{"id":"11","lastannounced":null,"lastseen":"2022-05-19T14:52Z","manufacturername":"_TZ3000_npzfdcof","modelid":"TS0001","name":"Karla room","swversion":null,"type":"On/Off light","uniqueid":"a4:c1:38:4c:e4:9e:0f:76-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"a4:c1:38:4c:e4:9e:0f:76-01"} (ret = 317)
14:52:24:046 Websocket 192.168.5.202:61018 send message: {"attr":{"id":"11","lastannounced":null,"lastseen":"2022-05-19T14:52Z","manufacturername":"_TZ3000_npzfdcof","modelid":"TS0001","name":"Karla room","swversion":null,"type":"On/Off light","uniqueid":"a4:c1:38:4c:e4:9e:0f:76-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"a4:c1:38:4c:e4:9e:0f:76-01"} (ret = 317)
14:52:24:710 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xC76B
14:52:24:712 neigbor 0xbc33acfffe18ad4d is unknown child
14:52:24:727 Websocket 172.18.0.1:41338 send message: {"attr":{"id":"4","lastannounced":"2022-03-28T08:39:25Z","lastseen":"2022-05-19T14:52Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW 806lm","name":"Dnevna ambijentalna","swversion":"2.1.022","type":"Dimmable light","uniqueid":"60:a4:23:ff:fe:e4:20:38-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"60:a4:23:ff:fe:e4:20:38-01"} (ret = 366)
14:52:24:730 Websocket 192.168.5.202:61018 send message: {"attr":{"id":"4","lastannounced":"2022-03-28T08:39:25Z","lastseen":"2022-05-19T14:52Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW 806lm","name":"Dnevna ambijentalna","swversion":"2.1.022","type":"Dimmable light","uniqueid":"60:a4:23:ff:fe:e4:20:38-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"60:a4:23:ff:fe:e4:20:38-01"} (ret = 366)
14:52:24:858 poll node 58:8e:81:ff:fe:61:aa:70-01
14:52:24:859 Poll light node livingroom ambient
14:52:24:911 Poll APS request to 0x588E81FFFE61AA70 cluster: 0x0006 dropped, values are fresh enough
14:52:25:287 CTRL skip creating node 0xBC33ACFFFE18AD4D while permit join is disabled
14:52:25:836 [INFO] - No button map for: TS0042, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00, zclSeq: 63
14:52:26:003 poll node 00:15:8d:00:06:37:17:d2-01-000c
14:52:26:005 Poll ZHAConsumption sensor node Consumption 20
14:52:26:266 Poll APS request to 0x00158D00063717D2 cluster: 0x0702 dropped, cluster doesn't exist
14:52:26:478 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:52:26:981 poll node 00:15:8d:00:06:37:17:d2-01-0b04
14:52:26:982 Poll ZHAPower sensor node Power 21
14:52:27:248 read attributes of 0x00158D00063717D2 cluster: 0x0B04: [ 
14:52:27:249 0x0505 
14:52:27:250 0x0508 
14:52:27:251 0x050B 
14:52:27:251 ]
14:52:27:252 add task 2378 type 19 to 0x00158D00063717D2 cluster 0x0B04 req.id 247
14:52:27:253 Poll APS request 247 to 0x00158D00063717D2 cluster: 0x0B04
14:52:27:294 Poll APS confirm 247 status: 0x00
14:52:27:296 Erase task req-id: 247, type: 19 zcl seqno: 58 send time 0, profileId: 0x0104, clusterId: 0x0B04
14:52:27:435 Node data 0x00158d00063717d2 profileId: 0x0104, clusterId: 0x0B04
14:52:27:439 0x00158D00063717D2: update ZCL value 0x01/0x0B04/0x0505 after 0 s

I can see that no button map for this device.

Smanar, 2022-05-22:
Yep 
>14:52:25:836 [INFO] - No button map for: TS0042, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00, zclSeq: 63

Hoping it 's the good device, you don't have 10 TS0042 used in same time ?

So you can edit the "button_maps.json" file, it is in the "devices" folder too.

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0"],
            "map": [
```

Just add yout device in the list "_TZ3000_t8hzpgnd"

vmercep, 2022-05-22:
OK, I have created buttonsmap in shared folder so I see switch now.... new problem
[image]
13:57:55:726 [INFO] - Button 1002 - TS0042, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 102
13:57:55:735 Websocket 172.18.0.1:41392 send message: {"e":"changed","id":"42","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-05-22T13:57:55.727"},"t":"event","uniqueid":"54:0f:57:ff:fe:30:35:c3-01-0006"} (ret = 165)
13:57:55:737 Websocket 192.168.5.202:61901 send message: {"e":"changed","id":"42","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2022-05-22T13:57:55.727"},"t":"event","uniqueid":"54:0f:57:ff:fe:30:35:c3-01-0006"} (ret = 165)
13:57:55:902 poll node 00:15:8d:00:06:37:17:d2-01-0b04

but nothing happens when I press button no 2
also if I assign this switch in switch editor to the light inside phoscon , nothing happens if I press button.

vmercep, 2022-05-22:
sorry, forget about it, on second restart it works....

thanks for your help

Smanar, 2022-05-22:
Have made the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6068
Tell me if you have issue.

closed 2022-05-23 by manup

sinus61, 2022-06-30:
The added DDF with PR is working, but missing battery events.
This one works for me:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_t8hzpgnd",
  "modelid": "TS0042",
  "product": "TS0042",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/_TZ3000_t8hzpgnd_2gang_bat_switch.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 4000,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}

Mimiix, 2022-06-30:
His ddf is and the other post, this topic is about another one.

antocpu, 2025-02-08:
Hello,

I don't know if my request is in the right place, so I apologize in advance if it isn't.

I bought the Loratap 2-button zigbee remote control model SS6200ZB, which seems to be compatible with Deconz (which this thread and the following page suggest: https://zigbee.blakadder.com/LoraTap_SS6200ZB.html).

However, I can't seem to add it to my installation. I'm using the headless version and the Phoscon web application.

I put the remote into pairing mode by holding down a button for 10 seconds (the LED flashes, as you can see on the video below) and I put the Conbee into pairing mode by pressing the button in the app, but after 3 minutes, the remote still hasn't been added.

https://github.com/user-attachments/assets/fd8f801c-ec51-4ab4-9112-6e335b2532fd

[image]

I'm using the latest Deconz beta version (this one: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.1-beta).

Could someone here give me an indication of how to pair this remote?

Thanks in advance,

Smanar, 2025-02-08:
Hello, it's tuya, you probably have a new clone, you need to use the GUI to know the model id and manufacture name, probably to add them to the DDF.

antocpu, 2025-02-08:
Thanks for your quick answer,

My model does not have the "pair button", so according to the manual, I have to press any button for 10 seconds until the yellow indicator light flashes (this seems to work) :

[image]

I'm connected to my raspberry pi running deconz through a VNC session.

What should I do now in the Deconz software ?

sanluca, 2025-02-08:
> Thanks for your quick answer,
>
> My model does not have the "pair button", so according to the manual, I have to press any button for 10 seconds until the yellow indicator light flashes (this seems to [...]

Hi,

I have two of these buttons but with the button in the center for pairing, aesthetically they are the same, anyway deconz sees them as a switch, they work fine the only drawback is that the battery status doesn't work

antocpu, 2025-02-08:
Ok, mine is the version without the pairing button.

Like @Smanar suggested, it seems to be a variation, thus the need for modifying the DDF, but I kind of struggle with this (I'm able to connect to my Pi through VNC, but the Deconz gui software won't launch...)

sanluca, 2025-02-08:
Just to be clear, to see the third button you need to disassemble the remote control

antocpu, 2025-02-08:
Thanks for the indication. I wasn't fully clear. I hence managed to find this pairing button, but when I hold it for long (say 10 sec) nothing happens :

[image]

sanluca, 2025-02-08:
Try flashing it again with the same button, but search using switches instead of sensors like you were doing before

antocpu, 2025-02-08:
I Tried using the switches menu but the device is no more recognised...

antocpu, 2025-02-08:
> Hello, it's tuya, you probably have a new clone, you need to use the GUI to know the model id and manufacture name, probably to add them to the DDF. I don't think the information is visible on logs, so if you have an headless version, you need to use X-fowarding or install deconz with a desktop environnement for test.
> 
> You haven't the model with the "pair button" ? (I can't read the video)
> 
> > To pair the device, open it with a screwdriver and press the pairing button for 5 seconds until the yellow indicator light flashes. For models without a pairing button, press and hold any of the buttons for 10 seconds until the yellow indicator light flashes.

hello @Smanar .

I finally managed to connect to Deconz (I needed to stop the service first (as mentionned [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1963#issuecomment-546040234)).

And I think I identified the device.

Here is a screenshot of the information provided in Deconz :

[image]

Can I do something with these to make it work ?

Thanks

Smanar, 2025-02-08:
Ok, so it's that the _TZ3000_fkvaniuu is new.
Now the next will depend of your OS ?

To make it short we will need to add "_TZ3000_fkvaniuu" to this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0042_2gang_remote.json
This part is easy but we need to add it too to button_maps.json and this will be not possible if you are using docker.

antocpu, 2025-02-08:
Here are my OS info :

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

Smanar, 2025-02-09:
Ok so will be easy.
You know how to use text editor ?

Close deconz.
Edit the file `/usr/share/deCONZ/devices/tuya/_TZ3000_TS0042_2gang_remote.json` to have (2 lines to add)
```
  "manufacturername": [
    "_TZ3000_5e235jpa",
    "_TZ3000_t8hzpgnd",
    "_TZ3000_tzvbimpq",
    "_TZ3000_i3rjdrwu",
    "_TZ3000_fkvaniuu"
  ],
  "modelid": [
    "TS0042",
    "TS0042",
    "TS0042",
    "TS0042",
    "TS0042"
  ],
```

Then edit the file /usr/share/deCONZ/devices/button_maps.json (1 line to edit) to have

```
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv", "_TZ3000_wbfgbpxq", "_TZ3000_fa9mlvja", "_TZ3000_tk3s5tyg", "_TZ3000_mrpevh8p", "_TZ3000_qgwcxxws", "_TZ3000_yj6k7vfo", "_TZ3000_axpdxqgu", "_TZ3000_a4xycprs", "_TZ3000_u3nv1jwk", "_TZ3000_fkvaniuu"],
            "map": [

```

Then relaod Deconz

Smanar, 2025-02-09:
For battery powered switches the API will return a "state/buttonevent" value on the form X00Y, X is the pressed button and Y the action.
So third app can display only 1 "widget" with all values.

Do you want to make a PR or want I make it ? to submit the device officialy ?

antocpu, 2025-02-09:
I did not know that, so it's now clearer for me.

If you do not matter, as I don't know how to do it, can you handle the PR ?

many thanks,

Smanar, 2025-02-09:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8117