dresden-elektronik/deconz-rest-plugin #5732: Aqara Cube Controller T1 Pro

ads1230, 2022-01-30:
## Device
- Product name: Aqara Cube Controller T1 Pro
- Manufacturer: LUMI
- Model identifier: lumi.remote.cagl02 / MFCZQ12LM
- Device type : Magic Cube

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Power configuration
[image]

### 01x0012 Multistate Input (Basic)
[image]

### On/Off
[image]

### Lumi Specific
[image]
[image]

### 02x0012 Multistate Input (Basic)
[image]

### 03x0012 Multistate Input (Basic)
[image]

ads1230, 2022-02-21:
Apologies for bumping the issue, don’t want it to close.

B1n00, 2022-11-25:
I observe the same issue with the device

mkcologne, 2022-11-29:
Put me on the List!

SwoopX, 2022-11-30:
This PR probably already holds everything for the device https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5838

Testing it would be ideal, which hopefully wouldn't involve any HA and mainly requires placement of the files.

B1n00, 2022-12-01:
> This PR probably already holds everything for the device #5838
> 
> Testing it would be ideal, which hopefully wouldn't involve any HA and mainly requires placement of the files.

It seems the Model specific to the PRO version is not include in the code: lumi.remote.cagl02 / MFCZQ12LM
Could you update so I can test?

fhinder, 2022-12-04:
I'm interested as well. 
The old "aqara magic cube" is working, the new one "aqara cube T1Pro"  is not recognised .

Doener92, 2022-12-05:
Hallo Zusammen, ich habe auch den neue Aqara Cube T1 Pro. Dieser wird leider auch nicht erkannt bzw. ist nicht nutzbar. Hilfe wäre nett.

Doener92, 2022-12-08:
Is there any update. Can someone help with the problem?

SwoopX, 2022-12-08:
As said before, the files from the mentioned PR are expected to work if the cube DDF is extended by the T1 Pro manufacturer and modelID. This should be tested please.

fhinder, 2022-12-12:
> Is there any update. Can someone help with the problem?

Pleas check #6608, if this is a solution for you. My setup is working.

closed 2023-01-10 by github-actions[bot]

stefanmechelen, 2023-01-26:
I don't know much about github, but I do understand it's possible to add the Cube T1 pro, I updated deconz (add on in HA) to 2.20.01). I can add my old first gen magic cube but I still can't add the T1 pro. Is that normal since this says it's solved. I'm confused about that.