dresden-elektronik/deconz-rest-plugin #5737: /groups/…/action/scene always returns "0"

2xlink, 2022-02-01:
## Describe the bug

Hi all. I think I discovered a bug with the /groups/…/action/scene condition. It seems to always return "0".

## Steps to reproduce the behavior

I added the following rule:

```json
{
    "actions": [
        {
            "address": "/groups/2/scenes/1/recall",
            "body": {},
            "method": "PUT"
        }
    ],
    "conditions": [
        {
            "address": "/sensors/68:0a:e2:ff:fe:12:34:56-01-1000/state/buttonevent",
            "operator": "eq",
            "value": "1002"
        },
        {
            "address": "/sensors/68:0a:e2:ff:fe:12:34:56-01-1000/state/lastupdated",
            "operator": "dx"
        },
        {
            "address": "/groups/2/action/scene",
            "operator": "eq",
            "value": "3"
        }
    ],
    "name": "Rule CALL_SCENE 1 WHEN OTHER",
    "periodic": 0,
    "status": "enabled"
}
```

The 3rd condition should check for the current scene of the group 2. When I call http://192.168.178.100:8080/api/<redacted>/groups/2/, I can see that it is set to "3":

```json
{
    "action": {
        "alert": "none",
        "bri": 128,
        "colormode": "hs",
        "ct": 0,
        "effect": "none",
        "hue": 0,
        "on": true,
        "sat": 128,
        "scene": "3",
        "xy": [
            0,
            0
        ]
    },
    "devicemembership": [],
    "etag": "<redacted>",
    "id": "2",
    "lights": [
        "3",
        "4"
    ],
    "name": "lights",
    "scenes": [
        {
            "id": "1",
            "lightcount": 2,
            "name": "All off",
            "transitiontime": 10
        },
        {
            "id": "2",
            "lightcount": 2,
            "name": "light1",
            "transitiontime": 10
        },
        {
            "id": "3",
            "lightcount": 2,
            "name": "light1 + light2",
            "transitiontime": 10
        }
    ],
    "state": {
        "all_on": true,
        "any_on": true
    },
    "type": "LightGroup"
}
```

## Expected behavior

The conditions should all be `true` and the action should be triggered. After some debugging, I found that the content of the variable always seems to be "0".

## Screenshots

--

## Environment

 - Host system: PC
 - Running method: Docker: deconzcommunity/deconz
 - Firmware version: 26580700
 - deCONZ version: 2.14.00
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Yes, several HDDs

## deCONZ Logs

Can supply if needed.

## Additional context

ebaauw, 2022-02-01:
You cannot include `action` attributes in rule conditions.  Instead, use a CLIPGenericStatus sensor to keep track of the current scene.

2xlink, 2022-02-01:
Thanks @ebaauw, with this I could successfully configure my setup. Do you maybe know whether such a functionality could be implemented in the future? I think that it would be very useful for many people :)
I would do it myself, but unfortunately I am pretty bad at C++ 😅

ebaauw, 2022-02-01:
The group `action` attributes are basically write-only (they just echo the last value set, and don't reflect the actual nor aggregated state of any light(s) in the group).  And they're not (yet?) linked to the event system.  In theory, that last point could be addressed, but in practice that would be quite some work, for something that's probably not too useful.  Having a proper `state` for groups (with more than just `all_on` and `any_on`) would be useful, I suppose, but that is quite challenging, especially if the group contains lights that don't support attribute reporting.

Anyway, I have a long list of higher priorities changes that I would want to make before this one.  Amongst others some proper error reporting by the API, alerting you that a rule condition on an `action` attribute won't do anything.

Mimiix, 2022-02-03:
As it seems this is not a bug, i'm closing this request.

Feature requests are (as mentioned in #5113 ) logged on the forums.

Kind regards,

closed 2022-02-03 by Mimiix