dresden-elektronik/deconz-rest-plugin #5738: THS317-ET by OWON

Stim04, 2022-02-01:
Hello,

I am trying to make this detectable in Jeedom with the deconz plug in but the best I could do is see it in deCONZ GUI for windows. I've been told on the discord (thx MiMix) that it's not supported yet so here I am. This is the first time I do that sorry if I forget something.

I've seen this equipment from different manufacturer (CASA.ia) has been treated on GitHub and i was hoping it would be detectable but i guess not.

CASA.ia version
https://zigbee.blakadder.com/CASA_IA_CTHS-317-ET.html
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5259

OWON version
https://zigbee.blakadder.com/OWON_THS317-ET.html

## Device

- Product name: Temperature Probe (only MAC address shown on deCONZ GUI)
- Manufacturer: OWON (unknown on Basic cluster)
- Model identifier: THS317-ET
- Device type :  
  - Sensor

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Do not hesitate to tell me if I forgot something

Thank you in advance

Mimiix, 2022-02-01:
@Smanar Would this be able with DDF?

Smanar, 2022-02-01:
Sure, will be fast and easy.
@Stim04 do you know what is DDF ?
Here you have some information https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet, and a video here https://www.youtube.com/watch?v=rgTQpOVg8w0

The procedure will depend of how is prefilled your DDF.

- Enable DDF.
- Make edit "edit DDF"
- If needed fill the "basic cluster identification", need the "model ID" and the "manufacture name"
- Set the status as "gold" (for tests)
- for sleeper, let "false" for the moment, but it s' probably true.
- If needed just drag and drop a ZHAtemperature from the right panel to the left.
- Save the DDF
- Make "hot reload" on the menu

The name need to be "temperature X" instead of "0xXXXX".

The value will be updated when you will read them in deconz (Need probably to wake up the device), if the value is updated automaticaly you have finished. Else it will work only 1 time, mean you need to configure the bind and the report.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel

This part is the more complicated, yu need first to drag & drop the cluster 0x0402 on the left panel, then the attribute 0x000 on the right one.
with luck values will be filled with standard value.

Then make again "hot reload".

You will probably miss too the battery after that.

Stim04, 2022-02-01:
I leave you a copy of the DDF I made since I am clueless about how good I did
[ths317-et.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7979862/ths317-et.zip)
.

Stim04, 2022-02-01:
Oh well.. puting Bronze to Gold just fixed it I think
[image]
Now I'm going to try and pair it on the debian version of deconz after puting my conbee back on my jeedom.. i copied the json here, is that fine ?
[image]

Smanar, 2022-02-01:
No, I don't think, it s more like a jeedom folder than a deconz one.
Take a look if you don't have a folder "/usr/share/deCONZ/devices/" or better "/home/deconzuser/.local/share/dresden-elektronik/deCONZ/devices"

Nothing is missing ? the battery is good ?

Stim04, 2022-02-01:
Well voltage (31) and percentage (200) has always been reporting in the cluster. Gonna look for another path in jeedom

Smanar, 2022-02-01:
>Well voltage (31) and percentage (200) has always been reporting in the cluster

You have set it yourself o was pre-filled ?

>Its true that it does not show the battery icon on the GUI

Can take time to appear.
You can check the device json in jeedom "information brute" (but can need time for somes values)

Here you have DDF included in deconz https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices
Can search thoses files on your machine. But this folder is reserved to deconz, can use it for test, but next time you will update deconz this folder will be reset.

Stim04, 2022-02-01:
> You have set it yourself o was pre-filled ?
It was zero before I read it, i did not fill anything

> Here you have DDF included in deconz https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices
> Can search thoses files on your machine. But this folder is reserved to deconz, can use it for test, but next time you will update deconz this folder will be reset.

Thanks I'm on it to try and find the right device folder with all that json stuff with the SSH console

Stim04, 2022-02-01:
Think I found it. All thanks to you and your list of devices
[image]

Smanar, 2022-02-01:
>It was zero before I read it, i did not fill anything. Do i need to change anything for the Max interval in the DDF? it's actually 0

Ha yes right, haven't see the value was 0
Defaut values for 0x0020 and 0x0021

```
            rq.minInterval = 300;
            rq.maxInterval = 3000;
            rq.reportableChange8bit = 1;
```

Smanar, 2022-02-01:
Lol, if it work i will keep the link to be used as tuto ^^

Smanar, 2022-02-01:
>Well temperature stay at zero but there might but a minor tweak to do on the DDF in the bindings with min and max interval like on power

Ha yes, on your DDF
```
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 0,
          "max": 0
        }
      ]
```

Will search basic value to use

Stim04, 2022-02-01:
So far in my tests it stays at 0. But I know we're closed to make this work.

Smanar, 2022-02-01:
For the lastseen ca be this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5675

Perhaps  it have DC ? you can try to re-include it if you still have no lastchange or lastseen after 1 hour.

Can you show the last DDF ?

Edit:
`  "sleeper": true,`

You have tried with this field as false ?

Stim04, 2022-02-01:
Here is last version i'm using.
[ths317-et.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7981544/ths317-et.zip)

Going to try with sleeper false. I already did alot of reset and re-pair with different values for the temp cluster max min interval.

Smanar, 2022-02-01:
Can you try with this one https://pastebin.com/UBkbsPhy

Stim04, 2022-02-01:
> Can you try with this one https://pastebin.com/UBkbsPhy

Let me make a coffee and a prayer before I do. The first in case the second fails

Smanar, 2022-02-01:
I  m making comparaison with a working DDf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-986296532

So you was right for
`"sleeper": true,`

and they have "forced" more thing than you

```
        {
          "name": "state/temperature",
          "awake": true,
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
```

If you use the editor, you will have thoses item when selecting state/temperature, else can just edit the DDF

Stim04, 2022-02-01:
getting closer but still reportinng 0 for temperature with your last additions
[ths317-et.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7981763/ths317-et.zip)

Smanar, 2022-02-01:
Arf sorry my bad

```
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
```

You have ep = 1, this device use 3 as endpoint.

When you read the attribute 0x0000  in the cluster 0x0402 you will  read the actual temperature AND it update the value in the json in same time, else there is a problem before.

Stim04, 2022-02-01:
Damn, getting little confused here as i started from scratch. Gonna try to figure that out.

Stim04, 2022-02-01:
Omg that last Json worked ! SW version is read right away but it takes a few minutes for the rest to come up.
@Smanar Thanks for your patience and help 🥇 
[image]
[image]

Here is the Json that saved my day 👍 
[ths317-et.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7982630/ths317-et.zip)

Mimiix, 2022-02-02:
@Stim04 can you please:
- Tell us how it works after a few days
- Submit a PR of this DDF if it works correctly.

Stim04, 2022-02-02:
@Mimiix Sure. I will  tell you how it goes after a few days. Not sure what is a PR and how do I do that.
For now it is working pretty well. It did not fail during the night.

Smanar, 2022-02-02:
Ha finallly ^^, was harder than I thought 

>We might have to put higher values for temperature request intervals as @Smanar suggested as it would extend battery life

Exactly, it's a personal choicel, but I prefer a long life battery than a report every minuts for a temperature.

Stim04, 2022-02-02:
> was harder than I thought

I probably messed up by not waiting long enough when i first used the json that detected SW version, or it was after we messed up the temperature state ep, can't remember for sure.

> Exactly, it's a personal choice, but I prefer a long life battery than a report every minute for a temperature
Probably a wiser choice for everyone including the planet.

Smanar, 2022-02-07:
> this was entirely a Jeedom generated issue.

I think more on deconz side ^^
It's a know issue, ATM as workarounf can use the good endpoint insteaod of "auto", and use "gold" status, and you are using all of them.
There is a new stable since yesterday, the v2.14.1 
And this one include https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5757

Stim04, 2022-02-07:
Thanks for the info. You're probably right but it started messing up right when I downgraded deconz to an anterior version on jeedom. I since found out how to fix everything and my Jeedom and deconz is now up to date. I will try to pass from 14.00 to 14.1 but for now everything with that sensoris working as expected. FF cluster could also be humidity sensor or PIR which is supposed to be available in with multi sensor (honestly I don't give a s*it as I bought it because it has an external temp prob I can put in the swimming pool but could be useful to sniff other commands if someone got a sniffer)

Smanar, 2022-02-08:
>Shall I try to open a PR with this DDF? Shall I modify the Gold status to Bronze ?

Sure, will be usefull for other

>Min and max interval 60 and 600 instead of this as you suggested?

You have made more test than me ^^, use the value you prefer

>What about refresh interval which is actually 120 ?

Ha yes, good catch, you can completely remove it.
This value is used for exemple for modele version, asked 1 time every 24h.
Totaly useless (and bad for battery) for temperature, because the device use reporting, if you use 120 here, it mean deconz will try to read the temperature itself every 120s.

Stim04, 2022-02-08:
So Bronze it is?
Modifying DDF and loading it on my deconz. Will make a PR for this if it works. On windows GUI i could check boxes to make Bronze loadable. I'm more sceptical about how the debian version will do on Jeedom

Smanar, 2022-02-08:
Right @Mimiix if you can confirm ^^.

From my memory, all submitted DDF need to be bronze, but user that not use GUI will be not able to enable them ?

Mimiix, 2022-02-08:
@Smanar yes indeed. Only if devs approve them as silver or higher we do so. Any user that we don't have as dev has to submit as bronze.

Smanar, 2022-02-09:
So the PR will be validated with status "bronze", and a dev need to switch it to "silver" later ? But when ?

Smanar, 2022-02-09:
Ok lol, so can be done using a text editor for the moment.
Jeedom is not using docker (not yet, they planned to that too :( ), so it s easier.

Stim04, 2022-02-09:
So i keep it gold on my Jeedom in the deconz/device folder for now if I understand correctly. So far sensor is working without refresh interval code line and higher intervals for temp cluster

Smanar, 2022-02-09:
>So far sensor is working without refresh interval code line and higher intervals for temp cluster

Yes, it's better for me.
If an user want closer report for regulation, it just a text file to edit. Better to keep a generic method for other.

>So i keep it gold on my Jeedom in the deconz/device folder

There is 2 folders used by deconz, 1 reserved to deconz, in this one files will be modified at every deconz update, you PR will be in this one, for you and all others users.
The second one never change, and will be prioritary on the first one, so can put your personnal DDF in this one.

> All DDF coming with deCONZ typically reside in /usr/share/deCONZ/devices/generic/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/deconzuser/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

Stim04, 2022-02-09:
Ok i think i understand it now. It was not easy to see the /.local folder on WinSCP but in SSH with putty i could find it. I moved the json accordinly to your last post, leaving it in gold for now.
[image]

[image]

Stim04, 2022-02-09:
As it keeps reporting even hours after a Jeedom reboot, I assume that I indeed placed the json at the right place and that I can now open the PR with that version of the DDF which I put in Bronze status as requested.

https://pastebin.com/YwiyAe76

First time I do a PR so forgive me in advance if I mess up on something. I did read the help section on github about the Pull Requests and will try to do like others do.

closed 2022-03-10 by github-actions[bot]

flopp999, 2022-08-30:
Hi, is this sensor compatible with ConBee II?
I have tried to add it in Phoscon but I do not see any new devices.
The sensor is flashing green(connected).

flopp999, 2022-08-30:
I am using Phoscon 2.17.1
Latest firmware on ConBee II.

flopp999, 2022-08-30:
> Hello, You probably need to edit the json, to enable the DDF
> 
> ` "status": "Bronze",`
> 
> Set the status to "Gold". This device is still on test.

With Gold it worked fine. Thank you

Smanar, 2022-08-30:
Tell us if you have issue, the one to check are :
- missing battery
- battery drain
- working or not
- the update frequency.

some information here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5777

flopp999, 2022-08-30:
I have two of this sensor.
Both was missing battery in the beginning but after appr. 30 min it showed up.

I will check them and get back to you here

Smanar, 2022-08-31:
Ok so you probably have the battery drain.
On deconz your device is like the first one on the first post, no more clusters ?

Can you totaly remove this bind on the DDF (it s the battery)

```
   {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },
```

flopp999, 2022-08-31:
> Ok so you probably have the battery drain. On deconz your device is like the first one on the first post, no more clusters ?
> 
> Can you totaly remove this bind on the DDF (it s the battery), and re-include the device (don't need to delete it, just re-include)
> 
> ```
>    {
>       "bind": "unicast",
>       "src.ep": 3,
>       "cl": "0x0001",
>       "report": [
>         {
>           "at": "0x0020",
>           "dt": "0x21",
>           "min": 300,
>           "max": 3000,
>           "change": "0x00000001"
>         }
>       ]
>     },
> ```
> 
> Can too remove in the state/temperature part
> 
> ```
>           "read": {
>             "at": "0x0000",
>             "cl": "0x0402",
>             "ep": 3,
>             "fn": "zcl"
>           },
> ```

I don't have GUI running I am using a headless Debian.
Can I run GUI on other OS and set the IP to my headless deconz?

But I will try to remove the lines and re-include one of them.

Smanar, 2022-08-31:
>Can I run GUI on other OS and set the IP to my headless deconz?

No, you can't have only 1 instance running desktop or headless.
But can just use a text editor to edit the json, I don't think you have a different model, I never see a poll cluster on this device.

Stim04, 2022-08-31:
Hello,
I would use a computer and access it in ssh with WinSCP to easily edit the json. That's what I did on Jeedom to transfer my file and easily edit it.
That's strange that I didn't have the battery drain problem. Mine wasn't reporting at crazy rates. Too bad the dog chew it and it felt in the swimming pool....

flopp999, 2022-09-08:
I have not changed DDF yet but it seems that the battery drain is not that bad.
Now the battery are 52 and 59 % and started around 85-90%.
I am using rechargeable IKEA batteries

flopp999, 2022-09-08:
Update very often
[image]
[image]
One has reported 656 times the other has reported 487 times last 24 hours.

flopp999, 2022-09-12:
{
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x00000014"
        }
        
        
I will try with 64 instead of 14 in change

mmeglitsch, 2022-11-10:
Working with latest version, on a Windows-based system: Followed instruction using 
[ths317-et.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9983404/ths317-et.json.txt)

After hot reload and re joining, it still does not show temperature data: 

`{
    "1": {
        "config": {
            "configured": false,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "eb32e0df6ccd1fdfa50858e396107bcf",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "status": null
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:07:d3:32-01"
    },
    "2": {
        "config": {
            "battery": null,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "54e6e0a0f752c67fcc431feb0f811b72",
        "lastannounced": "2022-11-10T17:29:25Z",
        "lastseen": "2022-11-10T17:34Z",
        "manufacturername": "OWON",
        "modelid": "THS317-ET",
        "name": "Temperature 2",
        "state": {
            "lastupdated": "none",
            "temperature": 0
        },
        "type": "ZHATemperature",
        "uniqueid": "3c:6a:2c:ff:fe:d2:45:5e-03-0402"
    }
}`

megiator, 2022-11-10:
Working with latest version, on a Windows-based system: Followed instruction using 
[ths317-et.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9983404/ths317-et.json.txt)

After hot reload and re joining, it still does not show temperature data: 

`{
    "1": {
        "config": {
            "configured": false,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "eb32e0df6ccd1fdfa50858e396107bcf",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "status": null
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:07:d3:32-01"
    },
    "2": {
        "config": {
            "battery": null,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "54e6e0a0f752c67fcc431feb0f811b72",
        "lastannounced": "2022-11-10T17:29:25Z",
        "lastseen": "2022-11-10T17:34Z",
        "manufacturername": "OWON",
        "modelid": "THS317-ET",
        "name": "Temperature 2",
        "state": {
            "lastupdated": "none",
            "temperature": 0
        },
        "type": "ZHATemperature",
        "uniqueid": "3c:6a:2c:ff:fe:d2:45:5e-03-0402"
    }
}`

Smanar, 2022-11-10:
Something to try.
Read the temperature in the endpoint 0x03, cluster 0x0402, attribute 0x0000.

If the value is updated in the API > Bind / report issue
Else there is something bad in the DDF, but all seem fine for me

```
          {
          "name": "state/temperature",
          "awake": true,
          "refresh.interval": 120,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 3,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 3,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
```

megiator, 2022-11-11:
After reload and pairing again, there is still no data in the Json: 

`    "2": {
        "config": {
            "battery": 0,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 3,
        "etag": "034f037315c8e88c5f4bc92f4a4cf1c4",
        "lastannounced": null,
        "lastseen": "2022-11-10T21:25Z",
        "manufacturername": "OWON",
        "modelid": "THS317-ET",
        "name": "Temperature 21",
        "state": {
            "lastupdated": "none",
            "temperature": 0
        },
        "type": "ZHATemperature",
        "uniqueid": "3c:6a:2c:ff:fe:d2:45:5e-03-0402"
    }`

I can see data in the deConz Application: 
[image]

Smanar, 2022-11-11:
>After reload and pairing again, there is still no data in the Json:

Just need to read the value like you have done for you capture, when you read a value in deconz, it (need) update too the API.
If the value is updated in the json, it mean the DDF is fine and it's a bind/report problem.
Else its from the DDF.

megiator, 2022-11-11:
Same again, tried with endpoint 0x03 and 0x01 - no changes and no data inside API, so it must be at the DDF side

Smanar, 2022-11-11:
>Same again, tried with endpoint 0x03 and 0x01

BTW you have 2 endpoints on your ? on the first post there is only 1 endpoint, the 0x03.
The DDF is only made for the endpoint 0x03.

Here my last try with the endpoint 0x03 using a fingerprint

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "THS317-ET",
  "product": "THS317-ET",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x03",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x03",
        "in": [
          "0x0000",
          "0x0402",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
            "name": "config/battery",
            "parse": {"cl": "0x0001", "at": "0x0021", "eval": "Item.val = Attr.val"},
            "awake": true
        },
        {
          "name": "state/lastupdated"
        },
          {
          "name": "state/temperature",
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 3,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 3,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 300,
          "max": 3000,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000014"
        }
      ]
    }
  ]
}
```

If this one not working, realy need to check in logs what is bad, in deconz / help / debug vieuw with flag "info" and "info_l2".

Ofc if your device work with the endpoint 0x01 its another story, need to replace all "0x03" by "0x01" and all "ep : 3" by "ep : 1"

megiator, 2022-11-11:
For some stange reason running now.

My config is: 
{
  "schema": "devcap1.schema.json",
  "manufacturername": "OWON",
  "modelid": "THS317-ET",
  "product": "THS317-ET",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x03",
        "in": [
          "0x0000",
          "0x0402",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val + R.item('config/offset').val",
            "fn": "zcl"
          },
          "default": 0
						   
						   
					
										  
					   
		   
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402"
				 
		 
						 
					   
					 
					  
								
		 
	   
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
				 
		 
						 
					   
					
					 
								
		 
	   
    }
  ]
}

Smanar, 2022-11-11:
Ha ? can you share a capture with your cluster list ?
It s the picture with all cluster in blue / gray, and with the endpoint in the title.

We probably need to make 2 DDF with a endpoint check, or use the "auto" feature for endpoint, but not sure we can use it for binding.

flopp999, 2022-12-04:
in HA
[image]

[20 more replies]