dresden-elektronik/deconz-rest-plugin #5740: ROB_200-007-0 only battery entity

besiktas97, 2022-02-01:
I have a 8 button switch:

https://www.robbshop.nl/robb-smarrt-draadloze-schakelaar-wit-8knops-zigbee

The only thing is that I can see the battery information and no other information nor entity's.
When I check Phoscon I can see that the switch is added. When I check deconz, I can see the following:

[image]

[image]

I already try to reset the device and pair it again -> No luck
Already added without resetting -> No Luck

[image]

closed 2022-02-01 by close-issue-app[bot]

Mimiix, 2022-02-01:
@besiktas97 You need to use the template.

Mimiix, 2022-02-01:
To add to that: I think its best to open an topic on the forum as your post probably doesn't comply with #5113.

besiktas97, 2022-02-01:
Where can I find the forum section so I can do that ? Thanks Mimmix

Mimiix, 2022-02-01:
I just replied to your issue @besiktas97