dresden-elektronik/deconz-rest-plugin #5741: ROB_200-007-0 only battery entity

besiktas97, 2022-02-01:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Respberry Pi 4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: (2.13.04 / 18-12-2021
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system?: No

## deCONZ Logs

## Additional context

Mimiix, 2022-02-01:
In HA if you go to Developer tools, Events and then listen for deconz_event.

If you click buttons, do they show? 

[image]

besiktas97, 2022-02-01:
Nothing is showing up when I do what you ask to me.

Mimiix, 2022-02-01:
Ah wait. This is fixed in the last beta (2.14.0).

You need to update to that.  The HA Addon is updated one stable releases. So you probably have to wait a while for that (expecting at least 1 week possibly longer).

Closing.,

closed 2022-02-01 by Mimiix

Mimiix, 2022-02-01:
The new Version of deconz is what you need. If you install deconz on your machine (beta) natively, you can use it already.

The ADdon needs to be updated to the new version. You have to wait for HA to do that for you and then you can update it there.

besiktas97, 2022-02-01:
Ok thank you very much.
Will wait for it ( hope that they will bring out soon the new update :D )

besiktas97, 2022-02-15:
I have installed the new update but I can't see entities of the buttons? I can see events from deconz when I push on the buttons.
How can I make automations so it can trigger for example the lights on/off for each button ?

Mimiix, 2022-02-15:
@besiktas97 That's something you need to ask at HA.