dresden-elektronik/deconz-rest-plugin #5745: Wesmartify essentials thermostate premium - _TZE200_jeaxp72v not fully working

bmgit26, 2022-02-02:
## Describe the bug

Similar to this [post](https://forum.phoscon.de/t/cant-set-temperature-for-eaxp72v-thermostat/541), for me the thermostat is connected to deconz, shows up in HA, but is not responding to turn on or heatpoint setting.

```
{
"config": {
"heatsetpoint": 2050,
"offset": 0,
"on": true,
"reachable": true,
"schedule": {},
"schedule_on": false
},
"ep": 1,
"etag": "28c04a523eacdde52426ed21a7d9b04d",
"lastannounced": "2022-01-27T10:41:30Z",
"lastseen": "2022-02-02T12:11Z",
"manufacturername": "_TZE200_jeaxp72v",
"modelid": "TS0601",
"name": "Thermostat 33",
"state": {
"lastupdated": "2022-02-02T11:27:41.033",
"on": false,
"temperature": 1650
},
"type": "ZHAThermostat",
"uniqueid": "84:71:27:ff:fe:75:d9:e8-01-0201"
}
```

## Screenshots

https://user-images.githubusercontent.com/3626088/152156459-c70c1c23-3004-4df4-b869-6df6c01e9d9a.mov

## Environment

 - Host system: Raspberry Pi 2
 - Running method: Raspbian 
 - Firmware version: 26720700
 - deCONZ version: 2.14.00
 - Device:  RaspBee II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

[deconz_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7986348/deconz_logs.txt)

Mimiix, 2022-02-02:
Awaiting merge to get this added:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5404

issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5745

I've reopened 5745.

Closing.

closed 2022-02-02 by Mimiix

bmgit26, 2022-02-03:
I compiled the not yet merged branch from #5404.
I see more infos now from the Rest-API coming from deconz/thermostat, but I still can't set a new heatpoint that is accepted by the thermostat:

```
{
"config": {
"heatsetpoint": 1500,
"locked": false,
"offset": 0,
"on": true,
"preset": null,
"reachable": true,
"schedule": {},
"schedule_on": false,
"setvalve": false,
"windowopen_set": false
},
"ep": 1,
"etag": "b37bf4433d050e018d6f414f69e49ca2",
"lastannounced": "2022-01-27T10:41:30Z",
"lastseen": "2022-02-03T10:02Z",
"manufacturername": "_TZE200_jeaxp72v",
"modelid": "TS0601",
"name": "Thermostat 33",
"state": {
"lastupdated": "2022-02-03T09:56:52.894",
"on": false,
"temperature": 1330
},
"type": "ZHAThermostat",
"uniqueid": "84:71:27:ff:fe:75:d9:e8-01-0201"
}
```

Can we reopen?

Mimiix, 2022-02-03:
I'd like to have it all in that other issue.  I really thought I reopened it already but I just did again.

Can you share your findings there?

bmgit26, 2022-02-03:
Sure, I can do.