dresden-elektronik/deconz-rest-plugin #5746: Device name

gmitch64, 2022-02-02:
## Device
•	Product name: Neo Temperature & Humidity Sensor (also Lux)
•	Manufacturer: _TZ3000_qaaysllp
•	Model identifier: TS0201
•	Device type :
	Multipurpose Sensor (Light, Temperature, Humidity)

## Screenshots

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

gmitch64, 2022-02-02:
I'm not 100% sure if this should be a new device, or a bug. The device IS being recognized, but it's only returning the Lux info in the JSON packet - there's no Temperature, Humidity or Battery Level being returned.

Currently running 2.13.04 on a Pi 3B+
deconz-dev/unknown,now 2.13.04-raspbian-buster-stable armhf [installed]
deconz/unknown,now 2.13.04-raspbian-buster-stable armhf [installed]

I have the dev tools installed, so I can compile test code for testing if needed.

If there's any other info that the devs need, please let me know.

G

Mimiix, 2022-02-02:
Hi,

Seems to be a double post. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4790#issue-864100128

Please use the search next time.

Ill reopen the other issue.

closed 2022-02-02 by Mimiix