dresden-elektronik/deconz-rest-plugin #5747: LK Wiser 4 Button Switch adopts but does not work in 2.14.00

mwaldhauer, 2022-02-03:
## Describe the bug
Button event are not created when adding switches. I have tried 3.

## Steps to reproduce the behavior
Output from API:
config           : @{battery=; on=True; reachable=False}
ep               : 21
etag             : 70c76eff8cfe0b0be5a2a332013555a5
lastannounced    :
lastseen         : 2022-02-01T04:59Z
manufacturername : Schneider Electric
mode             : 2
modelid          : FLS/AIRLINK/4
name             : TestSwitch
state            : @{buttonevent=; lastupdated=none}
type             : ZHASwitch
uniqueid         : 0c:43:14:ff:fe:46:1e:f8-15-0006

## Expected behavior
Buttenevents are created on adopt

## Screenshots
[image]

## Environment
 - Host system: Unraid
 - Running method: Docker container
 - Firmware version: 26720700
 - deCONZ version: 2.14.00
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Which? Z-wave stick

## deCONZ Logs
04:38:16:902 DEV found DDF for 0x588E81FFFE7BC86E, path: /opt/deCONZ/devices/fyrtur_block-out_roller_blind.json
04:38:16:902 DEV no DDF for 0x00158D000488BDE1, modelId: lumi.sensor_magnet.aq2
04:38:16:902 DEV create on-the-fly DDF for 0x00158D000488BDE1
04:38:16:903 DEV found DDF for 0x588E81FFFE7BC86E, path: /opt/deCONZ/devices/fyrtur_block-out_roller_blind.json
04:38:16:903 DEV found DDF for 0x00158D000488BDE1, path: 
04:38:16:905 dlg action: Read binding table
04:38:16:917 DEV no DDF for 0x001788010B9B34D9, modelId: LCA004
04:38:16:917 DEV create on-the-fly DDF for 0x001788010B9B34D9
04:38:16:917 DEV found DDF for 0x001788010B9AFA43, path: 
04:38:16:917 DEV found DDF for 0x001788010B9AFC21, path: 
04:38:16:917 DEV no DDF for 0x680AE2FFFEA4469A, modelId: BoxDIMZ 98425031
04:38:16:917 DEV create on-the-fly DDF for 0x680AE2FFFEA4469A
04:38:16:917 DEV no DDF for 0x00124B0021376B6F, modelId: GL-C-008
04:38:16:917 DEV create on-the-fly DDF for 0x00124B0021376B6F
04:38:16:917 DEV found DDF for 0x00124B0021383D9C, path: 
04:38:16:917 DEV found DDF for 0x00124B00213AF3C9, path: 
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, create item: config/battery
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/id
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/lastannounced
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/lastseen
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/manufacturername
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/modelid
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/name
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/swversion
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/type
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: attr/uniqueid
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: config/on
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: config/reachable
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, create item: state/buttonevent
04:38:16:918 sub-device: 68:0a:e2:ff:fe:11:67:f3-15-0006, has item: state/lastupdated
04:38:16:918 DEV found DDF for 0x680AE2FFFE1167F3, path: /usr/share/deCONZ/devices/wiser/fuga_4button_battery_switch.json
04:38:16:918 DEV found DDF for 0x001788010B9B1C27, path: 
04:38:16:918 DEV no DDF for 0xF0D1B800001280AF, modelId: Panel TW Z3
04:38:16:918 DEV create on-the-fly DDF for 0xF0D1B800001280AF
04:38:16:919 sub-device: 0c:43:14:ff:fe:46:1e:f8-15-0006, create item: config/battery
04:38:16:919 sub-device: 0c:43:14:ff:fe:46:1e:f8-15-0006, has item: attr/id
[... 1285 lines cut ...]
04:39:24:613 ZDP Mgmt_Lqi_rsp zdpSeq: 41 from 0x001788010B6337EA total: 13, startIndex: 9, listCount: 3
04:39:24:613     * neighbor: 0xF0D1B800001280AF (0x656F), LQI: 191, relation: 0x02, depth: 15, rxOnWHenIdle: 1
04:39:24:613     * neighbor: 0x7CB03EAA00AD80D8 (0xB591), LQI: 57, relation: 0x02, depth: 15, rxOnWHenIdle: 1
04:39:24:613     * neighbor: 0x0017880106F51483 (0xF13D), LQI: 141, relation: 0x01, depth: 2, rxOnWHenIdle: 0
04:39:24:813 APS-DATA.request id: 41, addrmode: 0x03, addr: 0x00124b0021383d9c, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
04:39:24:827 APS-DATA.confirm id: 41, status: 0x00 SUCCESS
04:39:24:827 APS-DATA.confirm request id: 41 -> erase from queue
04:39:24:869 APS-DATA.indication srcAddr: 0x5c1e, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -83
04:39:24:869 APS-DATA.request id: 41 erase from queue
04:39:24:869 ZCL got data for node=0x5C1E, cl=0x0008, at=0x0000, status=0x00, type=0x20
04:39:24:914 APS-DATA.request id: 42, addrmode: 0x03, addr: 0x00124b0021383d9c, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x0B queue: 0 len: 19 tx.options 0x00
04:39:24:927 APS-DATA.confirm id: 42, status: 0x00 SUCCESS
04:39:24:927 APS-DATA.confirm request id: 42 -> erase from queue
04:39:24:982 APS-DATA.indication srcAddr: 0x5c1e, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -82
04:39:24:982 APS-DATA.request id: 42 erase from queue
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x0001, status=0x00, type=0x20
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x0003, status=0x00, type=0x21
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x0004, status=0x00, type=0x21
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x0007, status=0x00, type=0x21
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x0008, status=0x00, type=0x30
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x4000, status=0x00, type=0x21
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x4001, status=0x00, type=0x30
04:39:24:982 ZCL got data for node=0x5C1E, cl=0x0300, at=0x4002, status=0x00, type=0x20

## Additional context

Mimiix, 2022-02-03:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5680/

It;'s that PR.

mwaldhauer, 2022-02-07:
Fix only worked for the 1st switch,  my 3 others will not pair correctly

hanskroner, 2022-02-07:
I'm seeing a similar situation in 2.14.01. I can see button_events from the first switch I joined, but not from the second.

Additionally, the switch that does generate button_events will only generate _S_BUTTON_ACTION_SHORT_RELEASED_ events. I've tried extending its button map in the .json file to add _S_BUTTON_ACTION_HOLD_ and _S_BUTTON_ACTION_LONG_RELEASED_ events, with no luck. Maybe I just don't understand the format for these maps? The documentation is thin, and I'm not clear on what the "Mode of operation" parameter is supposed to be.

mwaldhauer, 2022-02-09:
Do we need to open a new bug on this ? @Mimiix

Mimiix, 2022-02-09:
Im happy to reopen.

hanskroner, 2022-02-09:
I opened pull request [#5772](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5772) with changes to the button map that fixed both issues for me. I removed the devices from the network, reset them, and re-joined them with the extended button map in the pull request. Would appreciate if anyone else could give them a test!

mwaldhauer, 2022-02-09:
Would love to, but i only run in docker image so im not sure how to

manup, 2022-02-10:
> I opened pull request [#5772](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5772) with changes to the button map that fixed both issues for me. I removed the devices from the [...]

Nice thanks, I'll give it a test later on.

hanskroner, 2022-02-10:
> Would love to, but i only run in docker image so im not sure how to

If you followed the instructions for setting up your Docker from the deCONZ Community page, you've already mapped `/opt/deCONZ` form inside the container to somewhere (the instructions suggest `/opt/deconz`). In there you'll have a `devices` directory - plop the two files in my pull request in there - just be mindful that `fuga_4button_battery_switch.json` needs to be inside a `wiser` sub-directory.

mwaldhauer, 2022-02-18:
I have 2 switches working now, but it seems that if they try to join the network through another device, not the configuration tool the button events does not work

mwaldhauer, 2022-02-18:
Tried a third switch. First it joined through a lamp, didnt work. Second it joined through Configuration Tool, worked.

Also buttons are not registering in Home Assistant.

EjvindHald, 2022-02-20:
@mwaldhauer , @hanskroner 

Can you guys adopt the Fuga 4 button in Phoscon ?

I have tried to adopt many times, but after a while the device eventually blink red and nothing is created in Phoscon. The other new Wiser devices is adopted fine.

hanskroner, 2022-02-21:
@EjvindHald 
> I have tried to adopt many times, but after a while the device eventually blink red and nothing is created in Phoscon. The other new Wiser devices is adopted fine.

I have two of them. Both of them joined through the Phoscon UI. I made sure to reset them, following the button-holding procedure in their manual first. After applying the PR above, I could also set them up using Phoscon's Switch Editor - though that's not the way I'm using them at the moment.

@mwaldhauer
> Also buttons are not registering in Home Assistant.

They don't show up as buttons in HA. Instead, you listen for 'deconz_event' and filter by ID. You can go into the 'Developer Tools' section in HA and then the 'Events' tab. You can type the service name there and listen for events - press a button on your switch and HA will show you the event payload so you can see what it looks like.

EjvindHald, 2022-03-06:
> I opened pull request [#5772](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5772) with changes to the button map that fixed both issues for me. I removed the devices from the [...]

Thank you very, very much for this contribution. I can confirm that it is working absolutely perfect.

mwaldhauer, 2022-03-25:
I am not getting the correct button event from the lower 2 buttons...i get 1002 and 2002, not 3002 and 4002 which should be 3 and 4...

EjvindHald, 2022-03-25:
Keep button 3 and 4 in at the same  time for 20 seconds.

This is because the default in early firmware release was that button 3 and 4 were not enabled. They can be enabled by either the gateway or the long press explained above. Unfortunately, this is not documented anywhere.

closed 2022-04-23 by github-actions[bot]