dresden-elektronik/deconz-rest-plugin #5753: Xiaomi ZNCLDJ11LM - OPEN and STOP works, CLOSE gets rejected

gr08, 2022-02-04:
## Describe the bug

I have a Xiaomi Curtain motor ZNCLDJ11LM. When I use deconz app, I can open, close and stop + read position, which is pretty much all I need. But when I use Node-Red and deconz-OUT, I can only open and stop. JSON close will give an error.

I'm using these JSONs:
{ open: true } - working fine
{ stop: true} - working fine
{ close: true } - gives error

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]
[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian. Deconz headless image running as docker.
 - Firmware version: 26580700
 - deCONZ version: 2.13.4
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, SSD

## deCONZ Logs

## Additional context

I tested various others JSONs for closing that I was able to find but no success.

Any idea what could be the problem or what else I can look at?

I run various other sensors, switches and lights from various vendors without problem.

Mimiix, 2022-02-04:
What does the rest api report when you open and close using that?

I am tempting to close this issue as it doesn't comply with #5113 and as it looks like an issue in node red. Nevertheless, I give it a chance. Please read this careful and let me know if the api works. If that does work, it's an node red issue. https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

gr08, 2022-02-04:
Ah sorry, I didn't transcript the error message from the screenshot.

When I send open/stop, there is no response, simply the motor stars opening or stops. When I send close, the response is:
`deconz-out ERROR: parameter, close, not available`

I understand the concern it might be Node-red issue. I spent one month trying to figure this out and I lean towards this being a deconz issue which I why I tried to find help here. As I mentioned in my OP, with the same API I do control switches, lights and read many sensors without issue. I can also open/stop the very same xiaomi curtain motor.

Mimiix, 2022-02-04:
@zehir Is this a node red issue?

SwoopX, 2022-02-04:
It is, the API does not have any `close` parameter. Closing is `open: false`. Should be closed over here.

gr08, 2022-02-04:
`open: false` that's it. It's working. Sorry for wasting your time gentlemen, you are awesome!

Mimiix, 2022-02-04:
Case closed :)

closed 2022-02-04 by Mimiix

Zehir, 2022-02-05:
> It is, the API does not have any `close` parameter. Closing is `open: false`. Should be closed over here.

No it's not a node red issue ;) the node red plugin just help converting the rest api to node red. It's was not working because the close option do not exist in the rest api. It's also what response is (who came from the rest api)