dresden-elektronik/deconz-rest-plugin #5755: philips hue motion sensor sml003

sn0wf4ke, 2022-02-05:
## Device

- Product name: hue motion sensor
- Manufacturer: Signify Netherlands B.V.
- Model identifier: SML003
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

### notes

it works with the DDF of a phillips hue motion sensor sml001 when model is renamed, but the luminance sensor doesn`t work. motion and temperature are fine.

SwoopX, 2022-02-06:
You'd probably need to copy file `sml_light_level.js` to the same location as your newly created DDF. Once don, I'd recommend a deconz restart instead of a hot reload of the file.

sn0wf4ke, 2022-02-06:
> You'd probably need to copy file `sml_light_level.js` to the same location as your newly created DDF. Once don, I'd recommend a deconz restart instead of a hot reload of the file.

this does the trick... thank you
but if it is so simple... please add this DDF as SILVER to have the out of the box experience 👍 :

```json
{
    "schema": "devcap1.schema.json",
    "doc:path": "philips/sml003_motion_sensor.md",
    "doc:hdr": "Motion sensor 2. Gen (SML003)",
    "manufacturername": "$MF_PHILIPS",
    "modelid": "SML003",
    "product": "Motion Sensor 2. Gen (SML003)",
    "status": "Bronze",
    "sleeper": false,
    "md:known_issues": [ ],
    "subdevices": [
         {
            "type": "$TYPE_PRESENCE_SENSOR",
            "restapi": "/sensors",
            "uuid": [ "$address.ext", "0x02", "0x0406"],
            "fingerprint": { "profile": "0x0104", "device": "0x0107", "endpoint": "0x02", "in": ["0x0000", "0x0001", "0x0406"] },
            "items": [
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/alert",
                    "default": "none"
                },
                {
                    "name": "config/battery",
                    "parse": {"ep": 2, "cl": "0x0001", "at": "0x0021", "eval": "Item.val = Attr.val / 2"},
                    "read": {"ep": 2, "cl": "0x0001", "at": "0x0021"},
                    "refresh.interval": 7300,
                    "awake": true
                },
                {
                    "name": "config/delay",
                    "read": {"ep": 2, "cl": "0x0406", "at": "0x0010"},
                    "parse": {"ep": 2, "cl": "0x0406", "at": "0x0010", "eval": "Item.val = Attr.val"},
                    "refresh.interval": 84000
                },
                {
                    "name": "config/ledindication",
                    "read": {"ep": 2, "cl": "0x0000", "at": "0x0033", "mf": "0x100b"},
                    "parse": {"ep": 2, "cl": "0x0000", "at": "0x0033", "mf": "0x100b", "eval": "Item.val = Attr.val"},
                    "refresh.interval": 84000
                },
                {
                    "name": "config/on"
                },
[... 264 lines cut ...]
            "src.ep": 2,
            "cl": "0x0406",
            "report": [ {"at": "0x0000", "dt": "0x18", "min": 1, "max": 300 } ]
        },
        {
            "bind": "unicast",
            "src.ep": 2,
            "cl": "0x0400",
            "report": [ {"at": "0x0000", "dt": "0x21", "min": 5, "max": 300, "change": "0x07d0" } ]
        },
        {
            "bind": "unicast",
            "src.ep": 2,
            "cl": "0x0402",
            "report": [ {"at": "0x0000", "dt": "0x29", "min": 10, "max": 300, "change": "0x14" } ]
        },
        {
            "bind": "unicast",
            "src.ep": 2,
            "cl": "0x0001",
            "report": [ {"at": "0x0021", "dt": "0x20", "min": 7200, "max": 7200, "change": "0x00" } ]
        }
    ]
}
```

SwoopX, 2022-02-06:
Could you please try the DDF from the PR? I've noticed that the new one got a broader sensitivity range (0-4 instead of 0-2). Hope the changes work out as well.

sn0wf4ke, 2022-02-06:
done the testing and commented to PR

Mimiix, 2022-02-06:
Great job guys

closed 2022-02-06 by manup