dresden-elektronik/deconz-rest-plugin #5763: Sonoff ZBMINI-L

SarcasticShrimp, 2022-02-06:
## Device

- Product name: SONOFF ZBMINI-L Zigbee 3.0 Smart Switch
- Manufacturer: SONOOF
- Model identifier: ZBMINI-L
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch
  

## Screenshots

### Basic
[image]
[image]
### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

[image]

pyrech, 2022-02-28:
I guess this issue should be kept open until support is added even if we have to wait a bit until #5733 is done?

In the meantime I created https://forum.phoscon.de/t/sonoff-zbmini-l/1686 as requested in #5113

Mimiix, 2022-02-28:
Yes  but you need to keep bumping this open once in a while.

pyrech, 2022-03-30:
Doh, not the OP here but still need this feature request to be kept open 😅

Mimiix, 2022-03-30:
Sure!
@Smanar  Can  we make a DDF?

sir-Unknown, 2022-04-23:
Not the OP. Got the device working by adjusting the DDF. However would like to have the possibility to receive the toggle of the power on/off from the wire. I don't have the skills to add that.

Smanar, 2022-04-24:
>However would like to have the possibility to receive the toggle of the power on/off from the wire. I don't have the skills to add that.

You mean the return when using the device manualy ? It's the binding part https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel
Else if you can copy/paste the DDF I can complete it.

closed 2022-05-23 by github-actions[bot]

moecre, 2022-05-25:
> Not the OP. Got the device working by adjusting the DDF. However would like to have the possibility to receive the toggle of the power on/off from the wire. I don't have the skills to add that.

I'm confused right now. Does this mean the device is compatible/partly compatible? These devices are the missing link for my smart home and I'd like to know if I can start buying them. Thanks.

felixnurse, 2022-05-25:
I have both devices (zbmini and zbmini-l) if that helps. 
Let me know what I need to share from deconz and illlpaste here for you.thanks

Smanar, 2022-05-25:
Depend of device but for exemple for the ZBMINI-L, you can start a DDF 
- fill the basic cluster indentification
- drag and drop a "on off light"
- save the file, llok the name and path on the editor
- share the contain

Some part can be auti filled
The device can work direclty if you take "hot relaod" in the editor menu. You will just miss the binding, but you can try to set them too.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

ads1230, 2022-05-29:
`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "ZBMINI-L",
  "product": "ZBMINI-L",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/zbmini-l.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
`

Smanar, 2022-05-29:
You miss the binding/reporting but not sure you need it ?

Without it the device don't make state report, not sure it usefull for this kind of device but it prevent having different state on the api and on the reality.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "ZBMINI-L",
  "product": "ZBMINI-L",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    }
  ]
}
```

Have added the bind/report for the on/off cluster.

If it still working can make a PR to add it ?

ads1230, 2022-05-29:
> You miss the binding/reporting but not sure you need it ?
>
> Without it the device don't make state report, not sure it usefull for this kind of device but it prevent having different state on the api [...]

Working with your edits. Set up 3 devices today all working fine even after reboot.

Smanar, 2022-05-30:
Nice, do you want tio make a PR or wana I make it ? (to add it officialy)

ads1230, 2022-05-30:
> Nice, do you want to make a PR or wana I make it ? (to add it officialy)

Could you do it please. I'm not too savvy on these things.

Smanar, 2022-05-30:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6095
Thx Again