dresden-elektronik/deconz-rest-plugin #5769: Aqara Smart Wall Switch H1 (With Neutral, Single Rocker/ Double Rocker)

JPOliveGit, 2022-02-08:
## Aqara Smart Wall Switch H1 (With Neutral, Single Rocker/ Double Rocker)

- Product name: H1 Wired
- Manufacturer: Aqara
- Model identifier: WS-EUK03 / WS-EUK04
- Device type :
  - Switch

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-02-08:
Missing screenshots. Please open another issue with correct template usage.

closed 2022-02-08 by Mimiix

SwoopX, 2022-02-08:
No, already supported

JPOliveGit, 2022-02-17:
> 

it can't find it... SwoopX the wireless ones work well but the wired do not...