dresden-elektronik/deconz-rest-plugin #5773: Schneider Electric / Wiser Merten MEG5113-0300

KingTomaHawk, 2022-02-09:
## Device

- Product name: Schneider Electric / Wiser Merten MEG5113-0300 (maybe device under the switch is also relevant: MEG5165-0000)
- Manufacturer: Schneider Electric
- Model identifier: 1GANG/SHUTTER/1
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other: Window Covering Device

## Screenshots
[image]
[image]

### Basic (Endpoint 05 and 15 same)
[image]

### Identify (Endpoint 05 and 15 same)
[image]

### Groups
[image]

### Scenes
[image]

### Window Covering Device
[image]

### Diagnostics (Endpoint 05 and 15 same)
[image]

### innr specific (typo, should be merten specific)
[image]

KingTomaHawk, 2022-02-09:
Note: Swoop helped me already with discovering e000 attributes and FF17 cluster. 

Still some functions missing or unknown:
1. Roller Shutter Movement time was not configurable -> Solved! 0xE000 should me the movement time in seconds.
2. When setting roller shutter to percent position, first it drives completely up then down to desired position. Maybe calibration required?
3. 0x0001 from FF17 Cluster set itself once to Standard and the physical buttons stopped working. Setting it to Toggle solved it, default was "3" . Not sure what that attribute does.
4. 0xe010 and 0xe013 are read only.

jensflorian, 2022-02-09:
The device under the switch is indeed relevant. Depending on the PlusLink inset used (https://www.merten.de/produkte/wisertm-smart-home/lichtsteuerung-1/pluslink-elektronik-einsaetze-2.html) the module ID will be different.

For example using MEG5113-0300 together with MEG5161-0000 relay the device will issue the model ID: 1GANG/SWITCH/1
Accordingly the combination of the two-button module MEG5123-0300 with MEG5172-0000 dimmer will result in: 2GANG/DIMMER/2

Can you describe how you added FF17 cluster? I'm still struggling here.
One attribute should toggle a green LED used as orientation light in the dark. Another one probably toggles the auto-off timer.

According to this  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4534#issuecomment-1014980941_ , Dresden Electronic has some Wiser Modules for testing and implementation.
I'd really see to have proper DDFs including button-events from these devices.

KingTomaHawk, 2022-02-10:
Thank you for the info! Do you also have the roller shutter? 

Swoop is working on fixing some bugs, but you can get these clusters by cloning his merten branch from https://github.com/SwoopX/deconz-rest-plugin and replacing the general.xml in  /usr/share/deCONZ/zcl/ .

Always make a backup first :)

KingTomaHawk, 2022-02-13:
More secrets revealed from FF17 cluster:

0x0001:
unknown_0 disabled buttons
unknown_1 disabled buttons
unknown_2 no difference to unknown 3
unknown_3 default
unknown_4 disabled buttons
unknown_5 disabled buttons
unknown_6 disabled buttons
unknown_7 disabled buttons

0x0000:
unknown_0 default (orientation (green) led off / movement (red) led on)
unknown_1 orientation led on / movement led on
unknown_2 orientation led on / movement led off
unknown_3 orientation led off / movement led off
unknown_4 unknown effect
unknown_5 unknown effect
unknown_6 unknown effect
unknown_7 unknown effect

Something seems to have disabled my windows covering option :P 
I think there is a inverted motors option still hidden somewhere.

SwoopX, 2022-02-13:
@jensflorian I added some functionality to the aforementioned branch to do some attribute discovery on the 0xff17 cluster, that should work in your case as well (with expectedly some different results popping out). However, this should only work if the cluster is on endpoint 15. If yours is on a different cluster, the command must be slightly amended.

It gets triggered while reading any attribute on the `identify` cluster. I guess I was sane enough to also output the results as L2 debug.

jensflorian, 2022-02-14:
> Something seems to have disabled my windows covering option :P I think there is a inverted motors option still hidden somewhere.

Pressing the button 3 times and holding for 10sec should reset the device to factory defaults.

For 0x0000 the attribute editor still has the old innr description "Air pressure (mBar)", this should be changed to something as "LED settings" 

According to the options available in wiser app following modes are probably assigned to some of the clusters within 0xFF17:
- Activate timer (on/off)
- Stair light mode (on/off)  this means:
 For dimmer, lights will be dimmed down before timer goes off.
 For switch, lights will be shortly turned off and on again before timer goes finally off
- LED mode (on/off) on reduces significantly max ballast on dimmer
- Memory mode (on/off): device will remember last light intensity, otherwise use predefined light setting.
- Running time (for shutter), default is 2min, can be changed from 00:10 seconds to 05:00 mins. (this is probably 0xE000) ?

To get buttonevents (this should be identical for shutter, dimmer and switch) I tried to add ZHASwitch functionality to the DDF editor but it seems I'm missing something here. First I dragged basic cluster from endpoint 15 to the DDF bindings window:

[image]

Then I added ZHASwitch to the device settings (For the shutter additionally "Window Covering Device" can be added):
[image]

For buttonevents I selected parse zcl and added endpoint 15:
[image]

But in the description field, no last received button event is shown. What I am missing here?

SwoopX, 2022-02-14:
A ZHASwitch resource can only work with client clusters (grey) in your case for on/off, level control, scenes and window covering. I don't know if the current code appropriately create and handle those resources appropriately (yet). Even if it does, it would require a tailored button map to be created to emit any button events. The respective messages to create one should be in the debug log (INFO_L2 iirc).

For usage of true zigbee groups, note that the client cluster of a device is bound to a group. The device to be controlled must then have the respecive cluster in its list and must be part of that group. That's best done via the binding dropbox.

KingTomaHawk, 2022-02-14:
> * Running time (for shutter), default is 2min, can be changed from 00:10 seconds to 05:00 mins. (this is probably 0xE000) ?

Yes. I mentioned it in the second comment.

> * Activate timer (on/off)
> * Stair light mode (on/off) this means:
> For dimmer, lights will be dimmed down before timer goes off.
> For switch, lights will be shortly turned off and on again before [...]

I don't think that I will be able to identify these features, because my roller shutter module does not support it.

There are still some attributes that I have not changed. I think that some options of the attributes require another module under the button, so I will not find all the functions.

KingTomaHawk, 2022-02-27:
I think I'm done with this device. Nothing to discover anymore on my side.
But I'm not sure if the device is just very dumb or if it's a bug, but setting a lift position, starting from any other position then open state, causes the the roller shutter to first go full up, wait until the "movement time" is up, then go to desired position.
Swoop tried to find the root of this behavior, without success. So its possible that this is just the way the device works.

TLDR: That's all I could find out about this devices and if you want lift positions (maybe) don't buy this product. 😄

closed 2022-03-28 by github-actions[bot]

axellebot, 2022-08-23:
> Thank you for the info! Do you also have the roller shutter?

I recently bought 10 shutter but they are not working properly with home-assistant ZHA (mostly due to missing travel time calibration).
Is there a way I can help ?

axellebot, 2022-08-23:
There are multiple issues for SE device support with [Zigpy](https://github.com/zigpy/) :
 - [CH10AX/SWITCH/1](https://github.com/zigpy/zha-device-handlers/issues/1696)
 - [NHPB/SHUTTER/1](https://github.com/zigpy/zha-device-handlers/issues/1685)
 - [FLS/AIRLINK/4](https://github.com/zigpy/zha-device-handlers/issues/1623)

jensflorian, 2022-10-20:
First attempt with some basic DDF

```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "1GANG/SWITCH/1",
  "product": "1GANG/SWITCH/1",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x15",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

INFO2 log says:

20:14:25:452 [INFO] - No button map for: 1GANG/SWITCH/1, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 103

Can someone help me adding button maps to the above DDF?

jensflorian, 2022-10-21:
Nevermind, I just figured it out. I added the device in the button_maps.json

```
"modelids": ["FLS/AIRLINK/4", "FLS/SYSTEM-M/4", "1GANG/SWITCH/1"],
```
Now I can parse the button events.