dresden-elektronik/deconz-rest-plugin #5778: Lockwood T-Lock Deadbolt - YDD-D4F0 TSDB

Sian-Lee-SA, 2022-02-10:
I took a stab at making a DDF but could not get anything for locks. The best I could do was get the battery state. 'm not sure of the current stance on locks within deCONZ but I do wish to have this lock added.

## Device

- Product name: Lockwood T-Lock
- Manufacturer: Yale
- Model identifier: YDD-D4F0 TSDB
- Device type : Please remove all unrelated device types. 
  - Lock

## Screenshots

### Basic

### Identify
[image]
[image]
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

[image]
[image]

Smanar, 2022-02-10:
Hello, you are not alone ^^.

Someone else have started one here  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5778

If you have too much problem can you show your DDF ?

Mimiix, 2022-02-10:
Closing as it's duplicate

Smanar, 2022-02-11:
Oups, sorry, mistake.
It's not a duplicate, but someone else is trying DDF for doorlock https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5775

closed 2022-03-13 by github-actions[bot]

Smanar, 2022-03-13:
@Sian-Lee-SA you have give up ?

Sian-Lee-SA, 2022-03-13:
Never really took it on, played around a bit then decided to give Zigbee2MQTT a try which had everything working from get go. I also thought to explore Zigbee2MQTT because every new device I get seems to already be supported. But after using it for a while I have started to see the cracks in it. But their lock for my specific model has everything including the add user and other bits an pieces.

After a few weeks of using Zigbee2MQTT, I had decided to dust of my old Philips Hue bridge specifically for my lights as they weren't playing nice. So right now I have a hybrid setup and personally finding the hybrid setup more stable and compatible atm.