dresden-elektronik/deconz-rest-plugin #5785: Wrong gateway address in API

TimoPax, 2022-02-12:
## Describe the bug
Calling the API I get in return the right ipaddress, but the gateway-address is wrong.

http://192.168.29.11/api/apikey/config
{ "gateway":192.168.29.1 }

If "gateway" is meant to be the network gateway ip-address than in my case it's 192.168.29.254.
Through the API the gateway can't be changed (API DOC) the question is how does the software determes this address and why is it wrong?

The API-Version is 1.16.0. 

## Steps to reproduce the behavior
1. Goto http://yourip/api/apikey/config
2. Check the gateway.

## Expected behavior
I would expect to see the network gateway ip-address.

## Screenshots

## Environment
 - Host system: VM (proxmox)
 - Running method: debian11
 - Firmware version: (26720700)
 - deCONZ version: (2.14.1)
 - Device: ConBee II 
 - Do you use an USB extension cable: (yes / no) -- yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2022-02-12:
The gateway in this case is the deconz installation. Meaning zigbee gateqay. Not sure why yours return 1and not 11.

ebaauw, 2022-02-13:
`gateway` in `/config` is the default IP router in your home network, same as for the Hue API.  I don’t know how deCONZ determines this, my router is on `x.x.x.1`.

I don’t know if deCONZ even uses this.  Maybe it is (was?) used for the [gateway product](https://www.dresden-elektronik.de/produkt/phoscon-gateway.html), where deCONZ manages Raspberry Pi OS.  Same for the WiFi config.  Maybe it was just added to mimic the Hue API.

TimoPax, 2022-02-13:
Thank you. Like I said, it's not the Router IP nor the IP of the current Deconz installation.
It used to be the IP address of an old installation of Deconz.
Are you saying that it doesnt matter if it's the right address or not, because it's not being used?

ebaauw, 2022-02-13:
That’s what I suspect/hope.  I’ll try and scan the code to see if it’s used somewhere when I’ll be at my computer.

ebaauw, 2022-02-13:
Looks like it's generated on the fly when doing a GET of `/config`:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/80952d6c5bfc5a12d3527176602b8423948897de/rest_configuration.cpp#L1066-L1078

And that also explains why it ends with `.1` rather than showing the default router address.  For fun: check what this looks like with a netmask other than `255.255.255.0`.

TimoPax, 2022-02-13:
:-)
Well, that explains the .1 of course and why there is no way to configure it. Thank you very much for investigating.
Hopefully, nobody is counting on this value than. Is this a bug or should I could this ticket?

ebaauw, 2022-02-13:
It's a bug alright, but with (close to) zero impact.  I think there are many far more pressing issues.

manup, 2025-05-17:
Note to myself, this attribute should be removed. I think it's another relict from the Hue API and never really worked, nor is it used by any client as far as I know.