dresden-elektronik/deconz-rest-plugin #5786: _TZ3000_4upl1fcj

Spirit1240, 2022-02-12:
## Device

- Product name: Scene Switch
- Manufacturer: _TZ3000_4upl1fcj
- Model identifier: TS0041
- Device type : Switch

## Screenshots

### Basic

[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Node info

[image]

Mimiix, 2022-02-12:
Missing screenshots.

hastime, 2022-02-28:
@Mimiix - which screenshots do you need I will be happy to add them as I have one of these, I was trying to create a DDF for it, I have it showing up in Phoscon with the DDF now but I can't see any button events and in Phoscon I cannot add it to a group.

Thanks,

Kind regards

Mimiix, 2022-02-28:
Screenshots seem to be fine, but not all are read so values are missing. 

We want them to be on the opening post though so @Spirit1240 needs to add them.

hastime, 2022-02-28:
@Mimiix - No problem, I don't want to create a duplicate Device Request, can I ask here, the switch shows in Phoscon and deconz but I get no events in the API or on HA, I also cannot add it to any Phoscon groups, is this because the switch needs whitelisting? 

Also the switch light flashes every now again, is this because deconz is polling it for some reason? The four button version of this switch doesn't flash without user interaction.

Thanks again,

Kind regards.

hastime, 2022-03-03:
Hi, 

Sorry to keep chasing but is there anything I can do to try and progress this? Or do I need to wait until it goes stale and log a new device request? 

Kind regards

Mimiix, 2022-03-03:
@hastime feel free to duplicate this one and I'll close this one. Please note that even if your request is valid, implementation can be halted due to #5733. I don't know if this can be done with ddfs.

hastime, 2022-03-03:
> @hastime feel free to duplicate this one and I'll close this one. Please note that even if your request is valid, implementation can be halted due to #5733. I don't know if this can be done with ddfs. 

@Mimiix Thanks for this I will create a device request for this switch. I understand that due to #5733 it may not progress as fast. 

I think it's worth the delay to have the work on DDFs progressed, because as a user being able to set up your own devices would be awesome! 

As always, thanks for your help!

Kind regards,

Mimiix, 2022-03-03:
You are most welcome :)

Feel free to ping me in the new request.

Mimiix, 2022-03-04:
Closing as we have a new , complete issue.

closed 2022-03-04 by Mimiix

MBj1703, 2023-02-09:
Sorry for stupid question, but what happened to this request?
I have the same.