dresden-elektronik/deconz-rest-plugin #5787: Setting a schedule's localtime returns a defect result

meandthemachine, 2022-02-13:
Hello,

when calling the API for setting a schedule's `localtime` (only), on success, the corresponding route's handler always tries to return/reflect `time` which is `null` in such a case, as it hasn't been part of the request. 
Also, the returned slug contains `"time"` instead of `"localtime"`.
For example:

PUT body:
```
{"localtime": "W127/T12:00:00"}
```

Current response:
```
[{"success":{"/schedules/1/time":null}}]
```

Expected response:
```
[{"success":{"/schedules/1/localtime":"W127/T12:00:00"}}]
```

I think the issue is caused by this line, that wrongly tries to reflect the `time` field of the request, even if it hasn't been given.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/80952d6c5bfc5a12d3527176602b8423948897de/rest_schedules.cpp#L669

Thank you and cheers!

meandthemachine, 2022-02-13:
I opened the issue via source code line context menu "reference in new issue". That one missed the template. I think I've provided enough information, though. Maybe someone reads this, despite the bot closing it...

meandthemachine, 2022-02-13:
@manup Please consider reopening this issue. I had a look at your bug report template and I think I already gave all information that is applicable to and needed for this issue. It is a quite evident bug, I think. Thank you.

Mimiix, 2022-02-14:
@meandthemachine I am not able to reopen at all as the bot will close it again. We ask to use the template so that all reports are the same. There's quiet some info missing. Please create a report with the template

meandthemachine, 2022-02-14:
I apologize for that one sentence that I just removed from my previous comment. That was not OK at all. Please excuse me.

meandthemachine, 2022-02-14:
@Mimiix I had a look at the bot's README and edited my issue accordingly. So now, maybe it won't be closed again if you try to reopen it...

Mimiix, 2022-02-14:
@meandthemachine I understand your frustration, but it isn't doable for us to maintain all issues being differently. That's why we chose the template. Most of us are doing this in their free time (as we are not employed by DE). So anything the user can do to help us a bit is better for everyone in the end. 

In this example, the first thing we would ask is the current setup you run (and version) as that was missing from your first post. 

I'm happy to re-open  and apology accepted 🙂

Mimiix, 2022-02-14:
I've edited your post manually to comply to the template :)

meandthemachine, 2022-02-14:
Oh, thank you very much!
... yes, I somehow missed the screenshots heading...

You will see that setup and version are not relevant in this case, but I understand that you can't know that upfront.

meandthemachine, 2022-02-14:
**Off topic:** Regarding the different entry points for opening an issue (and potentially missing the templates) and regarding your bot, maybe you could (or even should?) try setting `blank_issues_enabled` to `false` for your repo, as described in [GitHub Docs Configuring the template chooser](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository#configuring-the-template-chooser), and see how that works out if somebody uses "reference in new issue", like I did.

This kind of bug that I faced [has already been reported to GitHub and been confirmed](https://github.community/t/how-to-use-issue-templates-with-reference-in-new-issue/2359) some years ago, but unfortunately it remains unclear, if or how it has been addressed and resolved.

Cheers!

PR #5806 (2022-03-15): Fixed REST response when setting a schedule's localtime attribute
This should fix #5787. Feel free to edit or let me know if I should change something. I haven't been programming C++ for 20 years. Thanks.
closed 2022-03-15 by github-actions[bot]