dresden-elektronik/deconz-rest-plugin #5792: Jung ZLL 8 gang Wall Swicth is identified as Remote instead of WS_4f_J_1

Kienz, 2022-02-14:
## Describe the bug
The Jung ZLL LS 5004 is identifed as "Remote" instead of "WS_4f_J_1".
My existing Jung 8 gang Wall Switches working fine - had to reset one and after adding it is identified wrong.

## Steps to reproduce the behavior
Pair a Jung ZLL 8 gang Wall Switch - it is displayed as "Remote" in Phoscon and deCONZ GUI.

## Expected behavior
See screenshot 2

## Screenshots
Wrong device:
[image]

Correct device (same type)
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: 2.14.01
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2022-02-14:
Can you show a screenshot of the basic cluster?

Kienz, 2022-02-15:
Correct device:
[image]

Wrong device:
[image]

I added the device with the Phoscon Web app.

Mimiix, 2022-02-15:
They are not the same devices. The Manufacturer name and Model identifier is what the device is programmed for. Meaning: That's what it is. They can look the same, but the chip is different.  This is identifeid properly, in a way that this is what the device says he is.

If the device is working properly, you can ignore it. If it's not, you need to open an device request issue and supply all required screenshots and information there to get it supported. Be adviced of #5733.

Closing this issue as it's not a bug in deCONZ.

Kienz, 2022-02-15:
They are the same device - https://downloads.jung.de/public/de/pdf/bedienungsanleitungen/de_ZLLx5004M_32596233_20190703.pdf

And it worked - I added the device a year ago and it worked like my other Jung ZLLx5004 devices. Yesterday I have to reset this device and now I can't pair it right.

Mimiix, 2022-02-15:
Do they show the same clusters?

Mimiix, 2022-02-15:
Are you sure it worked before? You didn't do a firmware update?

Kienz, 2022-02-15:
Yes it worked - sure. I have 4 of this devices and all of them worked.
Some of them have the firmware 10023202 and some 10033202.
Here are some new screenshots of the Clusters:

Device which doesn't work:
[image]

Device with firmware 10023202:
[image]

Device with firmware 10033202:
[image]

Kienz, 2022-02-15:
Why was this closed?

Kienz, 2022-02-15:
I downgraded to deCONZ 2.10.04 (deconz-2.10.04-qt5.deb) - now the pairing works correct.
Cluster Info:
[image]

[image]

Mimiix, 2022-02-15:
This doesn't make any changes, but now we can classify it as a bug indeed :) Re-opening:)

Kienz, 2022-02-15:
The pairing process works correctly with 2.10.4 and 2.12.6.

Eisenhauer1987, 2022-03-01:
Same Problem here. 4 already paired devices working Fine, the new switch pairs worng and is Not working. Please fix

Eisenhauer1987, 2022-03-14:
Hi, Problem still existing. What can we do for help to fix?

Mimiix, 2022-03-14:
@manup

manup, 2022-03-14:
I've put it on my test list. There is some quirky code to detect the JUNG/GIRA switches (some MAC address magic if I remember correctly) which was likely broken somehow.

closed 2022-03-16 by manup