dresden-elektronik/deconz-rest-plugin #5796: Tuya Smart Thermostat Radiator Valve TS0601

dnldsign, 2022-02-14:
## Device

- Product name: Tuya Smart Thermostat Radiator Valve (https://zigbee.blakadder.com/Tuya_TV02.html)
- Manufacturer: _TZE200_hue3yfsn
- Model identifier: TS0601
- Device type :
  - Thermostat (TRV)

Should be almost identical to device https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4675 (https://zigbee.blakadder.com/Haozee_TRV601.html)

## Screenshots

### Basic Cluster
[image]

### Node info
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### OTAU
[image]

### Tuya cluster
[image]

dnldsign, 2022-03-11:
Switched to zigbee2mqtt

closed 2022-03-11 by dnldsign

danli003, 2022-08-09:
I bought a number of these things. Is it possible to get support in deconz? I'd rather not switch to zigbee2mqtt due to the effort involved.

Asuanin, 2022-09-13:
Same as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5844