dresden-elektronik/deconz-rest-plugin #5798: Gledopto GL-C-002P 5in1 Zigbee Mini Ultra Thin LED Controller

chris-nite, 2022-02-15:
## Device

- Product name: Gledopto GL-C-002P 5in1 Zigbee Mini Ultra Thin LED Controller
- Manufacturer: Gledopto 
- Model identifier: GL-C-002P (currently falsely reported as GL-C-009P after binding)
- Device type : 
  - Light

### Basic
[image]
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

chris-nite, 2022-02-15:
The current status is, that the device properly is binded to the mesh in phoscon, but falsely identified as GL-C-009P. Technically so far it is not used as a repeater (it got Zigbee 3.0 onboard) as well as it doesn't correctly report the power state (is it on or off?) back to home assistant.

Additional info: This led-controller can be used for different kind of led-stripes (rgb/cct/dimming/etc.) through changing the opt-Setting on the controller itself. Based on this it is reported differently back to phoscon as colorextended or dimming light.

Mimiix, 2022-02-15:
The manufacturer name and model Id are presented by the device.  So if that is wrong, it's in the device firmware...

davidrustingha, 2022-02-25:
I have this device and it is working in home-assistant. However; the transition is off. It seems I am not able to set a transition lower than 1 second, so the device is always fading in and out.
Is there a way to change this?

closed 2022-03-26 by github-actions[bot]