dresden-elektronik/deconz-rest-plugin #5799: deCONZ does not restart after reboot device (Pi4 with Openhabian)

WJ4IoT, 2022-02-16:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi4 )
 - Installed = OpenHABian
 - Release Build = 3.2.0 release build
 - Release = Raspbian GNU/Linux 10 (buster)
 - Kernel = Linux 5.10.63-v7l+
 - Platform = Raspberry Pi 4 Model B Rev 1.1
 - Running method: (on same device as Openhabian installed with menu-option '29')
 - Firmware version: (26660700)
 - deCONZ version: (2.14.01 / 06/02/2022)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes and using a USB2.0 port
 - Is there any other USB or serial devices connected to the host system? If so: Which?
 - Yes, FTDI USB Cable for P1 Port Dutch Slimme Meter (DSMR), have tested already without this cable, no difference
 
## deCONZ Logs

[info.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8079213/info.txt)

## Additional context

Mimiix, 2022-02-16:
Hi,

As this is not a bug report as described by #5113 , i am closing this issue. Feel free to open an topic on the forums.

Kind regards,

closed 2022-02-16 by Mimiix

Artens, 2022-08-06:
I'm sorry to report that this issue is actually a bug in the documentation. After three evenings spent on finding how to start a headless server manually, only through this tangential issue description I was able to start the service.

Therefore, the topic is not closed in my opinion, as the documentation should correctly state (with options declared) which command will start the deCONZ service/server standalone from the GUI.

Item | Operational value | Description
command | deCONZ | main application (server?)
-option setting | -platform minimal | Platform startup, my guess is that this is without GUI (no idea what other values are accepted, but it started it on my machine)
--http-port=<port> | --http-port=8081 | Set the port to listen to for the Phoscon app
--ws-port=<port> | --http-port=8082 | Set the port to listen to for the Web Service? (did not get it to work).

Mimiix, 2022-08-07:
> I'm sorry to report that this issue is actually a bug in the documentation. After three evenings spent on finding how to start a headless server manually, only through this tangential issue description I was able to start the service.
> 
> Therefore, the topic is not closed in my opinion, as the documentation should correctly state (with options declared) which command will start the deCONZ service/server standalone from the GUI.
> 
> Item | Operational value | Description command | deCONZ | main application (server?) -option setting | -platform minimal | Platform startup, my guess is that this is without GUI (no idea what other values are accepted, but it started it on my machine) --http-port= | --http-port=8081 | Set the port to listen to for the Phoscon app --ws-port= | --http-port=8082 | Set the port to listen to for the Web Service? (did not get it to work).

You mean the wiki? Feel free to correct it yourself :)

WJ4IoT, 2022-08-11:
If I recall correctly after a half later, the issue was that after a reboot of your Pi (headless) deConz will not restart automatically.
I am not sure this problem will reappear each time when upgrading the firmware.
Important is file:
`sudo nano /lib/systemd/system/deconz.service`
And the line:
`ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8081`
(port at your desire). I have updated file:
`https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually`
Because I cannot check my remarks are still valid nowadays hope others can verify.