dresden-elektronik/deconz-rest-plugin #5803: RTX/Tuya 1 Gang Zigbee Dimmer Module (QS-Zigbee-D02-TRIAC-LN)

Loonmor, 2022-02-16:
## Device

- Product name: RTX 1 Gang Zigbee Dimmer Module QS-Zigbee-D02-TRIAC-LN
- Manufacturer: _TZ3210_ngqk6jia
- Model identifier: TS110E
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch
  - Other: Dimmer

## Screenshots
 
[image]
[image]

### Basic
[image]
### Groups
[image]
### Scenes
[image]
### On/Off
[image]
### Level Control
[image]

mpetro89, 2022-02-21:
Please also support 2 gang device:

Manufacturer Name: _TZ3210_wdexaypg
Model identifier: TS110E

Device also support led_type, minimum & maximum brightness set.
I found this topic:
https://github.com/Koenkk/zigbee2mqtt/issues/9878

My 2 gang device:

[image]

[image]

Basic:
[image]

ON/OFF:
[image]

Level Control:
[image]

spekkie3, 2022-03-01:
I get the : unsupported issue in mqtt 

I use mqtt addon with home assistant 

Can someone please help ?

closed 2022-03-30 by github-actions[bot]

SkyByte, 2022-04-03:
See issue #9878