dresden-elektronik/deconz-rest-plugin #5805: Bitron Video 902010/24A alarm not working

kayhannay, 2022-02-16:
## Bug description

I have a Bitron Video 902010/24A smoke detector, which is working fine with deconz so far. But I just tried to trigger the alarm, which should be possible according to #2886 and #3211. But in my case, nothing happens.

## Steps to reproduce the behavior

1. Run `curl -X PUT 'http://deconz/api/xyz/lights/5/state' --data '{"alert": "lselect"}'`
2. See the result `[{"success":{"/lights/5/state/alert":"lselect"}}]`
3. Listen to the Bitron detector, there is no sound

## Expected behavior

There should be an alarm sound.

## Screenshots

>{
  "etag": "7e7...",
  "hascolor": false,
  "lastannounced": "2021-01-19T08:20:52Z",
  "lastseen": "2022-02-16T15:48Z",
  "manufacturername": "Bitron Video",
  "modelid": "902010/24A",
  "name": "Warning device 5",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": "20170421",
  "type": "Warning device",
  "uniqueid": "00:12:4b:00:16:bb:42:09-01"
}

## Environment

 - Host system: PC
 - Running method: Ubuntu
 - Firmware version: 26720700
 - deCONZ version: 2.14.01 / 6.2.2022
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes USV

## deCONZ Logs

> 21:57:28:036 [INFO] - No button map for: SP 120, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05293000, zclSeq: 38
21:57:28:036 ZCL attribute report 0x00158D0003884B86 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:57:28:036 	payload: 0b05293000
21:57:28:037 Websocket 192.168.1.1:56368 send message: {"e":"changed","id":"10","r":"sensors","state":{"current":322,"lastupdated":"2022-02-16T20:57:28.035","power":48,"voltage":230},"t":"event","uniqueid":"00:15:8d:00:03:88:4b:86-01-0b04"} (ret = 185)
21:57:28:037 Websocket 192.168.1.2:36976 send message: {"e":"changed","id":"10","r":"sensors","state":{"current":322,"lastupdated":"2022-02-16T20:57:28.035","power":48,"voltage":230},"t":"event","uniqueid":"00:15:8d:00:03:88:4b:86-01-0b04"} (ret = 185)
21:57:28:545 poll node 00:12:4b:00:16:bb:42:09-01
21:57:28:895 add task 2104 type 34 to 0x00124B0016BB4209 cluster 0x0502 req.id 241
21:57:29:039 Node data 0x00158d0003884b86 profileId: 0x0104, clusterId: 0x0B04
21:57:29:040 0x00158D0003884B86: update ZCL value 0x01/0x0B04/0x050B after 0 s
21:57:29:040 [INFO] - No button map for: SP 120, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05292C00, zclSeq: 45
21:57:29:040 ZCL attribute report 0x00158D0003884B86 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:57:29:040 	payload: 0b05292c00
21:57:29:041 Websocket 192.168.1.1:56368 send message: {"e":"changed","id":"10","r":"sensors","state":{"current":322,"lastupdated":"2022-02-16T20:57:29.040","power":44,"voltage":230},"t":"event","uniqueid":"00:15:8d:00:03:88:4b:86-01-0b04"} (ret = 185)
21:57:29:041 Websocket 192.168.1.2:36976 send message: {"e":"changed","id":"10","r":"sensors","state":{"current":322,"lastupdated":"2022-02-16T20:57:29.040","power":44,"voltage":230},"t":"event","uniqueid":"00:15:8d:00:03:88:4b:86-01-0b04"} (ret = 185)
21:57:29:121 poll node 7c:b0:3e:aa:0a:08:63:51-03
21:57:29:121 Poll light node On/Off plug-in unit 6
21:57:29:173 Poll APS request to 0x7CB03EAA0A086351 cluster: 0x0006 dropped, values are fresh enough
21:57:29:647 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:57:30:078 poll node 7c:b0:3e:aa:0a:08:80:75-03
21:57:30:078 Poll light node On/Off plug-in unit 3
21:57:30:131 Poll APS request to 0x7CB03EAA0A088075 cluster: 0x0006 dropped, values are fresh enough
21:57:30:513 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x72AD

SwoopX, 2022-02-17:
Your API call is wrong. You have to add `duration` with an integer value between 1 and 300. I have to admit it seems to be missing in the API documentation though.

closed 2022-02-17 by SwoopX

kayhannay, 2022-02-17:
When I add this parameter I get an error:

`curl -X PUT 'http://deconz.hannay.de/api/58761B1520/lights/5/state' --data '{"alert": "select", "duration": 1}'`

`[
  {
    "error": {
      "address": "/lights/5/state",
      "description": "parameter, duration, not available",
      "type": 6
    }
  }
]`

Is there something wrong with my request?

SwoopX, 2022-02-17:
Indeed, you've used `select` this time instead of `lselect` 🙂

kayhannay, 2022-02-17:
With `duration` I get the error with `select` and `lselect`. This does not work here.
With `ontime` I get success, but still no sound or any other signal from the smoke detector.

SwoopX, 2022-02-17:
> `duration`? Isn't that `ontime`?

Indeed. Has the naming changed with your last refactor? Would have sworn it was duration, but haven't used it for ages.

ebaauw, 2022-02-17:
No, it's been `ontime` since I introduced it for the Heiman Siren, a couple of years ago.  `duration` is a `config` attribute for ZHAPresence and ZHAVibration.  I think `lselect` should work with a default `ontime` (of 300s?) when you don't specify any.  Not something my ears appreciate testing, though (nor my neighbours).

Sending commands to sleepy devices is somewhat unreliable; there's always a change that the message gets lost, if the device doesn't poll its parent in time.  Could also be that deCONZ won't even send the command, when it thinks the device is asleep (and it won't see the polls unless it's the parent).  That's why (properly implemented) sleepy sensors use `config.pending`, with relatively frequent (5 min) attribute reporting, to make sure a changed `config` setting reaches the device.

I find setting the siren on my Develco smoke sensors usually works, but with a varying delay.  The mains-powered Heiman siren was a lot more reliable and responsive.  That is, until its power transformer broke on me (which I only noticed, after its backup battery depleted).

Anyways, best setup some `ddx` rules to (try and) activate the siren a few times in a row in case a real alarm happens.  Of course, not an issue when the sensor itself detects the alarm and activates the siren internally.