dresden-elektronik/deconz-rest-plugin #5807: Aqara Smart Wall Switch H1 (With Neutral, Single Rocker/ Double Rocker)

JPOliveGit, 2022-02-17:
## Describe the bug
trying to connect the Aqara Smart Wall Switch H1 (With Neutral, Single Rocker/ Double Rocker), but it cannot connect, I'm using phoscon in home assistant with conbee II, I could connect the Aqara Smart Wall Switch H1 wireless but not the wired(neutral).

## Steps to reproduce the behavior
1. enter phoscon in HA
2. menu
3. switches
4. add new switch
5. when phoscon is waiting i put the switch in pairing mode but phoscon can detect the switch

## Expected behavior
should connect to the switch like it did on the wireless version

## Screenshots

## Environment

Phoscon:
- Vendor: Dresden Elektronik
- Product: ConBee II
- Version: 2.14.01 / 06/02/2022
- Firmware: 26660700

Deconz:
- Version: 2.14.01

 - Host system: Raspberry Pi
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.14.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no) -- only relevant for ConBee I/II

## deCONZ Logs

## Additional context

Mimiix, 2022-02-17:
@SwoopX cna you check here?

@JPOliveGit what does deconz show?

SwoopX, 2022-02-17:
I'm about to close this one as that very device was the basis for the last larger PR I did on Xiaomi switches. I have it here and paired it a gazillion times.

Btw, I can recommend a firmware update, as it already is quite old.

Mimiix, 2022-02-17:
If you have the exact same one working, I'm closing as this issue doesn't comply with #5113.

@JPOliveGit feel free to open an topic on the forum.

closed 2022-02-17 by Mimiix

JPOliveGit, 2022-02-18:
ok guys... I realized that it was working on deconz although phoscon couldn't see it, HA can see it, but still it doesn't appear on phoscon... this guys