dresden-elektronik/deconz-rest-plugin #5808: GLEDOPTO Switch Module GL-SD-002 not shown in Phoscon App

bruesche, 2022-02-18:
I try to integrate the GLEDOPTO Switch Module GL-SD-002 in Phoscon. It didn't appear in the app after pairing it. In the deconz-gui I can see a new node with following infos:

Manufacturer Name GLEDOPTO

Model Identifier GL-SD-002

Date Code 20210707

SW Build ID 10176801

What can I do?
Please advice.

Thx

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - deCONZ version: (2.12.06 & 2.14.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? USB Stick

## deCONZ Logs

## Additional context

Mimiix, 2022-02-18:
Can you share a screenshot of how it looks in deconz?

bruesche, 2022-02-18:
Here are the Screens:
[image]
[image]
[image]