dresden-elektronik/deconz-rest-plugin #5815: Aquara Smart Knob Switch H1

Adam-Langley, 2022-02-20:
## Device

- Product name: Aquara Smart Knob Switch H1 (wireless version).
- Manufacturer: LUMI
- Model identifier: lumi.remote.rkba01
- SKU: ZNXNKG02LM
- Device type : Please remove all unrelated device types. 
  - Basic
  - Identify
  - Power Configuration
  - Lumi Specific
  - Multistate Input (Basic)
  - Identify

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Power Configuration
[image]

### Lumi Specific
[image]
[image]
[image]

### Multistate Input (Basic)
[image]

### OTAU
[image]

[image]

Adam-Langley, 2022-03-13:
Ping - still relevant

theboxer, 2023-01-16:
@Adam-Langley does this switch work with deconz/phoscon? I'm about to purchase one...

theboxer, 2023-01-16:
@Adam-Langley I'm actually not using home assistant, but a homebridge and exposing it to the home app (apple). By z2m you mean this: https://www.zigbee2mqtt.io/?

theboxer, 2023-01-16:
@Adam-Langley I have Rpi4 + Conbee2. I'm just configuring the docker container and going to try it :) Thank you for the hint! btw. I'm on deCONZ's discord if you wanna chat more about the setup.

Mimiix, 2023-01-16:
> Hi @theboxer - yes that's right.
> 
> When you first look into it, you'll think - "this is a lot of different components, for what I used to do with 1, or 2... too much hassle?" I urge you to give it a try, and you'll see, once you've installed it, you're done. It works like a dream. HomeAssistant ("HA") releases updates (applied automatically if you like), almost every few days - so, they're so incredibly active, it's awesome. Just follow the instructions. You'll be installing
> 
> 1. Home Assistant (the environment)
> 2. Zigbee2Mqttt (the add on which talks to your zigbee hardware, and then passes the zigbee messages to/from the next component...)
> 3. An MQTT broker - probably just the built-in one called "Mosquitto". MQTT is a messaging system - think of it like a highway system, it connects things which publish messages, to things which want to listen to messages... It's how your zigbee events will eventually find their way into HomeAssistant
> 
> Looking at the list above... where's the Phoscon components?? Yep - they're gone. No need for them. Zigbee2Mqtt is compatible with the dresden devices.
> 
> Can I ask, what is your installation platform? For example, I'm doing everything above with a Pi4 + RaspBee II. But you could equally be running on a workstation+Conbee or such...

Hey guys, I'm all for helping each other, but we are going a bit off topic now. May i suggest you reach out to each other through discord or forums?

Adam-Langley, 2023-01-16:
Hey @Mimiix - sure thing no prob - was just about to do that just didnt want to jump the gun.
@theboxer sure thing - I'll look for you now.

Smanar, 2023-01-17:
> In fact, an unrelated thing - but huge reason to move… In Phoscon, if a connection to a device ever failed, and the only way to get it to respond is to re/add it, you lose everything related to that device. Z2m is HEAPS better here. got an unresponsive device? Hit the “add device” button in Z2m, force the device into discovery mode, and the connection is healed… no settings lost.

It's same in phoscon, you just need to DON'T delete the device, and make the same procedure, all will be invisible.

saxn-paule, 2023-03-11:
I bought two of those devices but I'm not able to pair them via Phoscon. What has to be done to make it work?

xADDRx, 2023-03-31:
HI,
any Solution here? Have the Same Problem

xADDRx, 2023-06-27:
Has anyone got it working yet?

sadpixel, 2023-11-21:
I also have this device and would love for it to work with DeCONZ. Happy to provide any information that may be needed, though it looks like the OP already added the required screenshots...

sadpixel, 2023-11-21:
Dear @Mimiix, can we please get this re-opened? I'll try to put together a test DDF based on the one for the standard Aqara H1-series wireless switch (lumi.remote.b28ac1 aka WRS-R02), but I'll probably need some help with the Multistage Input cluster (0012)...

Mimiix, 2023-11-21:
Happy to re open. @smanar would this be possible with DDF ?

Smanar, 2023-11-22:
Honnestly IDK, this device need lot of work.
- It seem it need a special request on inclusion.
- The Lumi specific cluster give a payload with angle/vitesse/speed, like for all rotating switch, not easy to convert in X00Y event. Need to use special field.

The problem will be not the 0x0012 cluster, on this one you will just have some value according to action.

Can make try, make a basic DDF, without buttonmap part. use the device and look at return on logs to make the buttonmap file.

And I m not sure the "special request" can be automatic, on other Xiaomi device, it's just an option.

```
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl:attr",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl:attr",
            "mf": "0x115f",
            "state.timeout": 2
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
```

closed 2023-12-22 by github-actions[bot]