dresden-elektronik/deconz-rest-plugin #5819: _TZ300_mq9f17dw - TS0203 - Tuya Smart Zigbee Door Window Contact Sensor ZigBee 3.0

Telk0, 2022-02-21:
## Device

- Product name: Tuya Smart Zigbee Door Window Contact Sensor ZigBee 3.0
- Manufacturer: _TZ300_mq9f17dw
- Model identifier: TS0203
- Device type : Sensor
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots
[image]
[image]

### Basic
[image]

### Identify
[image]

### Alarms
Non

### Device Temperature
Non

### Groups
Non

### Scenes
Non

### On/Off
Non

### Level Control
Non

### Color Control
Non

### Simple Metering
Non

### Diagnostics
Non

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

Telk0, 2022-02-21:
@Smanar 
Hello Smanar, This is the Door WIndow i have. Only diffrence i see is manufacture compared to: 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5301#issuecomment-1046035105

But it wont find it. Can you help please.
Thank you

Smanar, 2022-02-21:
So first bad news for you, not possible adding new device using code ATM ^^.
Then good news your device relay look standard, you you know DDF ? I think you can add it without code edition.

It's a door sensor ? like this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/wiser/contact_sensor_cct591011_as.json ?

You can start from scratch https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Or just dit the previous one, just need to edit thoses lines according to your device

```
  "manufacturername": "_TZ300_mq9f17dw",
  "modelid": "TS0203",
  "vendor": "Tuya",
  "product": "Contact sensor blablabla",
```

Telk0, 2022-02-22:
Hello @Smanar 

I do not have experience with DDF :)

Its a door sensor similar to the one i linked to above.
Will the wiser DDF work for that? changing the 4 items you mention? or is it posible to withdraw the DDF for https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5301#issuecomment-1046035105 without having the actual device.

Smanar, 2022-02-22:
The device in #5301 use the legacy code, it's no more possible. The priority is gived to the DDF.

And yes, creating a DDF with just editing thoses 4 lines can be enought, just use a new file name.
Its exactlky the same device, with same cluster, you have it here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4816

It's a door sensor that use the IAS cluster, the only difference can be the enrollement

Telk0, 2022-03-02:
@Smanar Is it possible to set deconz mode to Hybrid & Silver+Gold default. So it dosent reset on reboot.

closed 2022-03-31 by github-actions[bot]