dresden-elektronik/deconz-rest-plugin #5820: GLEDOPTO GL-SD-002 Switch Modul

bruesche, 2022-02-21:
## Device

- Product name: GLEDOPTO Zigbee Mini Switch Modul
- Manufacturer: GLEDOPTO
- Model identifier: GL-SD-002
- Device type : Switch
 

## Screenshots

[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-02-21:
Missing screenshots.

bruesche, 2022-02-21:
Sorry for that. I will take the screenshots. But I can do that in the next 14 days.. Sorry for that.

bruesche, 2022-03-04:
so, back again. Hope I get all the screenshots. I try:
[image]
[image]
[image]
[image]
[image]

bruesche, 2022-03-04:
Please give me advice if something is missing. Thx :)

closed 2022-03-09 by bruesche