dresden-elektronik/deconz-rest-plugin #5823: Tuya Zigbee 3.0 PIR Motion Sensor; _TZ3000_msl6wxk9; TS0202

J3susC, 2022-02-22:
## Device

- Product name: ZM-35ZH-Q
- Manufacturer: _TZ3000_msl6wxk9
- Model identifier: TS0202
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

[image]
[image]

### Basic

[image]
[image]

### Identify
[image]
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
Power
[image]
[image]

IAS
[image]
[image]

Mimiix, 2022-02-22:
Hi,

There's already a request for this device:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5584

Closing this one.

closed 2022-02-22 by Mimiix

zafnz, 2022-04-15:
Can confirm that the provided settings (almost) work, I had to fix some json (missing comma). My device is basically the same sensor, but a slightly different ID. (_TZ3000_otvn3lne)

Otherwise everything I'd expect to work, works.

[otvn3lne.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8492980/otvn3lne.txt)