dresden-elektronik/deconz-rest-plugin #5824: False smoke alarm with Heiman smoke detector

cosmo84, 2022-02-22:
## Describe the bug

I have several Heiman Smoke detectors in my network. I struggle with the issue that once every 1-2 weeks I receive a false alarm, saying that smoke was detected. The sensor itself is not beeping, so it is only a "virtual" alert sent to Homebridge. Quite scary when you are not at home and as it happened the first time ;-)

What I see in the GUI in the IAS ZONE cluster, that Alarm 1 is checked after it happens.

[image]

I can reset it by pressing the test alarm button and in the same time read the Zone Status

[image]

I have in total 13 of these smoke detectors in my network and so far this issue occured with 3 of them and not only a single one. So for the moment I exclude an issue with the sensor itself.

## Steps to reproduce the behavior

Cannot be reproduced manually, but happens regular every 1-2 weeks after a reset (see above)

## Expected behavior

No smoke detection when there is no smoke.

## Screenshots

See above, please let me know additional need for screenshots.

## Environment

[image]

 - Host system: NAS
 - Running method: Deconzcommunity Docker container
 - Firmware version: see screenshot
 - deCONZ version: see screenshot
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No.

## deCONZ Logs

Please let me know which logs you need, I am right now not brave enough to activate debug logging for a period of weeks.

Thank you very much.

ebaauw, 2022-02-22:
When _Alarm 1_ is set, deCONZ has received a report from the sensor that smoke was detected.  Since this doesn’t happen on all sensors, I doubt it’s a software issue in deCONZ.

I think these sensors use an optical sensor to detect smoke.  Did you double-check the sensors for dust or debris?  Where are the sensors located?  In a kitchen, bathroom, garage, or some place where there’s steam or dust in the air?

Did you double-check the batteries of the affected sensors?  By other means that looking at the value reported through deCONZ?  Does the issue persist when you power-cycle the sensors (remove the battery for 10s)?

cosmo84, 2022-02-22:
The sensors are all new and it started immediately after their installation. 

If smoke is detected, why is it only transmitted to deCONZ and no alarm is ringing? 

I must admit that it happens mostly in the garage, but it is big and good ventilated. It happened as well in the living room. 

Since they are all new, battery level is good and also reported good in the GUI

ebaauw, 2022-02-22:
> If smoke is detected, why is it only transmitted to deCONZ and no alarm is ringing?

I'm not saying they detect smoke, I'm saying they seem to report smoke detected.  You did test these sensors and confirm they make noise?  Almost depleted battery?

> I must admit that it happens mostly in the garage, but it is big and good ventilated. It happened as well in the living room.

How cold and humid were these rooms?  Could there be condensation on the optical sensor?  I would expect the sensor to make noise in these cases, though.

> Since they are all new, battery level is good and also reported good in the GUI

Could you humour me and measure the battery level or voltage?  Or swap the battery of one of the sensors reporting false positives with that of one that behaves well?

If that doesn't change the behaviour, enable in the debug view: ERROR, ERROR_L2, APS, APS_L2, and look for `APS-Data.Indication` messages from just before the alarm goes off, with `srcAddr` matching the NWK address from the sensor, `profile`: 0x0104, `cluster`: 0x0500.  We need those lines, and the following `asdu` line.  Also check for any error messages.

cosmo84, 2022-02-23:
When I press the test button they beep as expected, so I can't identify a general misbehavior so far. 
I can't measure the battery, but I will change it with the one of another sensor that had no issue so far. If that moves the false reporting to the other sensor, your theory about depleted battery seems most reasonable.  
If the issue stays on the same device, I will switch the full device, not only the battery. 

I hope these tests help to limit the possible root causes. 

Still it is strange to me, that a sensor is reporting smoke towards deCONZ, but stays completely calm, meaning no beep alarm. Let's see...

Thanks for your help and advices so far, I will update as soon as I have new information.

cosmo84, 2022-03-03:
It happened again, with the same sensor that has now a battery of a sensor without issues so far. I had the debug logging enabled and found these lines:

20:00:53:553 APS-DATA.indication srcAddr: 0xa17f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -39
20:00:53:553 	asdu: 090900200000640000
20:00:53:554 APS-DATA.request id: 166, addrmode: 0x02, addr: 0xa17f, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x04
20:00:53:554 	asdu (length: 5): 10090b0000
20:00:53:562 APS-DATA.indication srcAddr: 0xa17f, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -39
20:00:53:562 	asdu: 090a00210000640000
20:00:53:563 APS-DATA.request id: 168, addrmode: 0x02, addr: 0xa17f, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
20:00:53:563 	asdu (length: 5): 100a0b0000
20:00:53:574 APS-DATA.confirm id: 166, status: 0x00 SUCCESS
20:00:53:575 APS-DATA.confirm request id: 166 -> erase from queue
20:00:53:585 APS-DATA.confirm id: 168, status: 0x00 SUCCESS
20:00:53:585 APS-DATA.confirm request id: 168 -> erase from queue
20:00:53:635 aps request id: 166 finished, erase from queue
20:00:53:715 aps request id: 168 finished, erase from queue

SwoopX, 2022-03-03:
I'm afraid ebaauw was right while assuming this is not a deconz issue, as data is pretty clear on that: your device is misbehaving.

The marked 2 bytes hold the current IAS zone status 090a00**2100**00640000, which is in your case 0x0021, translating to the following bit represenation b0000000000100001. The lowest bit (most right hand) is set to 1, indicating alarm status is true, so the device falsly detects fire/smoke.

Seen other oddities around the IAS zone status before from some devices, but not in that significance.

Sjuulh, 2022-03-21:
Have the same issue on (at least) 3 of 5 heiman smoke detectors. HA gets the message there is smoke detected, but the device isn't making any noise. When I test it with smoke it performs as expected. Is there any fix available?

Mimiix, 2022-03-21:
@Sjuulh Other than getting new firmware from the manufacturer, no. This is a bug in the device firmware as it seems, so there's nothing deconz can do.

cosmo84, 2022-04-07:
Just a quick update from my side. Two weeks ago I took each smoke detector in my hands, made sure it is awake and read the IAS Zone (cluster 500) attributes. The zone status of all sensors was read with 0x0030. 
Since then I had no more false alarm. I am not an expert here, but maybe for some sensors the cluster attributes were not read correctly during the pairing and that might have caused the issue?

I will keep monitoring for 1-2 weeks. If it stays good, it will be a month without false alarm, never had before. Then we can assume the issue fixed.

cosmo84, 2022-04-29:
On my side it looks good, I had no more issues since then.

cosmo84, 2022-05-10:
Still no issues, for me the issue closed.

closed 2022-05-10 by cosmo84

georgbachmann, 2022-06-14:
@cosmo84 as I have the exact same issue as you describe with the same smoke detectors, could you maybe tell me how to read that IAS Zone. I am a zigbee and deconz noob, so I don't know where to do that :)
@nonitex is the issue resolved for you as well?
Thanks for the help!

tpjanssen, 2023-01-11:
@cosmo84 @nonitex @georgbachmann 

Can you guys please give an update on this issue? I'm experiencing the same issue.

Thank you.

cosmo84, 2023-01-11:
Hi,

for me the issue is more or less solved based on the procedure I shared earlier in this issue. 

Only one smoke sensor is every now and then reporting smoke, but it is in my garage and it might be related to lower temperature and higher humidity, just my personal assumption. I never checked the datasheet of the sensors so far if there is any limitation or recommendation about this. 
In all rooms of my home the issue never appeared again. Sorry, I can't help more than this I think. 

Best regards,

Stephan

georgbachmann, 2023-01-16:
For me the procedure didn't really work. I'd say it got less false alarms, but still...
I switched to zigbee2mqtt though a couple of days ago. Will need to see it the issue persists then...

Thanh-Binh, 2023-07-16:
Hello
I have also short false alarm every day with Heiman HS1SA-E: each of 3 smoke detectors beeps shortly onetime per day.

I really do not understand if those sensors are defect or is there any problem with the zigbee communication?

Can anyone help me? Thanks

georgbachmann, 2023-07-17:
I switched to zigbee2mqtt and never had a problem ever since. Switching is not super fun cause you need to re-pair devices, but once that's done you will have a far better experience :-)

Thanh-Binh, 2023-07-17:
@georgbachmann I really do not understand why it helps us? 
False alarm beeps come from sensors and I believe it is one way data from sensor to HA via deConZ. If your solution is right, the communication between sensor and deConZ Zigbee causes short alarm.

Couls you pls explain me a little more ?
Thanks

georgbachmann, 2023-07-17:
I don't know exactly what caused the false alarms. But ever since I switched to zigbee2mqtt I don't have them any more. But when I e.g. press the button on the fire-alarm it shows up in HA. So communication as such is still working. But without false alarms. I also switched my hardware to be fair from the Deconz-Stick to a Sonoff zigbee stick. So maybe that alone also helped.
So basically I can only tell you what I've experienced. No idea why/how it works now. But it does :)

Thanh-Binh, 2023-07-17:
@georgbachmann 
thanks for your infomation.
If I understood you well, the problem is solved after using Sonoff instead of deConZ.
Is it correct?

georgbachmann, 2023-07-17:
I don't know if it was the software or the hardware... I switched both and now everything works fine.
And in my personal opinion zigbee2mqtt is  much more intuitive, it offers over the air updates and so on...

Mimiix, 2023-07-17:
> I don't know if it was the software or the hardware... I switched both and now everything works fine. And in my personal opinion zigbee2mqtt is much more intuitive, it offers over the air updates and so on...

deCONZ also offers OTA updates ;)

Nevertheless, please keep it on-topic guys.

Thanh-Binh, 2023-07-17:
@georgbachmann 
one question for making sure that you have the same problem as me:
Does your sensor often generate a short beep before? or
Does HA send you often false alarm before?
Thanks

Thanh-Binh, 2023-07-18:
@klada thanks.
Currently I disconnected deConZ and will see if all sensors alone provide false alarm.

Thanh-Binh, 2023-07-20:
@georgbachmann @klada 
I had 2 days experiment with my sensor without deConZ, and did NOT have any more false alarm.
After plug-in again ConBee USB stick into HA, I have this problem.

Here is my conclusion:
The ZigBee communication between sensor and ConBee seems to cause false alarms
In combination with the experiment results of @georgbachmann with other ZigBee coordinator than ConBee, I believe that will be a bug with deConZ!!!

Any support from Dresden Electronik for solving that problem is helpful! Thanks

SwoopX, 2023-08-05:
> I believe that will be a bug with deConZ!!!

I consider it proven here (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5824#issuecomment-1057866782) that the device is the root cause. The ASDU is sent by the device and cannot be altered by deconz.