dresden-elektronik/deconz-rest-plugin #5826: Philips Hue Motion Sensor works even {"on": false}

marc965, 2022-02-22:
## Describe the bug

I tried one Philips Hue Indoor Motion Sensor with deCONZ integration and it works well except when setting {"on": false} with

deconz.configure: { "field": "/sensors/33/config", "data": {"on": false} }
the Motion Sensor is still working, but i want to disable it.

If i look on deCONZ Api /sensors/33/config is set correctly on: false
Trying with:

platform: command_line
switches:
bad_motion:
friendly_name: 'BadMotion'
command_on: curl -X PUT -d '{"on":true}' "http://192.168.133.41:40850/api/XXXXXXX/sensors/33/config"
command_off: curl -X PUT -d '{"on":false}' "http://192.168.133.41:40850/api/XXXXXXX/sensors/33/config"
command_state: 'curl http://192.168.133.41:40850/api/XXXXXXXX/sensors/33'
value_template: '{{ value_json.config.on }}'
is working too, but no change on sensors behavior.

With Philips Hue Integration this is working properly.

## Steps to reproduce the behavior

## Expected behavior

If {"on": false} the motion sensor had to do nothing.

## Screenshots

[image]
[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
   NAS: Synology VMM, Home Assistant core-2022.2.9, supervisor-2022.01.1, Home Assistant OS 7.4 

 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
   Home Assistant deCONZ Add-on 6.12.0
 
 - Firmware version: (26xxyy00)
   266B0700

 - deCONZ version: (2.xx.yy)
   2.14.01 / 6.2.2022

 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
   Conbee II

 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
   yes

 - Is there any other USB or serial devices connected to the host system? If so: Which?
   no

## deCONZ Logs

## Additional context

SwoopX, 2022-02-22:
This setting is just to disable event processing within rules, not to discard any events from the device itself.

closed 2022-02-22 by SwoopX

marc965, 2022-02-23:
> This setting is just to disable event processing within rules, not to discard any events from the device itself.

Hi SwoopX. I dont agree. With my Philips Hue i have directly effect on Sensor behaviour with this param.
What can i do to persuade you thats not only a disabling processing parameter?

Within the Home Assistant Philips Hue integration this parameter is used to directly disable and enable Sensor as switch entity.
[image]

ebaauw, 2022-02-23:
See #90.

marc965, 2022-02-23:
thx for helping.
can you tell me if this "issue" is considered?
any information when it could be fixed?

not entirely understand...