dresden-elektronik/deconz-rest-plugin #5831: Device name

Philap95, 2022-02-24:
## Device

- Product name: Nedis Zigbee Radiator Thermostat
- Manufacturer: _TZE200_hhrtiq0x
- Model identifier: TS0601
- Device type : 
  - Thermostat

Already mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5495#issuecomment-974677504

### Cover
[image]

## Screenshots

### Nodeinfo
[image]

### Basic
[image]

### Groups
[image]

[image]

### Scenes
[image]

[image]

[image]

### Tuya specific Cluster
[image]

### OTAU
[image]

### Time
- nothing

scobby, 2022-02-26:
Have two of them here, the thermostat was on sold on mydealz, think currently there will be many people who bought them

Philap95, 2022-02-26:
@scobby That is correct.
I ordered 6 pieces directly.
NBB has sold them for 25€ each.

fronsky, 2022-03-03:
is alternative to use zigbee2mqtt?

fronsky, 2022-03-03:
> But bad news for you, the DDF core don't support yet tuya cluster, and all other code addition are locked, so no way to include this device yet.

What do you mean with "DDF core"?

Philap95, 2022-03-03:
> is alternative to use zigbee2mqtt?
> 
> 

For now I switched to Home Assistant with ZHA, zigbee2mqtt should also work, see: https://zigbee.blakadder.com/Nedis_ZBHTR10WT.html

fronsky, 2022-03-04:
> > is alternative to use zigbee2mqtt?
> 
> For now I switched to Home Assistant with ZHA, zigbee2mqtt should also work, see: https://zigbee.blakadder.com/Nedis_ZBHTR10WT.html

Thanks for yr advise. I'll give it a try.

TheHaf, 2022-03-31:
I also ordered two of these a few days ago as there was another sale. I hope they can soon be supported. Anything else that can be done on the user side to help integration?

Smanar, 2022-03-31:
The DDF core can now support Tuya cluster, but I have never do that that yet, and not sure it will work but same than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5844#issuecomment-1084766582

TheHaf, 2022-03-31:
If I am not mistaken, this would be done by the deconz GUI, correct?
I have my Smart Home stuff running on a raspberry. Can I just unplug the Conbee II stick from there and plug it into my PC or is there a risk of breaking anything if I don't shutdown the deconz service first?

Smanar, 2022-03-31:
Can be done with text editor too. Can make a try direclty.

If you unplug a conbee, deconz will just stop, and you have a backup feature in phoscon if you are worried.

Try with this DDF, but same, having logs on headless device is harder, hoping for the result will be visible in the API

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_hhrtiq0x",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/locked"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve"
        },
        {
          "name": "state/windowopen"
        }
      ]
    }
  ]
}
```

It's totaly experimental, but I m trying to have the temperature and the setpoint

TheHaf, 2022-04-01:
I should have read your message first before trying anything, it looks like I broke something. 😫

I went with the route of plugging the stick into the PC. For some reason, it showed only four nodes there in the deconz GUI instead of the registered 50 nodes. So I didn't do anything and plugged it back into the Raspberry. Almost everything seems normal, but one of the lights won't let me change it's brightness and state, it will always turn back on and to a certain level, regardless of whether I change the level or turn it off in Phoscon or via a physical dimmer. In the API view I can see corresponding events that get somehow triggered for that light, changing it back to on and with that certain brightness level.
Could it be that plugging the stick into the PC (re)activated an older network setting that now fights with the real network settings? It might be that one of the four nodes was this very same light.
I tried restarting deconz but to no avail. I wonder whether I should replug the stick to the PC and delete the nodes there or maybe even click the "leave network" button there as it seems to be a different network. But I am afraid to break even more.

I'll need to fix this before I can continue with this thermostat...

Smanar, 2022-04-01:
> For some reason, it showed only four nodes there in the deconz GUI instead of the registered 50 nodes

Yep this is normal, the network config is on the gateway, but the devices configuration is on the host, if you want to have a complete zigbee network you need to backup on the old machine, and restore it on the new one.

But if you try only with text editor, you can use the headless machine, at least for the moment.

TheHaf, 2022-04-01:
Thank you for the explanation! That certainly clears it up for me.

I just tried using the json you provided after updating to the current beta release. I put the json contents into `~/.local/share/dresden-elektronik/deCONZ/devices/nedis-thermostat.json` and restarted the deconz service.
And lo and behold, the thermostat appears as a sensor:
[image]

And here is the API information:
```
{
    "config": {
        "battery": null,
        "heatsetpoint": null,
        "locked": null,
        "on": true,
        "reachable": true
    },
    "etag": "e2fa0b60601bd684413e6004e8a44318",
    "lastannounced": null,
    "lastseen": "2022-04-01T15:47Z",
    "manufacturername": "_TZE200_hhrtiq0x",
    "modelid": "TS0601",
    "name": "Thermostat 50",
    "state": {
        "lastupdated": "none",
        "temperature": 0,
        "valve": null,
        "windowopen": "Quarantine"
    },
    "type": "ZHAThermostat",
    "uniqueid": "50:32:5f:ff:fe:9f:36:c0-01-ef00"
}
```

As you can see, the state information is not correct, but I haven't installed the thermostat yet on the radiator, it still sits on my desk. Maybe that is the reason for the wrong state information? Or maybe it has something to do with the fact that I tried to add it to the Zigbee network when I was still on the old version of deconz.

Smanar, 2022-04-01:
No, I don't think, we just need to be on the last for test.
The first part is done, deconz can see the device, the 0 value can be a convertion error 

Can you tried to change the setpoint using manually the device ?

Perhaps we miss that

> The read function is special and must only be specified on one item (here state/on). It doesn't actually read a single value but triggers the Tuya device to report all datapoints.

Try to add it in the setheatpoint

```
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"},
        },
```

For the dpid I m almost sure it's the good one, if I m right your device use 0x02 0x03 and 0x02 0x03.

Unfortunately, the only way to reload a DDF on headless deconz, is restarting deconz ....

TheHaf, 2022-04-01:
Indeed, when I turn the thermostat, the `heatsetpoint` updates accordingly: `"heatsetpoint": 1900,`. I tried locking the thermostat, but the value for `locked` didn't change unfortunately.

I made the change from `none`  to `tuya` in the read config of heatsetpoint and now `lastupdated` and `temperature` also report:
```
{
    "config": {
        "battery": null,
        "heatsetpoint": 2200,
        "locked": null,
        "on": true,
        "reachable": true
    },
    "etag": "22a92423034f3a1da3f6bd1a064e3d3c",
    "lastannounced": null,
    "lastseen": "2022-04-01T16:37Z",
    "manufacturername": "_TZE200_hhrtiq0x",
    "modelid": "TS0601",
    "name": "Thermostat 50",
    "state": {
        "lastupdated": "2022-04-01T16:32:51.324",
        "temperature": 1960,
        "valve": null,
        "windowopen": "Quarantine"
    },
    "type": "ZHAThermostat",
    "uniqueid": "50:32:5f:ff:fe:9f:36:c0-01-ef00"
}
```
So we are slowly making progress. 😸 

Don't worry about the debugging process, editing the DDF and restarting deconz on headless is no problem for me, I just use two SSH shells and can iterate quickly. 😺

Smanar, 2022-04-01:
Nice ^^, was easier than I thought

To send the request we are using

>"write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},

The dpid is good, it s the same than for the read, the convertion too (it work for read with * 10), so the problem is probably dt.

```
          bool           0x10
          s32 value      0x2b
          enum           0x30
          8-bit bitmap   0x18
          16-bit bitmap  0x19
          32-bit bitmap  0x1b
```

But for me it's good too ...

Smanar, 2022-04-01:
We can try for another one for test, the childlock, on this device it s 0x01 0x07

```
        {
          "name": "config/locked"
        },
```

replace with 

```
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid":7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

TheHaf, 2022-04-02:
I streamlined the debugging even more using the remote extension in Visual Studio code - I love that tool.

I tested the changes on the childlock, reading works well now. Writing is the same as with the heatsetpoint, I can change the value in iobroker and see the change in the API response, but it won't propagate to the thermostat. So I am not sure about the conversion being the issue here.

Smanar, 2022-04-02:
>using the remote extension in Visual Studio code

What is it ?
I m asking because I realy need some logs to see what happen. On deconz if you enable the flags "DDF" "INFO" and "INFO_L2", you will be able to see the request send by deconz.

I think if you change the childlock mode on the device itself, the value is not updated on the API ?

TheHaf, 2022-04-02:
It's an extension that allows Visual Studio Code to directly browse and edit files on a remote machine using SSH, [see here for more info](https://code.visualstudio.com/docs/remote/ssh).

To enable the debug flags in headless mode, I started deconz in the foreground using the following command:
`/usr/bin/deCONZ -platform minimal --http-port=80 --dbg-info=2 --dbg-ddf=1`
Are these the correct parameters for the flags?

I get lots of logging, but the only thing that stood out when I tried to change a value using iobroker was the following line, which was there 11 times in a row:
`18:49:00:590 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x`

Regarding your last question, when I change the childlock on the device itself, the value is updated in the API, thanks to the changes to `config/locked`. That's what I meant in the second paragraph in my last comment, sorry if it was not written clearly.

Smanar, 2022-04-02:
Ha good question, I m not sure for "--dbg-ddf" but you have skipped the more used ^^ --dbg-info=1 (need both =1 and =2)

To have some line like
>DBG_Printf(DBG_INFO, "writeTuyaData, dpid: 0x%02X, type: 0x%02X, expr: %s\n",

If I m right you need to have (for the childlock)
>writeTuyaData, dpid: 0x07, type: 0x02b, expr: 1 (or 0)

TheHaf, 2022-04-03:
I checked the [wiki for the debug flags](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags), unfortunately it doesn't mention anything about the ddf debug flag yet. And I can't find anything in the code in the repository regarding the flag parsing.
Are you sure about writing `--dbg-info=1 --dbg-info=2`? When I do it this way, it only seems to print the messages from level 1. According to the wiki, level 2 should contain the level 1 messages as well, which seems to be the case, when I compare the output from using only level 1 with using only level 2.
Anyway, I can't find anything with writeTuyData regardless of which debug flags I use.
When I use only the dbg-ddf flag, I get the following lines when switching the childlock on the device itself.
```
09:58:11:908 50:32:5f:ff:fe:9f:36:c0-01-ef00/config/locked expression: Item.val = Attr.val; --> true
09:58:18:728 50:32:5f:ff:fe:9f:36:c0-01-ef00/config/locked expression: Item.val = Attr.val; --> false
```
However there is nothing when switching the value via iobroker.
When I use dbg-info=2 I still get the lines about `No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x`.

All of this was done after trying to use `Item.val == 1 ? 1 : 0;` in the eval of the config/locked entry.

Smanar, 2022-04-03:
Good question, in fact on the code there is both, and can select only 1 of them using the GUI

>DBG_Printf(DBG_INFO_L2, "No Tuya productId entry found for manufacturername: %s\n", qPrintable(manufacturerName->toString()));
>DBG_Printf(DBG_INFO, "writeTuyaData, dpid: 0x%02X, type: 0x%02X, expr: %s\n",

So I m not sure what happen if using both or only one .... (It's not on api code), but the line I need are just used with "info".

In fact I m trying to trace this fonction https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/device_access_fn.cpp#L797

As you can see all debug line are with "info"

TheHaf, 2022-04-03:
Indeed, the relevant lines in that function are all level 1. So it seems we don't even reach line 851 and are returning early on one of the pre-conditions.
Maybe there is something wrong with my changes? Here is what I currently have for the childlock:
```
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"}
        },
```
I re-tried it just now with ` --dbg-info=1 --dbg-ddf=1 --dbg-error=2`, there is unfortunately still nothing in the log when I change the value in iobroker. I also tried it with db-error set to 1, just in case, but no change in the outcome.

Smanar, 2022-04-03:
Ha yep sorry, I m asking on discord ATM.
It seem the write command can't work perfectly for the moment, I m not sure but we probably need the PR called "Add DDF for Danfoss Ally thermostat"

TheHaf, 2022-04-04:
Ok, well at least reading out the values seems to work except for the battery.
Maybe the PR you mentioned will help with the writing, so I keep my fingers crossed for that. 😸

kvanisterdael, 2022-04-07:
Same here: reading of values is ok. 
I can set the heatsetpoint via the PUT command and the change is reflected in the config I read via the REST API.
However on the display of the Thermostat the value is not updated.

Smanar, 2022-04-07:
Hello, the Swoop PR is merged, but now there is a problem to add our values need to edit the code. I can make change in blind mode but to go faster there is someone able to compile the code ?

I will explain how to do but need someone with a Tuya TRV (any) and a full/real Linux OS.

Smanar, 2022-04-07:
Perfect, I will make a code change to test tommorow or this week end if I don't have time.
Will be a test to enable config/heatsetpoint and config/childlock. I m asking some confirmations to swoop and manup.

So I will give you in same time the procedure to compile the lib, it will just replace 1 file (and can make a backup), so not realy dangerous, will be based on the last code, and ofc will use the previous DDF.

kvanisterdael, 2022-04-08:
While you're at it, maybe a test to disable the 'open window' detection as well, as this is enabled by default.

TheHaf, 2022-04-08:
I can also give it a try with the code change when it's comitted.

Smanar, 2022-04-08:
BTW I m seing on the future version, the 2.15.1, the childlock could be tested without recompilation, the code is ready for tuya too.

So waiting for this version will be released, to try the test version, the procedure is explained here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So it make 

```
sudo apt install deconz-dev
git clone --branch tuya_trv_test https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder. The procedure change only this file.

You need to use first the config/heatsetpoint to enable the tuya reporting (because of the `"read": {"fn": "tuya"}` in the DDF.
Then you can try to set a temperature (this command use value) and the childlock (this command use a bool)

If I m right you will see in log when doing the request  (with "info" and "info_l2")
```
writeTuyaData, dpid: 0x07, type: 0x010, expr: Item.val
Tuya write expression: Item.val--> 1
```

The DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_hhrtiq0x",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve"
        },
        {
          "name": "state/windowopen"
        }
      ]
    }
  ]
}
```

Have not enabled yet the "open window", it's a test, but if the childlock is working will be easy to add it.

macinsight, 2022-04-08:
Tried setting the temperature using `curl -X PUT http://192.168.178.28/api/E55CDE04B9/sensors/15/config -H "Content-Type: application/json" -d '{ "heatsetpoint": 1500`, this had no discernible effect (Althought the response *was* `success`.) The API reports the `heatsetpoint` correctly afterwards. The display on the TRV does not change. There also was no discernible change in temperature from 20 to 15 degrees over the testing period.

Also tried setting the childlock using `curl -X PUT http://192.168.178.28/api/E55CDE04B9/sensors/15/config -H "Content-Type: application/json" -d '{ "locked": true }'`, this also had no discernible effect. The display did not change either.

Also, I am unable to find `writeTuyaData, dpid: 0x07, type: 0x010, expr: Item.val
Tuya write expression: Item.val--> 1
` (Or anything concerning tuya, really), with the following launch arguments: `/usr/bin/deCONZ --platform minimal --dbg-info=2 --dbg-error=2`. 

The logs produced during testing will follow tomorrow.

Smanar, 2022-04-09:
Ha, there is a problem ^^, I don't know how to enable "info" and "info_l2" in command line. And you miss all "info"
Nevermind, have updated the code to display more debug line with "info_l2" , less talkative

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Will be faster this time, if you can send me the same logs when you make same procedure pls ?

macinsight, 2022-04-09:
Sure, gonna get to testing right now.

macinsight, 2022-04-09:
No noticeable change from previous testing, logs will follow.

Smanar, 2022-04-09:
Yep, I haven't updated code, just add loging.

TheHaf, 2022-04-09:
I also compiled it now and tried it out using the debug flag parameters `--dbg-info=2 --dbg-ddf=1 --dbg-error=2`. We should really try to figure out how these debug flags actually work altogether and then add it to the documentation. Could this be something from the Qt framework or some external logging library?

As for the result of my test, I also don't get any different result than before, the change is not propagated to the device. One thing that I note is the line with `debug test 3`, I guess that is one of your added debug messages, that seems to right where I tried to change the lock setting.
Here is the full log, the change was roughly around timestamp `17:56:16:175`, where the debug message can be seen:
```
(...)
17:56:13:851 Idle timer triggered
17:56:13:851 Force binding of attribute reporting for node Flurlicht
17:56:13:851 Force read attributes for ZHAOpenClose SensorNode Türsensor Büro
17:56:13:853 rule event /config/localtime: 17:56:12.850 -> 17:56:13.851 (1001ms)
17:56:14:426 poll node 00:17:88:01:04:40:82:ec-0b
17:56:14:426 Poll light node Wohndecke 1
17:56:14:479 read attributes of 0x00178801044082EC cluster: 0x0006: [ 17:56:14:479 0x0000 17:56:14:479 ]
17:56:14:480 add task 161 type 19 to 0x00178801044082EC cluster 0x0006 req.id 18
17:56:14:480 Poll APS request 18 to 0x00178801044082EC cluster: 0x0006
17:56:14:562 Poll APS confirm 18 status: 0x00
17:56:14:562 Erase task req-id: 18, type: 19 zcl seqno: 25 send time 0, profileId: 0x0104, clusterId: 0x0006
17:56:14:582 Node data 0x00178801044082ec profileId: 0x0104, clusterId: 0x0006
17:56:14:582 0x00178801044082EC: update ZCL value 0x0B/0x0006/0x0000 after 0 s
17:56:14:850 schedule 3 diff 16721, Sat Apr 9 22:34:55 2022
17:56:14:852 rule event /config/localtime: 17:56:13.851 -> 17:56:14.850 (999ms)
17:56:15:425 poll node 84:18:26:00:00:0e:24:23-03
17:56:15:425 Poll light node Esstisch 3
17:56:15:477 Poll APS request to 0x84182600000E2423 cluster: 0x0006 dropped, values are fresh enough
17:56:15:850 schedule 3 diff 16720, Sat Apr 9 22:34:55 2022
17:56:15:850 Wait 2s till query finished
17:56:15:852 rule event /config/localtime: 17:56:14.850 -> 17:56:15.850 (1000ms)
17:56:16:175 debug test 3
17:56:16:176 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:178 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:179 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:180 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:181 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:183 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:184 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:185 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:186 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:187 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:188 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
17:56:16:197 Websocket 192.168.178.57:8857 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
17:56:16:198 Websocket 127.0.0.1:40032 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
17:56:16:350 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
17:56:16:525 poll node 00:17:88:01:03:a1:f4:ed-0b
17:56:16:525 Poll light node Flurlicht
17:56:16:578 read attributes of 0x0017880103A1F4ED cluster: 0x0006: [ 17:56:16:578 0x0000 17:56:16:578 ]
17:56:16:578 add task 171 type 19 to 0x0017880103A1F4ED cluster 0x0006 req.id 31
17:56:16:578 Poll APS request 31 to 0x0017880103A1F4ED cluster: 0x0006
17:56:16:663 Poll APS confirm 31 status: 0x00
17:56:16:663 Erase task req-id: 31, type: 19 zcl seqno: 26 send time 0, profileId: 0x0104, clusterId: 0x0006
17:56:16:673 Node data 0x0017880103a1f4ed profileId: 0x0104, clusterId: 0x0006
17:56:16:673 0x0017880103A1F4ED: update ZCL value 0x0B/0x0006/0x0000 after 0 s
17:56:16:851 schedule 3 diff 16719, Sat Apr 9 22:34:55 2022
17:56:16:852 rule event /config/localtime: 17:56:15.850 -> 17:56:16.851 (1001ms)
17:56:17:000 APS-DATA.indication from child 0x4536
17:56:17:675 poll node 58:8e:81:ff:fe:d4:ae:49-01
17:56:17:675 Poll light node Blitzwolf-2
(...)
```

Smanar, 2022-04-09:
>I guess that is one of your added debug messages

Yep exactly.

>We should really try to figure out how these debug flags actually work altogether and then add it to the documentation

I m asking ATM.

I m seing 2 mac adress, the device actually used is the 50:32:5f:ff:fe:9f:36:c0-01-ef00 ?
So yes It seem there is something
```
17:56:22:039 SC tick --> StateRead
17:56:22:039 SC state change failed: 50:32:5f:ff:fe:9f:36:c0-01-ef00
```

It's like the code is waiting for a return from the device, but there is no logs indicating something was send to the device ...

Have added some more debug line, if you can make a last try ?
BTW have you the GUI, if yes, a screen log capture can be enought.

TheHaf, 2022-04-10:
Regarding the debug flags, after comparing the messages when using `--dbg-info=2` to the code, I am now convinced that the command line flag for level 2 contains the level 1 messages as well. For instance you changed the message `SC tick --> StateRead` in your latest commit to be in level 2 instead of level 1 but I did see it before as well when using the flag for level 2.

I have not seen how the messages come together in the UI yet, only the logging window with the check boxes. If the check boxes really toggle the levels disjunct, then the behavior of the command line flags is definitely different to the UI.

Yes, the device is the one with the id `50:32:5f:ff:fe:9f:36:c0-01-ef00`.

I tried out your latest changes just now and have additional log messages, at least `09:12:22:906 Device managed by DDF` during the time when the change occurred. Regarding the `debug test` messages, I can only see the ones numbered 3 and 7.

log output:
```
09:12:22:851 Poll light node Esstisch 3
09:12:22:851 Idle timer triggered
09:12:22:851 Force binding of attribute reporting for node Flurlicht
09:12:22:851 Force read attributes for ZHASwitch SensorNode TRADFRI Dimmer Küche
09:12:22:852 rule event /config/localtime: 09:12:21.850 -> 09:12:22.851 (1001ms)
09:12:22:857 read attributes of 0x84182600000E2423 cluster: 0x0006: [ 09:12:22:858 0x0000 09:12:22:858 ]
09:12:22:858 add task 177 type 19 to 0x84182600000E2423 cluster 0x0006 req.id 53
09:12:22:858 Poll APS request 53 to 0x84182600000E2423 cluster: 0x0006
09:12:22:906 debug test 3
09:12:22:906 Device managed by DDF
09:12:22:906 ressource ok
09:12:22:907 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:908 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:910 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:911 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:912 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:913 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:915 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:916 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:917 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:918 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:919 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
09:12:22:928 Websocket 127.0.0.1:37518 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
09:12:22:929 Websocket 192.168.178.57:11837 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
09:12:22:929 Websocket 127.0.0.1:37610 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
09:12:22:929 Websocket 192.168.178.57:11842 send message: {"config":{"battery":null,"heatsetpoint":600,"locked":true,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
09:12:22:929 debug test 7
09:12:22:942 Poll APS confirm 53 status: 0x00
09:12:22:942 Erase task req-id: 53, type: 19 zcl seqno: 155 send time 0, profileId: 0x0104, clusterId: 0x0006
09:12:22:969 Node data 0x84182600000e2423 profileId: 0x0104, clusterId: 0x0006
09:12:22:969 0x84182600000E2423: update ZCL value 0x03/0x0006/0x0000 after 0 s
09:12:23:850 schedule 2 diff 25449, Sun Apr 10 16:16:32 2022
09:12:23:850 schedule 3 diff 48152, Sun Apr 10 22:34:55 2022
09:12:23:852 rule event /config/localtime: 09:12:22.851 -> 09:12:23.850 (999ms)
09:12:23:925 poll node 00:17:88:01:03:a1:f4:ed-0b
09:12:23:925 Poll light node Flurlicht
09:12:23:977 read attributes of 0x0017880103A1F4ED cluster: 0x0006: [ 09:12:23:977 0x0000 09:12:23:977 ]
09:12:23:977 add task 183 type 19 to 0x0017880103A1F4ED cluster 0x0006 req.id 60
09:12:23:977 Poll APS request 60 to 0x0017880103A1F4ED cluster: 0x0006
09:12:24:061 Poll APS confirm 60 status: 0x00
09:12:24:061 Erase task req-id: 60, type: 19 zcl seqno: 156 send time 0, profileId: 0x0104, clusterId: 0x0006
09:12:24:084 Node data 0x0017880103a1f4ed profileId: 0x0104, clusterId: 0x0006
09:12:24:085 0x0017880103A1F4ED: update ZCL value 0x0B/0x0006/0x0000 after 0 s
09:12:24:675 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x1282
09:12:24:850 schedule 2 diff 25448, Sun Apr 10 16:16:32 2022
09:12:24:850 schedule 3 diff 48151, Sun Apr 10 22:34:55 2022
09:12:24:851 rule event /config/localtime: 09:12:23.850 -> 09:12:24.850 (1000ms)
09:12:25:025 poll node 58:8e:81:ff:fe:d4:ae:49-01
09:12:25:025 Poll light node Blitzwolf-2
09:12:25:077 read attributes of 0x588E81FFFED4AE49 cluster: 0x0006: [ 09:12:25:077 0x0000 09:12:25:077 ]
09:12:25:077 add task 188 type 19 to 0x588E81FFFED4AE49 cluster 0x0006 req.id 68
09:12:25:077 Poll APS request 68 to 0x588E81FFFED4AE49 cluster: 0x0006
09:12:25:161 Poll APS confirm 68 status: 0x00
[... cut ...]
09:12:29:850 schedule 3 diff 48146, Sun Apr 10 22:34:55 2022
09:12:29:852 rule event /config/localtime: 09:12:28.851 -> 09:12:29.850 (999ms)
09:12:29:925 poll node 84:18:26:00:00:0e:29:78-03
09:12:29:925 Poll light node Esstisch 4
09:12:29:977 read attributes of 0x84182600000E2978 cluster: 0x0006: [ 09:12:29:977 0x0000 09:12:29:977 ]
09:12:29:977 add task 214 type 19 to 0x84182600000E2978 cluster 0x0006 req.id 104
09:12:29:977 Poll APS request 104 to 0x84182600000E2978 cluster: 0x0006
09:12:30:061 Poll APS confirm 104 status: 0x00
```

Smanar, 2022-04-10:
> I am now convinced that the command line flag for level 2 contains the level 1 messages as well

Swoop say that too, but I have check some debug line from your logs and all are with DBG_INFO_L2 and realy miss lot of them.

But You still have a succed message when you make the request ?

```
09:12:22:906 debug test 3
09:12:22:906 Device managed by DDF
09:12:22:906 ressource ok
```

This is good new, we are in the fonction that manage parametre, and all values are fine, but 
```
                else if (rid.suffix == RConfigHeatSetpoint) // Signed integer
                {
                    DBG_Printf(DBG_INFO_L2, "debug test 4\n");
```

The code don't use the config/heatsetpoint at all ....
There is something wrong in the request or there is something in the code that break this part, but in this situation you can"t have a succed message ...

TheHaf, 2022-04-10:
Yes, I get a success message. I also tried it using curl instead of iobroker this time and curl responds for example when setting the lock with `[{"success":{"/sensors/50/config/locked":true}}]`.

I just tried setting a debug message before that long if/else if/... block to check the actual contents of rid.suffix.
For setting the lock I get `config/locked` and for setting a heatsetpoint I get `config/heatsetpoint` as the value of rid.suffix. So this seems to be right.

That made me realize though that I didn't clarify what exactly I performed in my last comment, sorry about any misunderstanding. I have now a log for setting the heatsetpoint as well, and there you can see the debug messages 4 and 5 as well:
```
16:00:59:381 Erase task req-id: 106, type: 19 zcl seqno: 75 send time 0, profileId: 0x0104, clusterId: 0x0006
16:00:59:402 Node data 0x84182600000e2978 profileId: 0x0104, clusterId: 0x0006
16:00:59:402 0x84182600000E2978: update ZCL value 0x03/0x0006/0x0000 after 0 s
16:00:59:850 schedule 2 diff 933, Sun Apr 10 16:16:32 2022
16:00:59:851 schedule 3 diff 23636, Sun Apr 10 22:34:55 2022
16:00:59:851 Wait 2s till query finished
16:00:59:852 rule event /config/localtime: 16:00:58.850 -> 16:00:59.851 (1001ms)
16:01:00:044 debug test 3
16:01:00:044 Device managed by DDF
16:01:00:044 ressource ok
16:01:00:044 i1-rid.suffix=config/heatsetpoint
16:01:00:044 i2-rid.suffix=config/heatsetpoint
16:01:00:044 debug test 4
16:01:00:046 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:048 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:050 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:051 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:053 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:055 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:057 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:058 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:060 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:062 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:064 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
16:01:00:064 debug test 5
16:01:00:075 Websocket 192.168.178.57:13100 send message: {"config":{"battery":null,"heatsetpoint":1500,"locked":false,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
16:01:00:076 Websocket 192.168.178.57:13108 send message: {"config":{"battery":null,"heatsetpoint":1500,"locked":false,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
16:01:00:076 Websocket 127.0.0.1:42208 send message: {"config":{"battery":null,"heatsetpoint":1500,"locked":false,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
16:01:00:076 debug test 7
16:01:00:350 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:01:00:350 poll node 84:18:26:00:00:0e:31:3e-03
16:01:00:350 Poll light node Esstisch 1
```

So it does get into the else if part.

The two lines before the `debug test 4` show that I can receive both messages with `DBG_INFO` and `DBG_INFO_L2` using my command line parameters.

The 11 lines with "No Tuya productId entry found for manufacturername" happen with changing the lock state as well, not just with changing the heatsetpoint. That seems to happen in `R_GetProductId` in `product_match.cpp`. It looks to me as if there is not yet a product entry for `_TZE200_hhrtiq0x`.

Smanar, 2022-04-10:
"No Tuya productId entry found for manufacturername"

This part will be removed soon ^^, it s the old legacy code.
Now we are using the DDF, and the DDF core can work only for new device, so device that are generaly not in this list.

So now it's better, debug line are all here. I think I have found the problem, have removed some debug line and add some more.
But this time you have too a code modification for the config/heatsetpoint. I m trying to write the value before waiting for a return.

TheHaf, 2022-04-10:
Ok, then I'll just ignore that part from now on, good to know.

I discarded my changes and pulled the current state of the branch again. And lo and behold, when I change the heatsetpoint now, the display on the device lights up and shows the new value. 👍
Here is the log:
```
ttributes for ZHASwitch SensorNode SYMFONISK Sound Controller
18:21:40:852 Node data 0x84182600000e313e profileId: 0x0104, clusterId: 0x0300
18:21:40:852 0x84182600000E313E: added ZCL value 0x03/0x0300/0x0007
18:21:40:852 0x84182600000E313E: added ZCL value 0x03/0x0300/0x0008
18:21:40:852 0x84182600000E313E: added ZCL value 0x03/0x0300/0x4001
18:21:40:853 rule event /config/localtime: 18:21:39.850 -> 18:21:40.851 (1001ms)
18:21:40:907 read attributes of 0x7CB03EAA00B1DE7B cluster: 0x0006: [ 18:21:40:907 0x0000 18:21:40:907 ]
18:21:40:907 add task 162 type 19 to 0x7CB03EAA00B1DE7B cluster 0x0006 req.id 25
18:21:40:907 Poll APS request 25 to 0x7CB03EAA00B1DE7B cluster: 0x0006
18:21:40:961 Poll APS confirm 25 status: 0x00
18:21:40:961 Erase task req-id: 25, type: 19 zcl seqno: 36 send time 0, profileId: 0x0104, clusterId: 0x0006
18:21:41:009 Node data 0x7cb03eaa00b1de7b profileId: 0x0104, clusterId: 0x0006
18:21:41:009 0x7CB03EAA00B1DE7B: added ZCL value 0x03/0x0006/0x0000
18:21:41:011 Websocket 192.168.178.57:9896 send message: {"attr":{"id":"2","lastannounced":"2021-10-26T09:33:39Z","lastseen":"2022-04-10T16:21Z","manufacturername":"OSRAM","modelid":"Plug 01","name":"Stehlampe","swversion":"V1.04.12","type":"On/Off plug-in unit","uniqueid":"7c:b0:3e:aa:00:b1:de:7b-03"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"7c:b0:3e:aa:00:b1:de:7b-03"} (ret = 335)
18:21:41:012 Websocket 192.168.178.57:9910 send message: {"attr":{"id":"2","lastannounced":"2021-10-26T09:33:39Z","lastseen":"2022-04-10T16:21Z","manufacturername":"OSRAM","modelid":"Plug 01","name":"Stehlampe","swversion":"V1.04.12","type":"On/Off plug-in unit","uniqueid":"7c:b0:3e:aa:00:b1:de:7b-03"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"7c:b0:3e:aa:00:b1:de:7b-03"} (ret = 335)
18:21:41:012 Websocket 192.168.178.57:9896 send message: {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 21034256)
18:21:41:012 Websocket 192.168.178.57:9910 send message: {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 21034256)
18:21:41:036 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:038 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:039 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:040 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:041 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:043 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:044 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:045 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:046 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:047 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:048 No Tuya productId entry found for manufacturername: _TZE200_hhrtiq0x
18:21:41:048 debug test 5
18:21:41:064 Websocket 192.168.178.57:9896 send message: {"config":{"battery":null,"heatsetpoint":1200,"locked":false,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
18:21:41:064 Websocket 192.168.178.57:9910 send message: {"config":{"battery":null,"heatsetpoint":1200,"locked":false,"on":true,"reachable":true},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 18)
18:21:41:064 debug test 7
18:21:41:064 SC tick --> StateCallFunction
18:21:41:064 debug test 2
18:21:41:064 debug test 1
18:21:41:064 writeTuyaData, dpid: 0x02, type: 0x2B, expr: Item.val / 10;
18:21:41:065 Tuya write expression: Item.val / 10; --> 120
18:21:41:781 poll node 84:18:26:00:00:0e:32:bf-03
18:21:41:781 Poll light node Esstisch 2
18:21:41:834 Poll APS request to 0x84182600000E32BF cluster: 0x0006 dropped, values are fresh enough
18:21:41:850 schedule 3 diff 15194, Sun Apr 10 22:34:55 2022
18:21:41:852 rule event /config/localtime: 18:21:40.851 -> 18:21:41.850 (999ms)
18:21:42:852 schedule 3 diff 15193, Sun Apr 10 22:34:55 2022
18:21:42:854 rule event /config/localtime: 18:21:41.850 -> 18:21:42.852 (1002ms)
18:21:42:925 poll node 90:fd:9f:ff:fe:08:b8:47-01
18:21:42:925 Poll light node Arbeitsfläche
18:21:42:977 read attributes of 0x90FD9FFFFE08B847 cluster: 0x0006: [ 18:21:42:977 0x0000 18:21:42:977 ]
18:21:42:977 add task 172 type 19 to 0x90FD9FFFFE08B847 cluster 0x0006 req.id 40
18:21:42:977 Poll APS request 40 to 0x90FD9FFFFE08B847 cluster: 0x0006
18:21:43:063 Poll APS confirm 40 status: 0x00
18:21:43:063 Erase task req-id: 40, type: 19 zcl seqno: 39 send time 0, profileId: 0x0104, clusterId: 0x0006
18:21:43:103 Node data 0x90fd9ffffe08b847 profileId: 0x0104, clusterId: 0x0006
18:21:43:103 0x90FD9FFFFE08B847: update ZCL value 0x01/0x0006/0x0000 after 0 s
18:21:43:850 schedule 3 diff 15192, Sun Apr 10 22:34:55 2022
18:21:43:852 rule event /config/localtime: 18:21:42.852 -> 18:21:43.850 (998ms)
```

Changing the locked status still won't update the device unfortunately, here is the log for that:
```
[... cut ...] :c0-01-ef00"},"e":"changed","id":"50","r":"sensors","t":"event","uniqueid":"50:32:5f:ff:fe:9f:36:c0-01-ef00"} (ret = 2128620552)
18:23:13:844 debug test 7
18:23:13:851 schedule 3 diff 15102, Sun Apr 10 22:34:55 2022
18:23:13:853 rule event /config/localtime: 18:23:12.851 -> 18:23:13.851 (1000ms)
18:23:13:854 APS-DATA.indication from child 0x4536
```

I tried it with the write eval for config/locked being `Item.val;` as well as `Item.val == 1 ? 1 : 0;`, to no avail. The log is from the first try.

Smanar, 2022-04-10:
Haa ^^, good new.
I have edited the code only for the setheatpint request (was a test)
Will add the other for tomorow.

TheHaf, 2022-04-11:
Oh, guess I should have taken a look at the latest commit. 😉
I tried changing the code for the locked state in the same way in lines 1577+:
```
StateChange change2(StateChange::StateCallFunction, SC_WriteZclAttribute, task.req.dstEndpoint());
change2.addTargetValue(rid.suffix, data.uinteger);
rsub->addStateChange(change2);
```

And it works! Most of the time. Actually, sometimes it won't update. I also noticed odd behavior with changing the heatsetpoint, sometimes it won't update, and sometimes it updates but it will return a weird value. For instance, I set it to 10°C and it does that but afterwards the API states the value 16.3°C. Or I set it to 16°c and it does that but the API will change to 9°C afterwards.

Smanar, 2022-04-11:
Something worring is you don't have thoses logs for childlock (and have added some debug line in a suspicious fonction)

17:56:22:039 SC tick --> StateRead
17:56:22:039 SC state change failed: 50:32:5f:ff:fe:9f:36:c0-01-ef00

Have added your code + the window open detection
DPID for windows detection is 18 = 0x12

```
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 18, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

```
18:21:41:064 writeTuyaData, dpid: 0x02, type: 0x2B, expr: Item.val / 10;
18:21:41:065 Tuya write expression: Item.val / 10; --> 120
```

And something strange, from my memory tuya answer immedialtly after a change, and I don't see return on your logs.

You don't have logs when you have values problem ?

And when you use --dbg-info=2, you need to have "info" AND "info_l2".

macinsight, 2022-04-11:
Confirmation from my side that thermostat control via deCONZ/HA combo works flawlessly.

Okay, almost, as it's continuously reporting as "off" and doesn't have the ability to set a mode (Manual, automatic and $third)

Many thanks to you!

Smanar, 2022-04-11:
>Okay, almost, as it's continuously reporting as "off" and doesn't have the ability to set a mode (Manual, automatic and $third)

Ha yes, right, will be the next step, but I m worried by the "random" result for @TheHaf . You don't have this kind of result ?
No answer or differents values ?

macinsight, 2022-04-11:
I'm unable to reproduce the issue with the "weird" value so far, but I can confirm that sometimes, it refuses to update. It's seemingly random, too.

Sadly, I don't have time to test it during the week, only on weekends. But it's fucking _great_ that you've got it where it is so far!

TheHaf, 2022-04-11:
One small test I did later today also did not result in weird values. The difference was, I did not run deconz in the foreground with the flags, but as a service. Maybe that has something to do with it, but I need to investigate further.

I am sorry I could not find the time today to do more testing than that. Maybe I can get you some logs with the wrong values tomorrow or Wednesday, I hope.

One other oddity I noticed was that the temperature read was also wrong. It showed a value of 0.5C (in iobroker) or 50 (raw value in the API) in a room that was about 19C.

Smanar, 2022-04-12:
Ok was wrong for the mode

There is 0x04 0x04 for the Preset with all values
There is 0x04 0x6A for the force mode, so a sort of off/manu/auto.

Have corrected the code.
So now the mode can work for read/write and preset onlt for read (but need to update the file , not the same name)

New json

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_hhrtiq0x",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 18, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/mode",
          "parse": {"fn": "tuya", "dpid": 106, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 106, "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 2 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/preset",
          "parse": {"fn": "tuya", "dpid": 4, "script": "tuya_trv_preset.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve"
        },
        {
          "name": "state/windowopen"
        }
      ]
    }
  ]
}
```

For the bad value, there is so much much thing possible, for exemple the device use the same dpid but with another type (for the code a device use a dpid only 1 time)
The legacy code can pass under the DDF one.
Or some other magic ^^, it s realy something untested yet, but I hope logs will be talkative enought to help us.

TheHaf, 2022-04-13:
> but need to update the file , not the same name
Do I understand that correctly, because of certain changes in the file I need to rename it so that deconz can read it correctly from scratch?

I was able to make only a quick test today (sorry, no logs today). When I tried setting the heatsetpoint to 10°C, it was shown as such on the device, but on the API it went to 9.2°C. Then The device lit up again with 10°c and the API showed 8.8°C. THEN the device lit up again with 10°C and finally the API also showed 10°C. That all happened within a minute or so. I used the updated file with a new name for that test. As I said earlier, unfortunately I did not have enough time to also log this behavior, hopefully tomorrow.

As for the strange read temperature value, after my last comment it went down to 0. Now it's at 10.5°C in iobroker, the API shows a value of 1050. The room has a temperature of over 20°C.

Smanar, 2022-04-14:
So can be a convertion issue, from my memory heatpoint for tuya can be only full value or with 0.5 °C precision, I never see a device that can use 9.2°C or 8.8°C

TheHaf, 2022-04-14:
Indeed, the device itself only lets me change the heatpoint in 0.5°C stepts. So that is really weird.

After making sure that I am on the most current git commit without any local changes, compiling and copying the lib and restarting the service, I now have completely strange behavior. I can't even get the locked value set reliably, it only allows me to set it to true and even then read values are inconsistent. I need to check again for any possible mistakes on my end before sending any logs.

Smanar, 2022-04-14:
NP, we have time.
Have updated the DDF too, but on the old version the config/lock was working and haven't changed this part.
Can have broken something when adding config/mode.

TheHaf, 2022-04-23:
Hi there! 😺 I finally found the time to do some more testing. I started from scratch, pulled your current branch, compiled it, copied the lib, added the json file including the extra config/heatpoint entry that you mentioned in your last post (didn't see anything related to it in the API though), started deconz with the debug flags and played around with the heatsetpoint configuration via iobroker.

The first thing I noticed was that the API still reported a wrong heatsetpoint value of 740 even after several restarts and the device still having it set to 11°C. Is there a cache that would explain this now very old value? I would have assumed that after a restart the value is being read again? Or is this read decoupled from deconz directly within the Conbee2 stick and deconz only accesses the values from the stick once it's active?

Anyway, after noticing that, I changed the value on the device itself first and set it to 14°C, here is the log:
```
18:38:30:612 Websocket 127.0.0.1:55006 send message: {"e":"changed","id":"34","r":"sensors","state":{"current":0,"lastupdated":"2022-04-23T16:38:30.610","power":0,"voltage":228},"t":"event","uniqueid":"58:8e:81:ff:fe:d4:ae:49-01-0b04"} (ret = 15)
18:38:30:612 Websocket 192.168.178.57:13530 send message: {"e":"changed","id":"34","r":"sensors","state":{"current":0,"lastupdated":"2022-04-23T16:38:30.610","power":0,"voltage":228},"t":"event","uniqueid":"58:8e:81:ff:fe:d4:ae:49-01-0b04"} (ret = 15)
18:38:30:613 Websocket 192.168.178.57:13594 send message: {"e":"changed","id":"34","r":"sensors","state":{"current":0,"lastupdated":"2022-04-23T16:38:30.610","power":0,"voltage":228},"t":"event","uniqueid":"58:8e:81:ff:fe:d4:ae:49-01-0b04"} (ret = 15)
18:38:30:813 Node data 0x84182600000e2423 profileId: 0x0104, clusterId: 0x0300
18:38:30:813 0x84182600000E2423: update ZCL value 0x03/0x0300/0x0007 after 0 s
18:38:30:813 ZCL attribute report 0x84182600000E2423 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000
18:38:30:813    payload: 0700217201
18:38:30:815 Websocket 127.0.0.1:55006 send message: {"attr":{"colorcapabilities":16,"ctmax":370,"ctmin":153,"id":"9","lastannounced":"2022-03-25T04:38:41Z","lastseen":"2022-04-23T16:38Z","manufacturername":"OSRAM","modelid":"Classic B40 TW - LIGHTIFY","name":"Esstisch 3","swversion":"V1.04.12","type":"Color temperature light","uniqueid":"84:18:26:00:00:0e:24:23-03"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0e:24:23-03"} (ret = 405)
18:38:30:815 Websocket 192.168.178.57:13530 send message: {"attr":{"colorcapabilities":16,"ctmax":370,"ctmin":153,"id":"9","lastannounced":"2022-03-25T04:38:41Z","lastseen":"2022-04-23T16:38Z","manufacturername":"OSRAM","modelid":"Classic B40 TW - LIGHTIFY","name":"Esstisch 3","swversion":"V1.04.12","type":"Color temperature light","uniqueid":"84:18:26:00:00:0e:24:23-03"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0e:24:23-03"} (ret = 405)
18:38:30:815 Websocket 192.168.178.57:13594 send message: {"attr":{"colorcapabilities":16,"ctmax":370,"ctmin":153,"id":"9","lastannounced":"2022-03-25T04:38:41Z","lastseen":"2022-04-23T16:38Z","manufacturername":"OSRAM","modelid":"Classic B40 TW - LIGHTIFY","name":"Esstisch 3","swversion":"V1.04.12","type":"Color temperature light","uniqueid":"84:18:26:00:00:0e:24:23-03"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0e:24:23-03"} (ret = 405)
18:38:30:828 Node data 0x84182600000e2423 profileId: 0x0104, clusterId: 0x0300
18:38:30:828 0x84182600000E2423: update ZCL value 0x03/0x0300/0x0008 after 0 s
18:38:30:828 ZCL attribute report 0x84182600000E2423 for cluster: 0x0300, ep: 0x03, frame control: 0x18, mfcode: 0x0000
18:38:30:828    payload: 08003002
18:38:31:190 poll node 58:8e:81:ff:fe:d3:8d:ce-01-0b04
18:38:31:190 Poll ZHAPower sensor node Power 32
18:38:31:350 schedule 3 diff 14184, Sat Apr 23 22:34:55 2022
18:38:31:350 Wait 0s till query finished
18:38:31:353 rule event /config/localtime: 18:38:30.351 -> 18:38:31.350 (999ms)
18:38:31:450 read attributes of 0x588E81FFFED38DCE cluster: 0x0B04: [ 18:38:31:450 0x0505 18:38:31:450 0x0508 18:38:31:450 0x050B 18:38:31:450 ]
18:38:31:450 add task 1095 type 19 to 0x588E81FFFED38DCE cluster 0x0B04 req.id 180
18:38:31:450 Poll APS request 180 to 0x588E81FFFED38DCE cluster: 0x0B04
18:38:31:470 Poll APS confirm 180 status: 0x00
18:38:31:470 Erase task req-id: 180, type: 19 zcl seqno: 237 send time 0, profileId: 0x0104, clusterId: 0x0B04
18:38:31:504 Node data 0x588e81fffed38dce profileId: 0x0104, clusterId: 0x0B04
18:38:31:504 0x588E81FFFED38DCE: update ZCL value 0x01/0x0B04/0x0505 after 0 s
18:38:31:504 0x588E81FFFED38DCE: update ZCL value 0x01/0x0B04/0x0508 after 0 s
18:38:31:505 0x588E81FFFED38DCE: update ZCL value 0x01/0x0B04/0x050B after 0 s
18:38:31:506 Websocket 127.0.0.1:55006 send message: {"e":"changed","id":"32","r":"sensors","state":{"current":420,"lastupdated":"2022-04-23T16:38:31.505","power":88,"voltage":228},"t":"event","uniqueid":"58:8e:81:ff:fe:d3:8d:ce-01-0b04"} (ret = 15)
18:38:31:506 Websocket 192.168.178.57:13530 send message: {"e":"changed","id":"32","r":"sensors","state":{"current":420,"lastupdated":"2022-04-23T16:38:31.505","power":88,"voltage":228},"t":"event","uniqueid":"58:8e:81:ff:fe:d3:8d:ce-01-0b04"} (ret = 15)
[... cut ...]
18:45:23:850 schedule 3 diff 13772, Sat Apr 23 22:34:55 2022
```

As you can see, this fires a lot of Tuya-related logs over the course of several seconds. The device does get set to 16°C, but the API reports 880 and iobroker 8,8°C again. I can also see this in the log, with the `TY_DATA_RESPONSE: seq 203, dpid: 0x02, type: 0x02, length: 4, val: 160` lines having the expected value of 160 and the `50:32:5f:ff:fe:9f:36:c0-01-ef00/config/heatsetpoint expression: Item.val = Attr.val * 10; --> 880` showing me 880.

Smanar, 2022-04-24:
> The first thing I noticed was that the API still reported a wrong heatsetpoint value of 740 even after several restarts and the device still having it set to 11°C. Is there a cache that would [...]

I think yes, all "config" value are stored in deconz, and it wait for new value from the api or from the device to update them (the device too make periodic report)

For this device the only convertion is * 10, so not normal having 880 for 140, and why always the same value ...

I m trying to re build the tuya request I think it was something like
`00 XX 02 02 00 04 00 00 00 8C`

How the code can find 88 somewhere ?
There is something bad, in the JS interpreter, but how it have worked before ?

Have made some change and add more debug line, you will see them around thoses lines 
```
18:45:17:928 Tuya parse expression
18:45:17:929 50:32:5f:ff:fe:9f:36:c0-01-ef00/config/heatsetpoint expression: Item.val = Attr.val * 10; --> 880
18:45:17:929 TY_DATA_REPORT: seq 120, dpid: 0x02, type: 0x02, length: 4, val: 160
```

TheHaf, 2022-04-24:
Thanks for your ongoing effort! I pulled your changes and re-tested. I don't see any of the `50:32:5f:ff:fe:9f:36:c0-01-ef00/config/heatsetpoint expression` anymore though.

As before, I changed the value first on the device, here is the log for changing it to 14°C, with the API returning 870 this time:
```
16:14:53:350 Wait 0s till query finished
16:14:53:352 rule event /config/localtime: 16:14:52.351 -> 16:14:53.350 (999ms)
16:14:53:850 poll node 84:18:26:00:00:0e:24:23-03
16:14:53:850 Poll light node Esstisch 3
16:14:53:900 Poll APS request to 0x84182600000E2423 cluster: 0x0006 dropped, values are fresh enough
16:14:54:197 Tuya parse expression
16:14:54:197 TY_DATA_RESPONSE: seq 283, dpid: 0x02, type: 0x02, length: 4, val: 135
16:14:54:197 Tuya parse expression
16:14:54:197 TY_DATA_RESPONSE: seq 283, dpid: 0x02, type: 0x02, length: 4, val: 135
16:14:54:197 Tuya parse expression
16:14:54:197 TY_DATA_RESPONSE: seq 283, dpid: 0x02, type: 0x02, length: 4, val: 135
16:14:54:197 Tuya parse expression
16:14:54:197 TY_DATA_RESPONSE: seq 283, dpid: 0x02, type: 0x02, length: 4, val: 135
16:14:54:197 Tuya parse expression
16:14:54:197 Tuya parse expression
16:14:54:197 TY_DATA_RESPONSE: seq 283, dpid: 0x02, type: 0x02, length: 4, val: 135
16:14:54:350 schedule 2 diff 98, Sun Apr 24 16:16:32 2022
16:14:54:350 schedule 3 diff 22801, Sun Apr 24 22:34:55 2022
16:14:54:350 Idle timer triggered
16:14:54:350 create binding for attribute reporting of cluster 0x0006
16:14:54:350 queue binding task for 0x7CB03EAA00AB3CF1, cluster 0x0006
16:14:54:350 update ddf Ceiling TW OSRAM index 68
16:14:54:350 create binding for attribute reporting of cluster 0x0008
16:14:54:350 queue binding task for 0x7CB03EAA00AB3CF1, cluster 0x0008
16:14:54:351 update ddf Ceiling TW OSRAM index 68
16:14:54:351 create binding for attribute reporting of cluster 0x0300
16:14:54:351 queue binding task for 0x7CB03EAA00AB3CF1, cluster 0x0300
16:14:54:351 update ddf Ceiling TW OSRAM index 68
16:14:54:351 Force binding of attribute reporting for node Kinderzimmerdecke
16:14:54:351 Force read attributes for ZHAOpenClose SensorNode Türsensor Wohnzimmer
16:14:54:351 rule event /config/localtime: 16:14:53.350 -> 16:14:54.350 (1000ms)
16:14:55:008 poll node 00:17:88:01:03:a1:f4:ed-0b
16:14:55:008 Poll light node Flurlicht
16:14:55:056 read attributes of 0x0017880103A1F4ED cluster: 0x0006: [ 16:14:55:056 0x0000 16:14:55:056 ]
16:14:55:056 add task 192 type 19 to 0x0017880103A1F4ED cluster 0x0006 req.id 62
16:14:55:056 Poll APS request 62 to 0x0017880103A1F4ED cluster: 0x0006
16:14:55:163 Poll APS confirm 62 status: 0x00
16:14:55:163 Erase task req-id: 62, type: 19 zcl seqno: 38 send time 0, profileId: 0x0104, clusterId: 0x0006
16:14:55:187 Node data 0x0017880103a1f4ed profileId: 0x0104, clusterId: 0x0006
16:14:55:187 0x0017880103A1F4ED: update ZCL value 0x0B/0x0006/0x0000 after 0 s
16:14:55:350 schedule 2 diff 97, Sun Apr 24 16:16:32 2022
16:14:55:350 schedule 3 diff 22800, Sun Apr 24 22:34:55 2022
16:14:55:352 rule event /config/localtime: 16:14:54.350 -> 16:14:55.350 (1000ms)
16:14:55:704 Node data 0x84182600000e2978 profileId: 0x0104, clusterId: 0x0006
16:14:55:704 0x84182600000E2978: update ZCL value 0x03/0x0006/0x0000 after 0 s
16:14:55:704 ZCL attribute report 0x84182600000E2978 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000
16:14:55:704    payload: 00001000
16:14:55:789 Node data 0x84182600000e313e profileId: 0x0104, clusterId: 0x0006
16:14:55:789 0x84182600000E313E: update ZCL value 0x03/0x0006/0x0000 after 0 s
16:14:55:789 ZCL attribute report 0x84182600000E313E for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000
16:14:55:789    payload: 00001000
16:14:56:175 poll node 58:8e:81:ff:fe:d4:ae:49-01
16:14:56:175 Poll light node Blitzwolf-2
16:14:56:227 read attributes of 0x588E81FFFED4AE49 cluster: 0x0006: [ 16:14:56:227 0x0000 16:14:56:227 ]
16:14:56:227 add task 198 type 19 to 0x588E81FFFED4AE49 cluster 0x0006 req.id 71
16:14:56:227 Poll APS request 71 to 0x588E81FFFED4AE49 cluster: 0x0006
16:14:56:264 Poll APS confirm 71 status: 0x00
16:14:56:264 Erase task req-id: 71, type: 19 zcl seqno: 39 send time 0, profileId: 0x0104, clusterId: 0x0006
16:14:56:268 Node data 0x588e81fffed4ae49 profileId: 0x0104, clusterId: 0x0006
16:14:56:268 0x588E81FFFED4AE49: update ZCL value 0x01/0x0006/0x0000 after 0 s
16:14:56:350 binding/unbinding timeout srcAddr: 0x90FD9FFFFE08B847, retry
16:14:56:351 schedule 2 diff 96, Sun Apr 24 16:16:32 2022
16:14:56:351 schedule 3 diff 22799, Sun Apr 24 22:34:55 2022
16:14:56:351 Wait 2s till query finished
16:14:56:352 rule event /config/localtime: 16:14:55.350 -> 16:14:56.351 (1001ms)
16:14:56:412 Bind response success for 0x84182600000e2423 ep: 0x03 cluster: 0x0008
[... 220 lines cut ...]
16:22:35:913 Tuya parse expression
16:22:35:913 Tuya parse expression
16:22:35:913 TY_DATA_REPORT: seq 241, dpid: 0x02, type: 0x02, length: 4, val: 160
16:22:36:010 Tuya parse expression
16:22:36:010 TY_DATA_RESPONSE: seq 304, dpid: 0x02, type: 0x02, length: 4, val: 160
16:22:36:010 Tuya parse expression
16:22:36:010 TY_DATA_RESPONSE: seq 304, dpid: 0x02, type: 0x02, length: 4, val: 160
16:22:36:010 Tuya parse expression
16:22:36:010 TY_DATA_RESPONSE: seq 304, dpid: 0x02, type: 0x02, length: 4, val: 160
16:22:36:010 Tuya parse expression
16:22:36:010 TY_DATA_RESPONSE: seq 304, dpid: 0x02, type: 0x02, length: 4, val: 160
16:22:36:010 Tuya parse expression
16:22:36:010 Tuya parse expression
[... cut ...]
16:22:36:227 add task 167 type 19 to 0x84182600000E2423 cluster 0x0006 req.id 34
16:22:36:228 Poll APS request 34 to 0x84182600000E2423 cluster: 0x0006
16:22:36:261 Poll APS confirm 34 status: 0x00
```

It tried some other values in iobroker, they all resulted in correct values on the device as well as the API. Finally, I set the value on the device again, the API still was stuck at 1600.

So in conclusion, at the moment changing the heatsetpoint value through iobroker/the API seems to work. Changing the value on the device doesn't seem to affect the value in deconz.

Smanar, 2022-04-24:
>I pulled your changes and re-tested. I don't see any of the 50:32:5f:ff:fe:9f:36:c0-01-ef00/config/heatsetpoint expression anymore though.

lol, was not the effect I wanted.
Can you tried whith the actual code (new version)

Else I I think have localised the problem, I think manup will be able to help me.

[79 more replies]